raspberrypi/firmware #416: trouble updating with rpi-update

mrpetrov, 2015-04-21:
~ $ date
Tue Apr 21 23:30:32 EEST 2015
 ~ $ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
curl: (7) Failed to connect to 199.27.75.133: Network is unreachable
 !!! Failed to download update for rpi-update!
 !!! Make sure you have ca-certificates installed and that the time is set correctly

Possible github issues?

popcornmix, 2015-04-22:
If you run:
```curl  https://raw.githubusercontent.com/Hexxeh/rpi-update/master/rpi-update```
what does it do?

mrpetrov, 2015-04-22:
I've had trouble on my end - rpi-update works fine after I lost some time searching for the problem, found it and fixed it.
Sorry for the noise...

closed 2015-04-22 by mrpetrov