raspberrypi/firmware #418: No way to determine which firmware belongs to which commit.

ali1234, 2015-04-28:
Running `vcgencmd version` outputs a version ID which references a commit hash on the private git repository containing the firmware source code. There is no way for a member of the public to determine which commit in this repository contains the matching firmware binary, other than by cloning the repository and testing every firmware version on a raspberry pi until they find the one with the correct ID.

XECDesign, 2015-04-28:
The raspberrypi-bootloader package (which installs the firmware) has the revision in the changelog.
`zcat /usr/share/doc/raspberrypi-bootloader/changelog.Debian.gz | head`
If you used rpi-update, it installs a /boot/.firmware_revision, containing the revision.
If you end up with mystery firmware outside of the bootloader package or any half-sensible tool, the date given by vcgencmd version is a good indicator of the firmware version.

ali1234, 2015-04-30:
Thanks. Do you mind if I repost this information on stack exchange? It might be easier for others to find it there. http://raspberrypi.stackexchange.com/questions/29991/how-do-i-find-the-firmware-repository-commit-which-matches-the-firmware-version

XECDesign, 2015-04-30:
Of course, please do.

XECDesign, 2015-05-06:
Based on a script provided by @ali1234, this gets auto-generated every day, if there's an update. http://repo.xecdesign.com/revisions.html

closed 2015-06-12 by XECDesign