raspberrypi/firmware #419: [OpenMAX] Field order for interlaced material randomly set

reufer, 2015-04-29:
Hi

There's an issue when playing interlaced video, captured from a German DVB broadcast:
www.reufer.ch/temp/00001.ts

The stream is 50i which is correctly detected by the video decoder, but the field order seems to be randomly set. Sometimes this leads to flickering which might disappear when restarting the stream.

I'm not sure if the stream is 100% sane, since ffprobe says it's 25p.

Regards,
Thomas

popcornmix, 2015-04-29:
So far I can't see this. Played the steram with kodi and command line omxplayer and the deinterlace seems good with no flicker. Can you explain what you are running, and what effect you see (and if you have to pause/seek/wait to see the issue)?

reufer, 2015-04-30:
The problem occurs only without deinterlacer (image fx set to OMX_ImageFilterNone), and setting the HDMI mode to 50i. Using progressive output mode and enabling the deinterlacer, there's no flickering. A seek operation (flushing tunnels, restarting clock) sometimes changes behavior and the video looks fine afterwards.

popcornmix, 2015-04-30:
You need to do more that just set a 1080i mode. 
By default we output a frame if it is due to be displayed every vsync. Without a deinterlacer, interlaced video is just progressive frames that are weaved together (that is what comes out of video decoder).
You need some way of ensuring you present these frames on the correct odd/even vsyncs.

See omxplayer and --native-deinterlace (which uses "vcgencmd hvs_update_fields" to force updates to only odd/even vsyncs).

reufer, 2015-05-02:
Setting hvs_update_fields solves the issue - thanks a lot for clarifying!

closed 2015-05-02 by reufer