raspberrypi/firmware #420: hdmi tv shows first always "raspbery" as cec device at each boot (noobs/raspbian/openelec)

beta-tester, 2015-04-29:
hi,
i don't know if i am right here with that issue.
i have a HDMI TV with three HDMI inputs, at each i have a RPi connected.
each RPi appers at boot into NOOBS, Raspbian and even OpenELEC first as "raspberry" as hdmi device name.
so in case i have NOOBS installed and boot into OpenELEC, that RPi will let the TV popup three times with a hdmi cec device name (1'st at boot into NOOBS, as "raspberry", 2'nd at boot into OpenELEC and finally at OpenELEC, when Kodi initialize the CEC service for remotecontrol. then the RPi is finally show as "Kodi").

i don't exactly know, where the first two "raspberry" popup comming from - from the firmware, from the bootcode/-loader, or from the kernel.
is there an option for the config.txt file to change that name at a very early stage of boot?

i know, there is an option **hdmi_ignore_cec_init=1** to do not init cec and then my tv does not popup the "raspberry" device name. and i do not use cec remote control two of my RPi (i use cec remotecontrolling my RPi only on that RPi, where OpenELEC is running)
BUT, the hdmi cec device names could be very hands to easily identify/select the different connected RPi on the TV menu - but for that i need to be able to change the default devicename "raspberry" to something unique.
the best would be to be able to do it in the config.txt file.
**hdmi_force_cec_name=MyRPi#1**

see this thread, where i asked, but got no helpful answer:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=108518

popcornmix, 2015-04-29:
You can change the initial "raspberry" CEC name with config.txt setting:
```cec_osd_name=MyRPi1```

However that shouldn't be visible with ```hdmi_ignore_cec_init=1```.
Once libcec is running (e.g. through OpenELEC), the firmware is not in control of CEC, so it's up to the application what is reported.

beta-tester, 2015-04-29:
that option ``cec_osd_name=MyRPi1`` i didn't know.
that's exactly what i was looking for.
thank you very much.

beta-tester, 2015-04-29:
one issue i have with that ``cec_osd_name=...`` option.
it seems that the firmware sets the cec ods name always with a fixed length.
even the name i gave is short, it will be shown "filled" with tailing blanks.
"Device (**MyRPi1___**)

compared to Kodi (using libcec), there the tailing blanks does not exist
"Device (**Kodi**)

Ruffio, 2017-01-07:
@popcornmix Is there a fixed length on cec ods?

JamesH65, 2017-12-19:
Had a quick look, could see no specific reason why it might add spaces. However, going to close this as a wont fix since its seems too much work to track it down, for almost zero benefit. Reopen if it is actually a issue.

closed 2017-12-19 by JamesH65

beta-tester, 2017-12-20:
on my old TV you could see if there are tailing blanks at the end or not.
but in the meantime i got a new/different TV.
that TV isn't showing how long the CEC string is in reallity.
so i can't verify if it is still an issue. sorry.