raspberrypi/firmware #421: HDMI mode switching breaks EGL

wm4, 2015-04-30:
Run any EGL application, use ``tvservice`` to change the video mode, the screen becomes black. EGL applications can't recover. It doesn't appear to raise an EGL_CONTEXT_LOST error either.

popcornmix, 2015-04-30:
This is expected behaviour. Any dispmanx layers will be lost when changing HDMI mode. The host code needs to destroy and recreate dispmanx layers (which includes EGL) when changing HDMI mode.

wm4, 2015-04-30:
Seriously? And you're supposed to register a tv service callback to detect this, or what?

Sometimes I wonder if this device was engineered to generate pain...

popcornmix, 2015-04-30:
Who is calling tvservice to change hdmi mode?
What do you expect to happen with your 1920x1080 overlay when someone switches to VGA mode?

wm4, 2015-04-30:
>Who is calling tvservice to change hdmi mode?

The user? Apparently it also happens when the HDMI cable is replugged, but I didn't retest it just now.

> What do you expect to happen with your 1920x1080 overlay when someone switches to VGA mode?

Something reasonable? Like clipping it, or at least doing something standard. And not do something completely unusual that breaks the architecture of everything.

popcornmix, 2015-04-30:
No, unplugging hdmi doesn't cause this behaviour.
You would need to run tvservice to set the HDMI mode (or use the tvservice API explicitly).

julianscheel, 2015-05-04:
@popcornmix Is it possible to rebind the eGL context to a new dispmanx layer without recreating the whole context? I think this would help to support dispman reconfiguration in stuff like Qt5, which currently requires applications to be restarted after mode changes.

popcornmix, 2015-05-04:
The context contains a reference to display which presumably will have changed (potentially to something with a different resolution), so I imagine that is not going to work.
Anything that calls eglGetCurrentDisplay:
https://github.com/raspberrypi/userland/blob/master/interface/khronos/egl/egl_client.c#L2140 
is likely to get confused.

Ruffio, 2017-04-19:
@wm4 can this issue be closed?

wm4, 2017-04-19:
There is no indication that this has been fixed, if you mean that.

JamesH65, 2017-12-19:
Closing as a wont fix. The amount of work to change the current behaviour is not cost effective, given this is something that is rarely done.

closed 2017-12-19 by JamesH65

popcornmix, 2017-12-19:
Also the arm side GL driver is where we are heading and that won't have this specific issue.
(I haven't checked its behaviour but it won't be using the same code as the firmware display/gl driver).

sirus20x6, 2017-12-28:
what about using tv service to turn a monitor off and then back on again? I don't care if the context is unusable while the monitor is off, but when it comes back on hdmi should be in the same mode as before.

popcornmix, 2017-12-28:
Use ``vcgencmd display_power 0`` to turn off display and ``vcgencmd display_power 1`` to turn it back on again. This doesn't tear down the list of displayed elements like using the tvservice so may behave as you desire.