raspberrypi/firmware #423: Mailbox SET_ALLOCATE_MEM, SET_LOCK_MEM and/or mmap(2) are not working correctly on RPi 2?

Terminus-IMRC, 2015-05-02:
Hi. Sorry for my terrible English.

Mailbox SET_ALLOCATE_MEM on RPi 2 returns non-zero handle correctly.
And SET_LOCK_MEM returns an adress which seems to be sane correctly.
And mmap(2) mapping above memory address returns an address which seems to be sane correctly.
But writing to a memory area which is allocated by using mailbox and mmap(2)'ed hangs the linux kernel.

So it seems that mailbox SET_ALLOCATE_MEM, SET_LOCK_MEM and/or mmap(2) are not working correctly on RPi 2.

For this reason, qpu-tutorial in https://github.com/hermanhermitage/videocoreiv-qpu, for example, hangs the linux kernel on RPi 2 while it works correctly on RPi 1.

Regards,
Terminus-IMRC.

popcornmix, 2015-05-02:
Look at this commit:
https://github.com/raspberrypi/userland/commit/3b81b91c18ff19f97033e146a9f3262ca631f0e9

You are likely to need to do something similar when converting from physic to bus addresses.

Terminus-IMRC, 2015-05-03:
Thanks for your reply, popcornmix.
AND'ing bus address with ~0xC0000000 in converting to physical address solved my problems.

(I think it should be ~0xF0000000 for the case that users use another memory flag, though. However, this issue is not the place to claim this.)

Thank you very much!

Terminus-IMRC, 2015-05-03:
No. It must be ~0xC0000000. Sigh.

closed 2015-05-03 by Terminus-IMRC