raspberrypi/firmware #424: video_decode hanging with MPEG2 PES

darkyp, 2015-05-04:
I wrote about my issue in the raspberry forum (https://www.raspberrypi.org/forums/viewtopic.php?f=67&t=108694) but as I did not receive any replies I thought I might be posting at the wrong place. Excuse me if this it not the right place either.

As to the problem, I think I came up with a reproduceable case and I beg for your help. 

Here: http://92.247.92.99/dbg.m2v I uploaded a test file with a MPEG2 PES. The file is 211MB, 2712 packets of 81920 bytes each that are fed into the video decoder this way. The decoder output port is not connected, on FillBufferDone my code just returns the buffer for filling again (in a separate thread).

The decoder will hang at sending 2692 packets without decoding any more and without returning the buffers. Debug information (OMX_GetDebugInformation) shows Received 2712, Transmitted 2692.

Interesting findings:
- if sending from packet number 100 (file offset 8 192 000) the decoder will hang at 2609, i.e. Receiver 2629, Transmitted 2609;
- if sending from packet number 200, decoder hangs at 2692, Received 2712;
- if sending from packet number 300 - the whole file is processed and the decoder does not hang (Received = Transmitted = 2712)

Can someone please test the file and suggest on how to resolve the issue?

Raspberry kernel version that I'm working on: Linux version 3.18.11-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #781 SMP PREEMPT Tue Apr 21 18:07:59 BST 2015

Thank you!

darkyp, 2015-05-08:
Guys (gals?) any insights on this? Has anyone reproduced the problem? Is it actually a problem or is it a problem only with my software / configuration / hardware?

I tried feeding preparsed PES packets and not the 80k stream buffers as they come and I have the decoder stalling again.

popcornmix, 2015-05-09:
I've tried playing dbg.m2v with mplayer, vlc and kodi on linux and none of them will decode a single frame. So I think start with a file that is playable on one of those players and then focus on getting it to play on Raspberry Pi.

darkyp, 2015-05-10:
OK, I uploaded a new file (http://92.247.92.99/dbg_1.m2v) produced with the following command:

dd bs=73331 if=dbg.m2v of=dbg_1.m2v skip=1

The file should now be playable in VLC.

popcornmix, 2015-05-10:
While the updated file plays with vlc, it fails with mplayer and kodi (on x86 linux), so I'm not convinced it is a valid stream.

darkyp, 2015-05-10:
omxplayer will render the stream up to a point though and hang.

I do not think mplayer and kodi should be the criteria that decides upon the validity of a stream especially when not playing it at all. There are certainly other products that verify a stream's validity and if it is not valid there should be a reason specified.

As you wish. I'm merely trying to help the community in building a better product by submitting these reports.

darkyp, 2015-05-12:
Some more information. Playing the dbg_1.mpg file using omxplayer the decoding will stop at 08:47. When it stops, using the left arrow key to go back 30 seconds decoding will go past 08:47 and stop at 12:53, going back to 12:23 this time the file is played till the end (12:59).

darkyp, 2015-05-15:
Here are two more files that cause omxplayer to hang:
http://92.247.92.99/dbg_2.mpg
http://92.247.92.99/dbg_3.mpg

These are smaller files (minute or less). They play fine on Windows: VLC, mplayer and Windows Media Player with ffdshow using libavcodec.

I can produce more. Let me know if needed.

darkyp, 2015-05-15:
Just bought a MPEG2 licence for an older raspberry 1 and the above files play OK with omxplayer. 

Kernel: Linux version 3.12.22+ (dc4@dc4-arm-01) (gcc version 4.7.2 20120731 (prerelease) (crosstool-NG linaro-1.13.1+bzr2458 - Linaro GCC 2012.08) ) #691 PREEMPT Wed Jun 18 18:29:58 BST 2014

omxplayer: Build date: Sat, 22 Mar 2014 20:58:15 +0000, Version   : 39e6342 [master]

darkyp, 2015-05-16:
The issue with the files occurs since the commit from 06.04.2015 (4e2845e0cbe7e701f0431e66a26f730de264e7a6, bump to 3.8.11). I tested with the previous firmwares and the files play fine.

darkyp, 2015-05-18:
Just checking in to update the status of this issue: reverted back to 3.18.9-v7+ from Mar 7 and now everything is OK.

dbg_1 to dbg_3 played fine with 3.18.10, but had problems with four other files. 

Now with 3.18.9 after running for 24 hours and continuously decoding 5 simultaneous live streams no hang so far. 

By going back I remember now, I updated because of the new support for some SPI TFT screen, but I do not currently use it because of another issue with it, probably a matter for a new post. Just briefly describing it - every time the screen is updated the kernel (although PREEMPT) seems to take too much time being busy with SPI and packets get dropped from the DVB USB device. Turning off the blinking cursor solves the problem, but any screen update will cause packets being missed from the DVB stream.

julianscheel, 2015-05-18:
@darkyp You might want to try running the latest kernel, but downgrading only the firmware (start(_x).elf, fixup.dat, bootcode.bin in /boot)
It's more likely that it's caused by firmware than by the kernel, I'd say.

darkyp, 2015-05-18:
@julianscheel Yes, you're right, my writing was misleading. I was referring more to commits than to kernel, though I wrote it so. Wanted to reference the commit actually. I'm aware that video_decode is not part of the kernel but the binary blobs.

popcornmix, 2015-12-07:
Is this still an issue?

Ruffio, 2016-06-29:
@darkyp is this still an issue? Else please close it. Thanks.

darkyp, 2016-07-04:
Unfortunately I cannot test it. I sold the berries as the project could not progress with unreliable sw/hw that would randomly hang even at tasks such as the "trivial" mpeg2 decode. 
I left example files once for testing.

closed 2016-07-04 by darkyp