raspberrypi/firmware #426: Firmware occasionally hangs when using the camera with gst-rpicamsrc

ali1234, 2015-05-10:
I am using gst-rpicamsrc to generate an rtsp stream, and changing the annotation every few frames. After a while the firmware hangs, at which point any invocation of vcgencmd will also hang. The processes will eventually become zombies which cannot be killed. dmesg says the following:

[10201.226896] INFO: task rtpsession-rtcp:12421 blocked for more than 120 seconds.
[10201.226936]       Not tainted 3.18.11+ #781
[10201.226949] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[10201.226963] rtpsession-rtcp D c054cbcc     0 12421   2314 0x00000005
[10201.227045] [<c054cbcc>] (__schedule) from [<c054cfa4>] (schedule+0x40/0x8c)
[10201.227084] [<c054cfa4>] (schedule) from [<c054fffc>] (schedule_timeout+0x214/0x29c)
[10201.227115] [<c054fffc>] (schedule_timeout) from [<c054efb4>] (__down+0x90/0xd4)
[10201.227162] [<c054efb4>] (__down) from [<c0056e3c>] (down+0xc0/0xc4)
[10201.227210] [<c0056e3c>] (down) from [<c0385ac4>] (vchiq_release+0x124/0x330)
[10201.227254] [<c0385ac4>] (vchiq_release) from [<c0139af4>] (__fput+0x90/0x20c)
[10201.227286] [<c0139af4>] (__fput) from [<c0139cd8>] (____fput+0x18/0x1c)
[10201.227325] [<c0139cd8>] (____fput) from [<c003e454>] (task_work_run+0x9c/0xcc)
[10201.227366] [<c003e454>] (task_work_run) from [<c0025ea0>] (do_exit+0x2ec/0xa58)
[10201.227401] [<c0025ea0>] (do_exit) from [<c00266a8>] (do_group_exit+0x50/0xf0)
[10201.227439] [<c00266a8>] (do_group_exit) from [<c0031ab4>] (get_signal+0x1bc/0x6fc)
[10201.227486] [<c0031ab4>] (get_signal) from [<c0011ff0>] (do_signal+0x74/0x400)
[10201.227524] [<c0011ff0>] (do_signal) from [<c0012550>] (do_work_pending+0xc0/0x110)
[10201.227553] [<c0012550>] (do_work_pending) from [<c000e858>] (work_pending+0xc/0x20)
[10321.228073] INFO: task rtpsession-rtcp:12421 blocked for more than 120 seconds.
[10321.228112]       Not tainted 3.18.11+ #781
[10321.228126] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[10321.228140] rtpsession-rtcp D c054cbcc     0 12421   2314 0x00000005
[10321.228221] [<c054cbcc>] (__schedule) from [<c054cfa4>] (schedule+0x40/0x8c)
[10321.228259] [<c054cfa4>] (schedule) from [<c054fffc>] (schedule_timeout+0x214/0x29c)
[10321.228292] [<c054fffc>] (schedule_timeout) from [<c054efb4>] (__down+0x90/0xd4)
[10321.228340] [<c054efb4>] (__down) from [<c0056e3c>] (down+0xc0/0xc4)
[10321.228391] [<c0056e3c>] (down) from [<c0385ac4>] (vchiq_release+0x124/0x330)
[10321.228438] [<c0385ac4>] (vchiq_release) from [<c0139af4>] (__fput+0x90/0x20c)
[10321.228469] [<c0139af4>] (__fput) from [<c0139cd8>] (____fput+0x18/0x1c)
[10321.228506] [<c0139cd8>] (____fput) from [<c003e454>] (task_work_run+0x9c/0xcc)
[10321.228550] [<c003e454>] (task_work_run) from [<c0025ea0>] (do_exit+0x2ec/0xa58)
[10321.228582] [<c0025ea0>] (do_exit) from [<c00266a8>] (do_group_exit+0x50/0xf0)
[10321.228620] [<c00266a8>] (do_group_exit) from [<c0031ab4>] (get_signal+0x1bc/0x6fc)
[10321.228667] [<c0031ab4>] (get_signal) from [<c0011ff0>] (do_signal+0x74/0x400)
[10321.228704] [<c0011ff0>] (do_signal) from [<c0012550>] (do_work_pending+0xc0/0x110)
[10321.228736] [<c0012550>] (do_work_pending) from [<c000e858>] (work_pending+0xc/0x20)
[10441.229324] INFO: task rtpsession-rtcp:12421 blocked for more than 120 seconds.
[10441.229383]       Not tainted 3.18.11+ #781
[10441.229400] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[10441.229415] rtpsession-rtcp D c054cbcc     0 12421   2314 0x00000005
[10441.229498] [<c054cbcc>] (__schedule) from [<c054cfa4>] (schedule+0x40/0x8c)
[10441.229560] [<c054cfa4>] (schedule) from [<c054fffc>] (schedule_timeout+0x214/0x29c)
[10441.229596] [<c054fffc>] (schedule_timeout) from [<c054efb4>] (__down+0x90/0xd4)
[10441.229664] [<c054efb4>] (__down) from [<c0056e3c>] (down+0xc0/0xc4)
[10441.229717] [<c0056e3c>] (down) from [<c0385ac4>] (vchiq_release+0x124/0x330)
[10441.229784] [<c0385ac4>] (vchiq_release) from [<c0139af4>] (__fput+0x90/0x20c)
[10441.229819] [<c0139af4>] (__fput) from [<c0139cd8>] (____fput+0x18/0x1c)
[10441.229860] [<c0139cd8>] (____fput) from [<c003e454>] (task_work_run+0x9c/0xcc)
[10441.229926] [<c003e454>] (task_work_run) from [<c0025ea0>] (do_exit+0x2ec/0xa58)
[10441.229959] [<c0025ea0>] (do_exit) from [<c00266a8>] (do_group_exit+0x50/0xf0)
[10441.230003] [<c00266a8>] (do_group_exit) from [<c0031ab4>] (get_signal+0x1bc/0x6fc)
[10441.230072] [<c0031ab4>] (get_signal) from [<c0011ff0>] (do_signal+0x74/0x400)
[10441.230110] [<c0011ff0>] (do_signal) from [<c0012550>] (do_work_pending+0xc0/0x110)
[10441.230162] [<c0012550>] (do_work_pending) from [<c000e858>] (work_pending+0xc/0x20)
[10441.230186] INFO: task VCHIQ completio:12467 blocked for more than 120 seconds.
[10441.230203]       Not tainted 3.18.11+ #781
[10441.230214] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[10441.230229] VCHIQ completio D c054cbcc     0 12467  12449 0x00000005
[10441.230299] [<c054cbcc>] (__schedule) from [<c054cfa4>] (schedule+0x40/0x8c)
[10441.230335] [<c054cfa4>] (schedule) from [<c054fffc>] (schedule_timeout+0x214/0x29c)
[10441.230367] [<c054fffc>] (schedule_timeout) from [<c054efb4>] (__down+0x90/0xd4)
[10441.230425] [<c054efb4>] (__down) from [<c0056e3c>] (down+0xc0/0xc4)
[10441.230466] [<c0056e3c>] (down) from [<c0385ac4>] (vchiq_release+0x124/0x330)
[10441.230525] [<c0385ac4>] (vchiq_release) from [<c0139af4>] (__fput+0x90/0x20c)
[10441.230558] [<c0139af4>] (__fput) from [<c0139cd8>] (____fput+0x18/0x1c)
[10441.230591] [<c0139cd8>] (____fput) from [<c003e454>] (task_work_run+0x9c/0xcc)
[10441.230649] [<c003e454>] (task_work_run) from [<c0025ea0>] (do_exit+0x2ec/0xa58)
[10441.230683] [<c0025ea0>] (do_exit) from [<c00266a8>] (do_group_exit+0x50/0xf0)
[10441.230717] [<c00266a8>] (do_group_exit) from [<c0031ab4>] (get_signal+0x1bc/0x6fc)
[10441.230776] [<c0031ab4>] (get_signal) from [<c0012128>] (do_signal+0x1ac/0x400)
[10441.230812] [<c0012128>] (do_signal) from [<c0012550>] (do_work_pending+0xc0/0x110)
[10441.230841] [<c0012550>] (do_work_pending) from [<c000e858>] (work_pending+0xc/0x20)
[10561.230609] INFO: task rtpsession-rtcp:12421 blocked for more than 120 seconds.
[10561.230648]       Not tainted 3.18.11+ #781
[10561.230665] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[10561.230679] rtpsession-rtcp D c054cbcc     0 12421   2314 0x00000005
[10561.230757] [<c054cbcc>] (__schedule) from [<c054cfa4>] (schedule+0x40/0x8c)
[10561.230799] [<c054cfa4>] (schedule) from [<c054fffc>] (schedule_timeout+0x214/0x29c)
[10561.230831] [<c054fffc>] (schedule_timeout) from [<c054efb4>] (__down+0x90/0xd4)
[10561.230877] [<c054efb4>] (__down) from [<c0056e3c>] (down+0xc0/0xc4)
[10561.230927] [<c0056e3c>] (down) from [<c0385ac4>] (vchiq_release+0x124/0x330)
[10561.230975] [<c0385ac4>] (vchiq_release) from [<c0139af4>] (__fput+0x90/0x20c)
[10561.231007] [<c0139af4>] (__fput) from [<c0139cd8>] (____fput+0x18/0x1c)
[10561.231047] [<c0139cd8>] (____fput) from [<c003e454>] (task_work_run+0x9c/0xcc)
[10561.231089] [<c003e454>] (task_work_run) from [<c0025ea0>] (do_exit+0x2ec/0xa58)
[10561.231119] [<c0025ea0>] (do_exit) from [<c00266a8>] (do_group_exit+0x50/0xf0)
[10561.231161] [<c00266a8>] (do_group_exit) from [<c0031ab4>] (get_signal+0x1bc/0x6fc)
[10561.231208] [<c0031ab4>] (get_signal) from [<c0011ff0>] (do_signal+0x74/0x400)
[10561.231240] [<c0011ff0>] (do_signal) from [<c0012550>] (do_work_pending+0xc0/0x110)
[10561.231268] [<c0012550>] (do_work_pending) from [<c000e858>] (work_pending+0xc/0x20)
[10561.231291] INFO: task VCHIQ completio:12467 blocked for more than 120 seconds.
[10561.231307]       Not tainted 3.18.11+ #781
[10561.231319] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[10561.231331] VCHIQ completio D c054cbcc     0 12467      1 0x00000005
[10561.231379] [<c054cbcc>] (__schedule) from [<c054cfa4>] (schedule+0x40/0x8c)
[10561.231413] [<c054cfa4>] (schedule) from [<c054fffc>] (schedule_timeout+0x214/0x29c)
[10561.231443] [<c054fffc>] (schedule_timeout) from [<c054efb4>] (__down+0x90/0xd4)
[10561.231477] [<c054efb4>] (__down) from [<c0056e3c>] (down+0xc0/0xc4)
[10561.231514] [<c0056e3c>] (down) from [<c0385ac4>] (vchiq_release+0x124/0x330)
[10561.231552] [<c0385ac4>] (vchiq_release) from [<c0139af4>] (__fput+0x90/0x20c)
[10561.231583] [<c0139af4>] (__fput) from [<c0139cd8>] (____fput+0x18/0x1c)
[10561.231616] [<c0139cd8>] (____fput) from [<c003e454>] (task_work_run+0x9c/0xcc)
[10561.231651] [<c003e454>] (task_work_run) from [<c0025ea0>] (do_exit+0x2ec/0xa58)
[10561.231684] [<c0025ea0>] (do_exit) from [<c00266a8>] (do_group_exit+0x50/0xf0)
[10561.231717] [<c00266a8>] (do_group_exit) from [<c0031ab4>] (get_signal+0x1bc/0x6fc)
[10561.231755] [<c0031ab4>] (get_signal) from [<c0012128>] (do_signal+0x1ac/0x400)
[10561.231789] [<c0012128>] (do_signal) from [<c0012550>] (do_work_pending+0xc0/0x110)
[10561.231820] [<c0012550>] (do_work_pending) from [<c000e858>] (work_pending+0xc/0x20)

ali1234, 2015-05-10:
My rpi-update .firmware_revision is 5b0cbedacf45e111f02d925fa5b1cec9041fb279 - gst-rpicamsrc needs the V3 annotation API. Nobody else seems to be having this problem, but nobody else is updating the annotation as frequently as I do, so that would be where I'd start looking for problems.

pelwell, 2015-05-11:
If you can create a self-contained test-case, ideally one that doesn't require an RTSP (or other) client, then this has a chance of getting looked at.

Ruffio, 2015-06-26:
@ali1234 whats the status on this one?

Ruffio, 2016-06-29:
@ali1234 can this be closed? I pinged you a year ago, but so far no response... Please close this issue or provide a self-contained test-case. Thanks.

Ruffio, 2016-12-30:
Please consider to close this issue due to lack of activity. Last activity was on 10 May 2015, and @ali1234 has been pinged three times since then to come up with a testcase, but without any response.

closed 2016-12-30 by popcornmix