raspberrypi/firmware #427: i2c does not work with 3.18.13-v7+

3PO, 2015-05-11:
Hi,

i2c does not work with commit 9354845

```
raspberrypi ~ # modprobe -v i2c-dev
insmod /lib/modules/3.18.13-v7+/kernel/drivers/i2c/i2c-dev.ko
modprobe: ERROR: could not insert 'i2c_dev': Invalid argument
raspberrypi ~ # modprobe -v i2c-bcm2708
insmod /lib/modules/3.18.13-v7+/kernel/drivers/i2c/busses/i2c-bcm2708.ko
modprobe: ERROR: could not insert 'i2c_bcm2708': Invalid argument
´´´

´´´
raspberrypi ~ # uname -a
Linux raspberrypi 3.18.13-v7+ #784 SMP PREEMPT Sat May 9 15:57:36 BST 2015 armv7l BCM2709 GNU/Linux
raspberrypi ~ #
´´´

pelwell, 2015-05-13:
It works for me:
```
pi@raspberrypi:~$ uname -a
Linux raspberrypi 3.18.13-v7+ #764 SMP PREEMPT Tue May 12 17:53:56 BST 2015 armv7l GNU/Linux
pi@raspberrypi:~$ lsmod
Module                  Size  Used by
lirc_dev                8181  0
rc_core                16932  1 lirc_dev
i2c_dev                 6027  0
snd_bcm2835            18365  0
snd_soc_bcm2708_i2s     6637  0
regmap_mmio             2961  1 snd_soc_bcm2708_i2s
snd_soc_core          136798  1 snd_soc_bcm2708_i2s
snd_compress            7610  1 snd_soc_core
snd_pcm_dmaengine       3359  1 snd_soc_core
snd_pcm                73475  3 snd_bcm2835,snd_soc_core,snd_pcm_dmaengine
snd_seq                53078  0
snd_seq_device          5628  1 snd_seq
snd_timer              17784  2 snd_pcm,snd_seq
snd                    51038  7 snd_bcm2835,snd_soc_core,snd_timer,snd_pcm,snd_seq,snd_seq_device,snd_compress
rtc_pcf8523             3124  0
i2c_bcm2708             4990  0
evdev                   9950  3
uio_pdrv_genirq         2958  0
uio                     8119  1 uio_pdrv_genirq
```
The rtc_pcf8523 is a real time clock, and that is working.

Have you updated your modules? Use "modules" as a build target to make them, and "modules_install" to copy them to /lib/modules/<version>/ and update the module dependencies.

3PO, 2015-05-13:
I dont know, what was wrong, but now i can load the modules. :)

```
raspberrypi ~ # lsmod
Module                  Size  Used by
ipv6                  333976  16
i2c_dev                 6027  0
i2c_bcm2708             4990  0
uio_pdrv_genirq         2958  0
uio                     8119  1 uio_pdrv_genirq
raspberrypi ~ #
```
But my DS3132 RTC doesnt work. :(

@pelwell,
can you please post your "cmdline.txt"?

pelwell, 2015-05-13:
Here it is, but I doubt it will help you:
```
debug dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 ip=dhcp root=/dev/nfs nfsroot=192.168.0.154:/usr/share/piroot,udp,vers=3 rootfstype=nfs elevator=deadline rootwait loglevel=8 earlyprintk
```
There is a dedicated DS3132 driver, but a) it doesn't support all of the features and b) it isn't designed to work with device tree. For a basic RTC driver you can substitute the DS3131 driver (which is actually provided by the DS1307 module...). Add
```
dtoverlay=i2c-rtc,ds3131
```
to your config.txt.

3PO, 2015-05-13:
THX. But it does not work. ;)

```
raspberrypi ~ # ls -la /sys/class/i2c-adapter
total 0
drwxr-xr-x  2 root root 0 May 13 13:25 .
drwxr-xr-x 43 root root 0 Jan  1  1970 ..
raspberrypi ~ #
```

My cmdline.txt:
```
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p3 rootfstype=ext4 elevator=deadline rootwait dtoverlay=rtc-i2c,ds3231

popcornmix, 2015-05-13:
dtoverlay=rtc-i2c,ds3231 goes in config.txt not cmdline.txt

3PO, 2015-05-13:
"dtoverlay=rtc-i2c,ds3231" in "config.txt" does not work.
No Device in "/sys/class/i2c-adapter". :(

pelwell, 2015-05-13:
That's because it should be i2c-rtc, not rtc-i2c.

pelwell, 2015-05-13:
I'm closing this. Any further questions can be asked in the Forums. I suggest here: https://www.raspberrypi.org/forums/viewforum.php?f=44

closed 2015-05-13 by pelwell