raspberrypi/firmware #428: 'vc_vchi_sm_init: failed to open VCHI service' when gpu=16M

diederikdehaas, 2015-05-15:
When specifying `gpu_mem=16` in `/boot/config.txt` you get:
`vc_vchi_sm_init: failed to open VCHI service` 
This does not happen with `gpu_mem=32`.

```
root@rasppi-1b:~# vcgencmd get_mem gpu
gpu=16M
root@rasppi-1b:~# dmesg | grep vc
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=1024 bcm2708.boardrev=0xe bcm2708.serial=0xdcc57b89 smsc95xx.macaddr=B8:27:EB:C5:7B:89 bcm2708_fb.fbswap=1 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 elevator=deadline root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
[    0.143861] bcm2708_vcio: mailbox at f200b880
[    1.263315] vc-mem: phys_addr:0x00000000 mem_base=0x1fa00000 mem_size:0x20000000(512 MiB)
[    1.465847] vc-cma: Videocore CMA driver
[    1.469907] vc-cma: vc_cma_base      = 0x00000000
[    1.474715] vc-cma: vc_cma_size      = 0x00000000 (0 MiB)
[    1.480378] vc-cma: vc_cma_initial   = 0x00000000 (0 MiB)
[    1.513888] vchiq: vchiq_init_state: slot_zero = 0xdec80000, is_master = 0
[    2.435292] vc-sm: Videocore shared memory driver
[    2.465087] [vc_sm_connected_init]: start
[    2.495438] vc_vchi_sm_init: failed to open VCHI service (-1)
[    2.501172] [vc_sm_connected_init]: failed to initialize shared memory service
[    2.535570] [vc_sm_connected_init]: end - returning -1
```
This happens with both `raspberrypi-bootloader` version `1.20150421-1` and after running `rpi-update` (and rebooting) 
```
root@rasppi-1b:~# cat /boot/.firmware_revision 
03b44377dd1f410900d93ca92d39cdacba3544ac
```

popcornmix, 2015-05-15:
Yes, that is expected. Most vchiq services on GPU are disabled when using cut down firmware (which you get with gpu_mem=16), so vc_sm is not supported. Perhaps the kernel log message should be silenced, but it is harmless.

diederikdehaas, 2015-05-15:
Ok, thanks

closed 2015-05-15 by diederikdehaas

diederikdehaas, 2015-05-15:
I think it's better if it weren't silenced since it is informative.
If ppl see the message they can put it in a search engine, find this issue and determine for themselves if it's a problem or not.

Kr4ntz, 2016-07-12:
Totally, absolutely true, @diederikdehaas .

Lewiscowles1986, 2016-09-30:
what is the threshold to enable?

pelwell, 2016-09-30:
I'm not sure what you are asking. Can you rephrase your question?

popcornmix, 2016-09-30:
``gpu_mem=32`` uses the full ``start.elf`` file and so supports VCHI services like ``vc_vchi_sm_init``.
Any less than that (e.g. ``gpu_mem=31``) will use the cut-down firmware (``start_cd.elf``) and not have the services.

descartes, 2018-10-01:
I arrived here as a consequence of the predicted search engine result above. There appear to be many other entries in the RPi forum & flora (eco system) explaining this issue. As it's not technically an error, maybe a message along the lines of "GPU memory setting too low to initialise VCHI service - x, y & z functions will not be available".

Jibun-no-Kage, 2022-12-03:
Just a comment, but giving the 'impression' of an error, is misleading and confusing.  So, I would strongly suggest that the message be made information or a notification, NOT an error, the use of RED text in dmesg is by convention for errors.  Please change the message to any other color than RED, or better change it to an informational or warning that is NOT flagged by dmesg as an error.  This would be the most user friendly approach, no?

pelwell, 2022-12-03:
Commenting on long-closed issues is not the done thing - it notifies everyone who has previously commented on or watched the issue. Open a new issue, if you must.

Jibun-no-Kage, 2022-12-03:
Since, I could not figure out via the web interface how to reply only to you pelwell, I am broadcasting again to the group...

Unless my intent was to do just that.  I am raising a principle, a suggested best practice question that is applicable to all the parties concerned.  Not address a new issue, per se.  If this was a regression, or a new issue, etc., I would agree with you, that opening a new issue is applicable.  But this is a unique opportunity to address a mindset,if  you will, IMHO.

Please consider my previous suggestion, it would benefit many users.  There has been countless times, over my enterprise IT career where that extra bit of polish or consideration, have saved 1000s of man hours, major $$$, etc., because a more informative reporting methodology was considered and implemented in due course, case in point, DELL learned this lesson well, from my past employer, but i digress.