raspberrypi/firmware #429: image_dx/deinterlace: GPU memory leak (regression)

julianscheel, 2015-05-18:
Commit b6e3bf1d5318c8953de521ad06fdb3185bedd6d4 introduces a gpu-side memory leak for me. It seems that whenever the image_fx filter (in advanced mode) is recreated one or two 621K blocks like the following are lost
```
$ vcdbg reloc
...
[  90] 0x1b53a020: used 621K (refcount 1 lock count 0, size   635904, d1rual) 'deinterlace'
...
```
The loss is persistent over the ARM side application lifecycle.

I am trying to make up a little test-snippet for this, but maybe it's obvious to you even without a snippet what went wrong?

popcornmix, 2015-05-18:
Don't worry about the test code for now.
The information you've provided narrows it down pretty well - I'll let you know if I can't reproduce it.

julianscheel, 2015-05-18:
Perfect, thank you

popcornmix, 2015-05-18:
Okay, that seemed to be straightforward - will be fixed in next firmware update

julianscheel, 2015-05-18:
Great, any ETA for the next firmware?

popcornmix, 2015-05-18:
There is a kernel change that wants to go out soon, so shouldn't be long. Tonight or tomorrow I expect.

popcornmix, 2015-05-18:
Can you rpi-update and test?

julianscheel, 2015-05-18:
Looks good! Thank you :)

closed 2015-05-18 by julianscheel