raspberrypi/firmware #432: 'bcm2708-rpi-cm.dtb' cause boot failure of Compute Module

GTrannoy, 2015-05-29:
Compute module doesn't boot when I add the last version of 'bcm2708-rpi-cm.dtb' in the boot directory of emmc.

Stop after "DMA channels allocated":
"...
[    2.118027] bcm2835-cpufreq: min=700000 max=700000
[    2.124773] sdhci: Secure Digital Host Controller Interface driver
[    2.132682] sdhci: Copyright(c) Pierre Ossman
[    2.139270] mmc-bcm2835 20300000.mmc: DMA channels allocated
[    2.187373] sdhci-pltfm: SDHCI platform a"

boot normally without this .dtb file.

I tested to add manually 'bcm2708-rpi-cm.dtb' from the firmware repository because it is not include in the last image of raspbian (2015-05-05).

Previous file I grab from the net couldn't be decompressed with 'dtc' (error blob flag, something like that) but now it is ok, I can convert in .dts and see content.

I suspect a problem in mmc or soc declarations but not expert...

GTrannoy, 2015-05-29:
Please,

Which configuration file is loaded when 'bcm2708-rpi-cm.dtb' is not present ???
Lack of 'bcm2708-rpi-cm.dtb could cause problem with i2s ?

pelwell, 2015-05-29:
There is no fallback - if the correct file for the target (in this case bcm2708-rpi-cm.dtb) is not present, Device Tree is disabled,

Are you running the latest kernel (sudo rpi-update)? Just grabbing a random .dtb file and expecting it to work with your kernel is not guaranteed to work - you should update them together.

Please remember that Compute Module support using stock kernels is still its infancy, and there may be teething problems, but it is best to start by using rpi-update to get the latest release before reporting a problem. If that still shows the problem, then you can include the contents of /boot/.firmware_revision in your report.

GTrannoy, 2015-05-29:
I understand possible bug with young Compute Module. No problem ;-)

Ok. So my Device Tree isn't enable...

Yes, 'sudo rpi-update' is done. I tried download latest release of rpi-3.18.y and recompile several times (because I work on an audio codec but probably blocked without DT...).

Last week, I tried to find the good dtb but I saw the bcm2708-rpi-cm.dtb have been updated 6 days ago and tried it this morning (official from "https://github.com/raspberrypi/firmware/tree/master/boot").
=> Stop loading after DMA channels allocated.

Without this file, 'sudo vcdbg log msg' report lack but Compute Module boots, probably without Device Tree support.

I use the latest firmware that I have just downloading.

GTrannoy, 2015-05-29:
I found the problem ! :-)
It was me... I work with a custom board in which USB from computer is switched between RPi-USB_slave or USB-serial transceiver. Moreover I used other MOSFETs drived by GPIO47 which could be more sensitive than theses ones used on the Compute Module dev kit.
So I observe interferences induced by fast switching of GPIO47 linked to emmc activity.  USB-serial fails and I thought to a boot failure but it was my terminal which stop responding...
Temporarily I solved my problem by deactivating the emmc activity status in dtb and I will correct my board later.

Nevertheless, I have completed the 'bcm2708-rpi-cm.dtb' with additional settings accordingly to others dtb. I will commit on my fork.

closed 2015-05-29 by GTrannoy