raspberrypi/firmware #434: Wrong revision number for older Raspberry Pi

AndrewFromMelbourne, 2015-06-06:
Since the firmware commit "kernel: Fix bcm2835-mmc DMA allocation error handling" (Hexxeh commit a1621d67b20f04acdac602d0b24923ec3cdbfb66) my Raspberry Pi Model B is reporting the wrong (or at least different) revision number. Up until that commit, my Raspberry Pi was reporting revision 000f, it now reports revision 0002. I can revert back to an earlier revision (Hexxeh commit 4f95c894c6a96b265d98f04a889e469833f7e996) and the revision number goes back to the original 000f. I am guessing it was this: "Replacing board rev functions with board_info library"

I have tried the following firmware revisions. The table shows the model reported in dmesg and the revision number from /proc/cpuinfo.

commit | dmesg | revision
----------|----------|-----------
4f95c894c6a96b265d98f04a889e469833f7e996 | Raspberry Pi Model B | 000f
a1621d67b20f04acdac602d0b24923ec3cdbfb66 | Raspberry Pi Model B Rev 2 | 0002
74fca0c56a378f78ace0148b7d439d177ead3c54 | Raspberry Pi Model B Rev 2 | 0002

Revision 74fca0c56a378f78ace0148b7d439d177ead3c54 was "Fix for identifying older model B Pi", but this doesn't appear to have fixed the issue for my Raspberry Pi.

I also have the following Raspberry Pi revisions. None of them appear to be affected:-

model | revision
--------|-----------
Model B | 0002
Model A | 0008
Model B+ | 0010
Model B Pi2 | a010041

pelwell, 2015-06-06:
The old version numbers weren't always used consistently - with multiple suppliers there was some accidental version number re-use - and so the code to derive useful information from the version numbers was quite involved, involving a number of translations.

Your Model B that continues to report revision 0002 must be a Model B Rev 1 with 256MB. I suspect that the other Model B that used to report 000f is actually a Model B Rev 1, boardrev 0002 again, but with 512MB fitted. This was then internally translated to 000f (a Model B Rev 2, all of which had 512MB).

One of the effects of the rewrite has been to separate the version number reported by the device from the derived characteristics. The value appearing in /proc/cpuinfo used to be a massaged version number, but now it is the original value as reported by the device. The model string, which now includes the  revision number, is being calculated after the translation, hence the difference.

Before deciding how to proceed, I'd like to know the following:
1) Are my guesses correct - 0002 has 256MB and what was previously 000f has 512MB?
2) Does the Pi function correctly - there were some material changes between Rev 1 and Rev 2 (swapping I2C busses, for a start) that would cause problems if the board was mis-identified.
3) Has the change caused software problems?

pelwell, 2015-06-06:
And thanks for buying all those Pi's!

AndrewFromMelbourne, 2015-06-06:
Hi @pelwell. Thanks for the reply. 

1)

I can't find the board revision on the PCB. It just says Raspberry Pi (c) 2011.12. The board in question can be seen on the [Raspi.TV Raspberry Pi family portrait](http://raspi.tv/wp-content/uploads/2015/02/Pi-Family-Pi2-cropped_1500.jpg), identified as B Rev 2 (China).

This Raspberry Pi has 512MB of Ram

2)

It functions correctly for me. I am not using i2c on this board, but I have manually loaded the module(s). The i2c device is created as /dev/i2c-1, which appears to be correct to me.

The concern I have is that there is code around that reads the revision number from /proc/cpuinfo and uses the revision to identify the i2c device etc. I know wiringPi uses the revision number. Code I have written for some work I was involved in also makes a decision about which device to open based on the revision. I have some code on github that uses the revision to report a number of attributes ([here](https://github.com/AndrewFromMelbourne/raspberry_pi_revision))

3)

I haven't seen any software problems:- 

This Raspberry Pi has a [320x240 spi TFT](http://www.dx.com/p/mztx-pi-ext-2-5-ips-full-angle-lcd-display-extension-board-module-for-raspberry-pi-green-331576) set up to use fbtft and this is working correctly.

For several reasons, this Raspberry Pi sees the most use out of the five that I own. I use it mostly for software development (via ssh).

@pelwell I hope that helps. Please let me know if you want any more information, or want me to test anything in particular.

pelwell, 2015-06-07:
Thanks, Andrew, that's given me all the information I need. I've submitted a one line change that should make it into one of the next two updates.

AndrewFromMelbourne, 2015-06-07:
Thanks @pelwell, it works for me. Revision is back to 000f.

closed 2015-06-07 by AndrewFromMelbourne