raspberrypi/firmware #435: DispmanX snapshot using resource smaller than display with display rotated.

AndrewFromMelbourne, 2015-06-06:
I have a bug report for the utility I wrote called [raspi2png](https://github.com/AndrewFromMelbourne/raspi2png) that uses the DispmanX snapshot to make a screen shot of the Raspberry Pi. The [issue](https://github.com/AndrewFromMelbourne/raspi2png/issues/6) is that when the Raspberry Pi display is rotated, and a snapshot is created using a resource that has dimensions that are different to the display, the snapshot is blank. 

I have a small example program on gist ([snapshot_resize.c](https://gist.github.com/AndrewFromMelbourne/ac047bcb3d3f813d4764)) that shows the problem. If you run the example the program takes a snapshot with width and height set to half the display width and height and then displays the snapshot centered on the display.

The issue is present with the latest firmware release (31/5/2015 [revision 74fca0c56a378f78ace0148b7d439d177ead3c54](https://github.com/Hexxeh/rpi-firmware/commit/74fca0c56a378f78ace0148b7d439d177ead3c54)), I attempted to see if the issue is new, but I went back as far as 8/11/2014 ([revision 537ce36354588a7a4a3d58323e85ca8e35ec5f1f](https://github.com/Hexxeh/rpi-firmware/commit/537ce36354588a7a4a3d58323e85ca8e35ec5f1f)) and the issue was still present.

popcornmix, 2015-06-16:
Can you test this firmware:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_rotate_snapshot.zip
Please check it doesn't break any other snapshot (especially rotated screen) use cases.

AndrewFromMelbourne, 2015-06-16:
Thank @popcornmix! 

I have tested the linked firmware using all the variations of display_rotate I know of. Using rotation of 0, 1, 2, 3 and then using those rotations with horizontal (0x10000) and vertical flip (0x20000) and both (0x30000).

I have retested all the previous firmware bugs that I have raised and closed. I can find no regression at all.

Thanks for looking at the issue. Works well for me.

popcornmix, 2015-06-17:
Should be in latest rpi-update firmware

AndrewFromMelbourne, 2015-06-17:
Thanks very much @popcornmix. I have updated to 33a6707cf1c96b8a2b5dac2ac9dead590db9fcaa and can confirm that the issue is fixed. Thanks for all the help!

closed 2015-06-17 by AndrewFromMelbourne