raspberrypi/firmware #436: Binaries in hardfp/opt/vc/lib do not have hard-float flag in elf header

rogerjames99, 2015-06-09:
The binaries in the hardfp/opt/vc/lib directory do not have the hard-float flag set in the ELF header. This means that later versions of the ld.so loader will not find them via the ldconfig cache even though they appear to be in there. This may also apply to other parts of the hardfp tree, but I have not checked. Can someone with access check in a rebuild with the correct flags set.

popcornmix, 2015-06-09:
Can you explain exactly how you are detecting this and what you think should be done differently to fix this? The libraries are built with ``-mcpu=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard`` so you'll have to be more specific about what is being requested.

rogerjames99, 2015-06-09:
Mea culpa,

I just rechecked and realised I had used the files from the wrong part 
of the tree. Please kill this issue.

The adafruit pitft install kills the /opt/vc part of the tree and I had 
used the wrong files to replace it. Loads of strange missing so file 
errors! Took a while to twig as the ldconfig cache looked like the files 
where present. Maybe I should suggest to the loader guys that they put 
out a warning if a file is ignored due to wrong flags.

Sorry,

Roger

closed 2015-06-09 by popcornmix