raspberrypi/firmware #437: javafx animation stuttering

predrag-viceic, 2015-06-14:
Hi,
I have tried several kernels around these and it seems that the last working kernel is this one:

https://github.com/Hexxeh/rpi-firmware/commit/c815d34e024b4ceec65fff59e1448429bd909e9f

and the first failing this one:

https://github.com/Hexxeh/rpi-firmware/commit/6f1db5b734a8cf5217cafd983b1ff9c84c1c5013

My app translates images in ImageView(s) from top to bottom at slow speed and stuttering is fairly visible with the latter kernels. I would gladly help in finding out what it is. I have an openjfx for ARM build environment available.

Best,
Predrag

popcornmix, 2015-06-14:
I assume this is only apparent when overclocked? If you remove "gpu_freq=400" does the different between firmwares disappear?

predrag-viceic, 2015-06-14:
no. The differences are still here. I suspect a timer issue, without any proof..

popcornmix, 2015-06-14:
Can you confirm if it is a firmware issue or kernel issue?
e.g. just revert start*.elf and fixup*.dat and see if that fixes it.

predrag-viceic, 2015-06-14:
ok. With start.elf and fixup.dat from the old, working commit (c815d34e024b4ceec65fff59e1448429bd909e9f )and everything else from the newer commits (+ 6f1db5b734a8cf5217cafd983b1ff9c84c1c5013) it works, ie the animation doesn't stutter. I suppose it's a firmware problem then?

popcornmix, 2015-06-14:
Are you sure it's not overclock related. E.g. with an empty config.txt do you still see a performance change between the two versions? (I'm asking as the main change there was overclock related).

predrag-viceic, 2015-06-14:
It doesn't seem to be overclock related as even with the following config.txt, the difference is fairly visible.

hdmi_force_hotplug=1
config_hdmi_boost=4
disable_overscan=1
gpu_mem=384
framebuffer_width=768
framebuffer_height=1024
display_rotate=3 
hdmi_mode=16
hdmi_group=2
disable_splash=1
boot_delay=0

popcornmix, 2015-06-14:
Still present with display_rotate=3 removed? (that will significantly harm performance).
Do you see the issue with /opt/vc/src/hello_pi/hello_teapot demo?
If not is there a simple way of me being able to reproduce this? (I don't use java and don't imagine the issue is java related, so would prefer being able to avoid installing a lot of packages).
Otherwise I may be able to produce a few test builds to confirm the exact change that was significant.

predrag-viceic, 2015-06-14:
You've nailed it.. without display_rotate=3  there is no stutter anymore. However, my app works with portrait oriented touch screens, so what could be done to make this work ?

popcornmix, 2015-06-14:
Well if display_rotate=3 worked with earlier firmware then it should be possible to get that behaviour back, but I need to either reproduce the issue here, or provide you with some test builds to narrow down exactly what is happening.

predrag-viceic, 2015-06-14:
If you could provide me with few test builds, I would be glad to try them all ! After all it's just the matter of swapping two files :)

popcornmix, 2015-06-14:
Okay will do. Probably tomorrow.

popcornmix, 2015-06-15:
Here are two firmware updates that hopefully will narrow down the change to a single commit:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_8d7f1950.zip
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_780a6a76.zip

Can you let me know if these are good/bad.

predrag-viceic, 2015-06-15:
Firmware firmware_8d7f1950 solves the issue ! You're awsome !

popcornmix, 2015-06-15:
Unfortunately not the fix as this is an older firmware. Can you confirm 780a6a76 is bad?
If so then we know the specific commit responsible.

predrag-viceic, 2015-06-15:
I understand... 780a6a76 is bad.

popcornmix, 2015-06-15:
The identified commit was quite large. There were changes to:
1) switching to turbo mode and dependency of related clocks
2) additional locking when powering domains (protecting against a turbo state change)
3) Slight changes to PLL setup

(1) is the largest change, but I don't think that can be the problem as you still have the problem without overclocking. (2) is the change I suspect is causing the issue, although the locks are quite cheap and I believe only grabbed when powering domain - perhaps 60 times a second, so that is surprising it is noticeable. I don't imagine it is (3). 

Start with a build with just (1) in to narrow it down:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_b5e40cd43.zip

predrag-viceic, 2015-06-15:
firmware_b5e40cd43 is good

popcornmix, 2015-06-15:
This includes the PLL change:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_5b95db5f.zip

predrag-viceic, 2015-06-15:
firmware_5b95db5 is also good

popcornmix, 2015-06-15:
Thanks. I've just had the Eureka moment. Took me a while as the bug isn't in the changed code - it just happens to provoke it. The bug is in the code that submits the rotated display update. It's holding a lock it shouldn't when waiting for the next vsync. I'll see if I can refactor that.

popcornmix, 2015-06-15:
This should be a latest firmware with the rotate/lock fix in place:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_rotate_fix.zip

predrag-viceic, 2015-06-16:
This one is good, the stutter is gone. Thanks !

popcornmix, 2015-06-16:
Great. Will be in next rpi-update firmware.

popcornmix, 2015-06-17:
Should be in latest rpi-update firmware

predrag-viceic, 2015-06-18:
Cool, I have tested and it works perfectly ! You can close the issue. Many thanks!

closed 2015-06-18 by popcornmix