raspberrypi/firmware #438: Vertical padding of about 100px with HDMI in portrait orientation

predrag-viceic, 2015-06-14:
The firmware ()16ab84b6ee1c31e749b104b75801fc92eaf53fd6 moves the display up on y axis by about 100px when the HDMI is in portrait orientation. The previous firmware (9ad0e7528ce9d356e7c96fd7b2ed6d3103955d28) is ok . This is my /boot/config.txt :

arm_freq=1000
hdmi_force_hotplug=1
config_hdmi_boost=4
disable_overscan=1
core_freq=500
sdram_freq=500
over_voltage=6
gpu_mem=384
gpu_freq=400
initial_turbo=60
framebuffer_width=768
framebuffer_height=1024
display_rotate=3
hdmi_mode=16
hdmi_group=2
disable_splash=1
boot_delay=0
force_turbo=0

popcornmix, 2015-06-15:
I can reproduce this. We now clip dispmanx elements on all four edges (used to just clip negative coordinates). We are using the unrotated screen dimensions which for you is likely to clip to 768x768.

Swapping the clip coordinated when rotated seems to fix it. Can you confirm this test build is okay:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_rotate_clip.zip

predrag-viceic, 2015-06-15:
Yes , it works now ! I have underestimated the vertical translation, it wa indeed more than 100px. Thanks !

popcornmix, 2015-06-15:
Okay this fix should be in next rpi-update firmware.

popcornmix, 2015-06-17:
Should be in latest rpi-update firmware

predrag-viceic, 2015-06-18:
Cool, I have tested and it works perfectly ! You can close the issue. Many thanks!

closed 2015-06-18 by popcornmix