raspberrypi/firmware #439: Reading frequencies always return 0

Ruffio, 2015-06-16:
Hi.

When I execute the following commands I always get 0 in return:

vcgencmd get_config core_freq
vcgencmd get_config gpu_freq
vcgencmd get_config h264_freq
 vcgencmd get_config isp_freq
vcgencmd get_config v3d_freq

Is that expected?

pi@raspberrypi ~ $ vcgencmd get_config core_freq
core_freq=0

vcgencmd get_config arm_freq is no problem:
2015-06-16 12:25:36 DEBUG - Executing command: vcgencmd get_config arm_freq
2015-06-16 12:25:36 DEBUG - Result of command: arm_freq=900

popcornmix, 2015-06-16:
Yes, if they are at default settings. On Pi2 arm_freq=900 is actually an explicit overclock enabled by default so it does show.
Add ``core_freq=260`` to config.txt and then it will show (after a reboot).

If you want to know the current frequencies then:
```
vcgencmd measure_clock core
vcgencmd measure_clock arm
vcgencmd measure_clock h264
vcgencmd measure_clock isp
vcgencmd measure_clock v3d
```
would be the correct commands.

Ruffio, 2015-06-16:
Thanks. The 'new' commands all shows 250 MHz (idle?).

I understand the argument about that only overclocked values are shown and if not overclocked 0 is shown. However since Pi2 is by default overclocked to 900 MHz this breaks the consistency. In my opinion it would be more consistent to always show the (max configured) values.

popcornmix, 2015-06-16:
The command prints the values in the "config" structure that are non-zero. The "config" structure is mostly zero on boot, but entries in config.txt may override it.

There are a few exceptions where a non-zero value is implicitly filled in. This is generally for a previously optional setting which has proved useful and the default has been swapped. On pi2 arm_freq falls into this category as arm_freq=900 is an implicit config option set by default.

``vcgencmd get_config`` is purely a debugging command to confirm the config settings entered by the user have been recognised. It just displays the contents of the "config" struct. I wouldn't recommended using this for determining settings like current clock speed (there are better calls for that).

closed 2015-06-16 by Ruffio