raspberrypi/firmware #441: tvservice-client: Failed to connect to TV service: -1 (Multiple omxplayer windows)

gsirin, 2015-06-20:
Hi,

when I start 4 Instances of omxplayer in windows mode like this:

```
screen -d -m -S first omxplayer --win "960 540 1920 1080" http://download.blender.org/peach/bigbuckbunny_movies/big_buck_bunny_480p_h264.mov

screen -d -m -S second omxplayer --win "960 0 1920 540" http://download.blender.org/peach/bigbuckbunny_movies/big_buck_bunny_480p_h264.mov

screen -d -m -S third omxplayer --win "0 0 960 540" http://download.blender.org/peach/bigbuckbunny_movies/big_buck_bunny_480p_h264.mov

screen -d -m -S fourth omxplayer --win "0 540 960 1280" http://download.blender.org/peach/bigbuckbunny_movies/big_buck_bunny_480p_h264.mov
```

than it is not possible to use the "tvservice" binary any longer. (What I actually need for taking screenshots of the buffer and/or listening to cec-client events)
I keep getting the error message: tvservice-client: Failed to connect to TV service: -1

I tried to extend the RAM for the GPU and tried to play other streams. Without success.When I stop one video/omxplayer instance or only start 3 instances everything works as expected!?

Can reproduce this on
- Raspberry Pi Model A (raspian)
- Raspberry Pi 2 Model B (raspian minimal)

Any hints or ideas which limitation could be the source?
Thank you in advance!

MilhouseVH, 2015-06-20:
The information given in #373 might explain it.

gsirin, 2015-06-20:
Thanks a lot for the linked issue. Any ideas how to avoid running into this limitation or adjust it to my needs?

popcornmix, 2015-06-20:
You'll need to rebuild the apps without calling bcm_host_init() which starts all vc services (like gencmd). Just starting the service before the call to use the service and then stopping it straight after would avoid the issue.

gsirin, 2015-06-20:
So I fork omxplayer and remove the bcm_host_init() calls and compile it. Right?
Would it be possible to add a switch in future versions to avoid this problem? Or is it too specific to add that possibility for everybody?

robw83, 2015-06-22:
I too am having exactly the same issue. I can only run three screens but get the same error message. If you figure it out; do let me know!

robw83, 2015-06-22:
Hi there

Think I've solved it; mostly :)

I've just tried changing the gpu ram to 512mb (I have a 1gb ram pi) by doing the following:-

1) sudo nano /boot/config.txt

2) Putting in a line, or modiying the line which says "gpu_mem" so we have:-

gpu_mem = 512

3) Rebooting the machine

4) Then, I am able to load 6 omxplayer screens. On the fourth screen I still get the tvservice error, but it loads anyway!

Thanks
Rob ;)

gsirin, 2015-06-22:
Hi Rob,

yepp I have plenty of GPU Ram available and getting the tvservice error in omxplayer as well but the video is playing fine. There is a limitation of 4 connections to (do not know exactly what) a system service for accessing some information or whatever and when this limit is reached you get the error. No problem in the omxplayer case because the video plays but when it comes to grab the buffer for screenshots or try to use the cec-client for hdmi commands than there is no connection left and the operations are not longer possible :(

JamesH65, 2017-12-19:
Closing this issue as questions answered/issue resolved.

closed 2017-12-19 by JamesH65