raspberrypi/firmware #442: No more /proc/config.gz?

jorti, 2015-06-22:
I'm trying to find the config file of the 4.0.5 kernel, but it looks like it isn't available anymore in /proc/config.gz

Is it available somewhere?

popcornmix, 2015-06-22:
It is now built as a module to save some RAM. Try this first:
``sudo modprobe configs``

jorti, 2015-06-22:
Fantastic. Than you.

closed 2015-06-22 by jorti

krummrey, 2016-12-17:
modprobe: ERROR: ../libkmod/libkmod.c:557 kmod_search_moddep() could not open moddep file '/lib/modules/4.4.26-v7+/modules.dep.bin'

Any way to get around this?

popcornmix, 2016-12-17:
@krummrey working here:
```
pi@domnfs:~ $ sudo modprobe configs
pi@domnfs:~ $ zcat /proc/config.gz |grep SND_BCM2835
CONFIG_SND_BCM2835=m
CONFIG_SND_BCM2835_SOC_I2S=m
```
I can only assume you have something corrupt in your install.
Are you running raspbian jessie? Are you up to date (``sudo apt-get update && sudo apt-get dist-upgrade``)? What does ``uname -a`` report?

Marietto2008, 2020-05-01:
root@DESKTOP-N9UN2H3-WLS2:/usr/src/4.19.43-microsoft-standard# sudo cp /proc/config.gz config.gz
cp: cannot stat '/proc/config.gz': No such file or directory

Marietto2008, 2020-05-01:
root@DESKTOP-N9UN2H3:/usr/src/4.19.43-microsoft-standard# sudo modprobe configs
modprobe: FATAL: Module configs not found in directory /lib/modules/4.4.0-19041-Microsoft

pelwell, 2020-05-02:
It doesn't look like you are using a Raspberry Pi, so you shouldn't be posting here.

Marietto2008, 2020-05-02:
linux works the same everywhere....

JamesH65, 2020-05-02:
> the basic working of linux is the same everywhere....

True, but this isn't "the basic workings of linux" as its dependant on the configuration of the kernel build. Which is mostly defined by whatever distribution it is using. So, no, thing is not something we are going to answer, because we cannot answer it.

More to the point, why should we provide technical support to people who are not using our product?

pelwell, 2020-05-02:
Just Google "Linux config.gz".

deomaximo, 2021-02-17:
> It is now built as a module to save some RAM. Try this first:
> `sudo modprobe configs`
> 
> (or add `configs` to /etc/modules if you always want it available).

Thank you so much!!