raspberrypi/firmware #444: sd card corruption and error transferring data

zimmy73, 2015-06-26:
Hello all,

I'm experiencing problems on more than 10 Raspberry B+, Raspbian GNU/Linux 7,
running mrtg, check_mk agent and nothing else,
sd card kingston 8GB class10 (SDC10/8GBSP)

Here is the mount command:
/dev/root on / type ext4 (rw,noatime,data=ordered)
devtmpfs on /dev type devtmpfs (rw,relatime,size=218304k,nr_inodes=54576,mode=755)
tmpfs on /run type tmpfs (rw,nosuid,noexec,relatime,size=44516k,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /run/shm type tmpfs (rw,nosuid,nodev,noexec,relatime,size=89020k)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
/dev/mmcblk0p1 on /boot type vfat (ro,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
none on /var/log type tmpfs (rw,noatime,size=1024k)
none on /var/tmp type tmpfs (rw,noatime,size=1024k)
none on /tmp type tmpfs (rw,noatime,size=1024k)

as you can see the /boot partition is ro, and the /var and /tmp are in-memory filesystem;
swap is disabled

With kernel 3.18.11+, after more or less 20-30 days of good running I have noticed that
they started to have problems in succession, reporting error data transfering on the kern.log file.
 
mmcblk0: error -84 transferring data, sector 0, nr 8, cmd response 0x900, card status 0x0

After a reboot, all the raspberry devices were not able to start, nothing displayed on the HDMI output and the red and green
leds always on ... the sd card was corrupted and I was not able to built it from scratch because it resulted not readable
both from linux and windows.

Now I have done a raspi-update on an "still working" raspberry, loading kernel 4.0.6+, I'm waiting, hoping that the problem was related to
the kernel modules.

Do you know something about this issue?
Do you have suggestions for me?

Any help would be appreciated.

Many thanks

pelwell, 2015-06-26:
There has been a lot of discussion and investigation of SD card corruption, and although the results were not entirely conclusive we have learnt a few things:
* With the first 3.12 release we remove the NO_CMD23 quirk that improved behaviour on some cards but at the expense of some Samsungs. Later a debug option was added so it could be put back.
* Just prior to the first 3.13 release, @popcornmix added extra locking around critical register accesses. 
* Around the same time, a driver for the second SD card interface (called sdhost) was introduced. It is currently still optional, but it can be enabled by adding `dtoverlay=sdhost` to your config.txt file. There is some evidence to suggest that it doesn't suffer from some hardware problems that the other interface does, but the driver is still relatively new.

The most obvious thing to try is to switch to using the sdhost interface. As it says above, just add `dtoverlay=sdhost` to your config.txt files.

zimmy73, 2015-06-26:
Hi Phil,
first of all many thanks for your quick reply.
I'm following your suggestion.

Just to add an information, I have a Raspberry B, with Raspbian GNU/Linux 7 Kernel 3.12.23+ and 
an SD class 4, configured like the Raspberry B+ ... the problem is not rising ... it is running since two months.
What do you think about?

If it's possible I would ask your feedback about the corrupted SD cards:
at this stage I have 5 SD cards not working.
Windows is not able to read the cards and linux reports
mmc0: error -110 whilst initialising SD card
Do you think that there is a way to recover the SD cards or I need to trash them?

Many thanks

zimmy73, 2015-07-03:
Hi Phil,
where I can find information about the sdhost driver and its development?
... I cannot find any information about this issue.
Now the Raspberry B+ are working with the sdhost driver:
kernel: [    2.259525] mmc0: sdhost-bcm2835 loaded - DMA enabled (>2)
I need to wait some days in order to see if the problem is still present.
Let me know something if you can.
Many thanks

pelwell, 2015-07-03:
The sdhost driver uses the second, Broadcom-developed SD card interface on the BCM2835. It has some quirks and issues (sharing the core clock is one of them, requiring slightly more CPU support is another) but so does the Arasan (mmc) interface, and for some cards sdhost seems to work better than mmc, plus with a more flexible clock divider you can hit 50MHz from a 250MHz core clock (the mmc block is limited to an even divider, so you are stuck at 41.6MHz) and get a slight speed-up.

Ruffio, 2016-06-29:
@zimmy73 has your issue been resolved? Then please close this issue. Thanks.

Ruffio, 2017-01-14:
@popcornmix please consider closing this issue

closed 2017-01-14 by popcornmix