raspberrypi/firmware #445: config.txt option to change Default Camera LED Pin

ric96, 2015-07-06:
I required for my project to change the default camera led to another pin, although i was able to do it using  dts-blob.bin method but i found it kind of tedious for such a basic task, we already have a config option to turn off the led it would be nice to have an option to change the pin as well.
thnx.

6by9, 2015-07-06:
config.txt has no mechanism for changing the direction or mode of a GPIO, so you'd get weird results if, for example, you assigned the LED to an input pin.
dt-blob is the correct way to do this. The config.txt option for disabling the LED predated dt-blob, and in some ways ought to be removed in favour of it.

ric96, 2015-07-06:
Then an easier management system should be created or integrated with raspi-config to facilitate this, if disabling the led is also being removed form config.txt then a management system is a must as the led causes serious red tint even ina fairly lit environment.

popcornmix, 2015-07-06:
Ping @pelwell in case any dtoverlay scheme for firmware dtblob is planned.
But for now dt-blob.bin is the way to do this.

pelwell, 2015-07-07:
I agree that creating a custom dt-blob.bin isn't an elegant solution - there are too many steps in involved, there is no easy way to distribute a patch, and it is easy to forget what is in there.

We have discussed the possibility of integrating the dt-blob.bin functionality into the kernel bcm280*.dtb  files. If we did so then the existing overlay/parameter mechanism would automatically be capable of doing this sort of task. But that's the sort of change that requires a lot of consideration - it's not in our schedule, and it won't happen in the next few months.

ric96, 2015-07-07:
Another problem with the dtb is than you have to use the whole format as provided in the example file, for instance just using:

/dts-v1/;

/ {
   videocore {
    pins_2b2 { // Pi 2 Model B rev 1.1
         pin_config {
            pin@default {
               polarity = "active_high";
               termination = "pull_down";
               startup_state = "inactive";
               function = "input";
            }; // pin
            pin@p22 { function = "output"; termination = "pull_down"; }; // Camera LED
          }; //pinconfig
          
          pin_defines {
           pin_define@CAMERA_0_LED {
               type = "internal";
               number = <22>;
            };
          }; //pindefines
    };
   };
  };

Won't boot the system.

JamesH65, 2017-12-19:
@6by9 @pelwell This is an old one, not sure what to do with it. Didn't something to do with the LED get added since this was last referenced?

pelwell, 2017-12-19:
The recent LED change was to propagate the firmware's camera LED and shutdown pins to Device Tree for Linux drivers to access, so the information flow is in the wrong direction to help here.

I have considered making the HAT GPIO configuration capabilities available via config.txt. It wouldn't be user-friendly - a simple hexadecimal encoding of https://github.com/raspberrypi/hats/blob/master/eeprm-format.md#gpio-map-atom-data-type0x0002 to avoid bloating the firmware any more than necessary - called something like `gpiomap`, but that would only take care of half of the requirement; the `CAMERA_0_LED` setting would still be wrong.

I suspect that a complete solution would require a reasonable amount of gpioman hacking to expose the necessary functionality, for questionable gains.

6by9, 2017-12-19:
(Typo in Phil's link above - should be https://github.com/raspberrypi/hats/blob/master/eeprom-format.md#gpio-map-atom-data-type0x0002)
Otherwise I agree that it's far from being a priority, unless a dtoverlay style approach could be adopted with minimal effort (I haven't checked which order dt-blob.bin anf config.txt are processed in).

closed 2018-10-29 by JamesH65