raspberrypi/firmware #446: Syncing to display with MMAL video output

wm4, 2015-07-06:
To avoid judder, you want to have a way to queue video frames for display in a controlled and predictable way. For example, you want to know or control at which vsync a specific frame queued to the MMAL output will be displayed. It's not quite obvious how to achieve this.

``vc_dispmanx_vsync_callback()`` sounds useful in this context, and it looks like the callback is indeed fired at the rate the display refresh rate indicates. I tried the following algorithm:

- send the video frame that should be displayed to the MMAL_COMPONENT_DEFAULT_VIDEO_RENDERER component
- wait until the next vsync event happens
- repeat the previous step if you want the video frame to be visible for multiple vsyncs
- send the next frame

Note that there is no additional waiting, so the next frame will be typically sent as soon as a vsync callback is fired. (With the possibility of some vsync callbacks being without a frame to account for repeated frames.)

This assumes the MMAL output will display the frame it receives as soon as possible, i.e. the next vsync event. This also assumes that the component doesn't queue frames internally, and can render the frame within a vsync. Additionally, the vsync callback and the way the MMAL component works must be related (for example, the vsync callback could be called after internal back and back buffers were switched, and the MMAL component also synchronizes to this event).

The results I get with this method are mixed. Is this perhaps not the right way?

popcornmix, 2015-07-06:
Sounds similar to how kodi works (with mmal and "sync video to display" enabled).

There is a clock that counts vsyncs:
https://github.com/xbmc/xbmc/blob/master/xbmc/video/videosync/VideoSyncPi.cpp#L49
Using vc_dispmanx_vsync_callback:
https://github.com/xbmc/xbmc/blob/master/xbmc/linux/RBP.cpp#L176

and generally you submit the new frame shortly after the vsync event. The GPU will have no trouble rendering the frame within a vsync. The video_render component has no queueing - when next vsync is due you will get the latest frame submitted (if you have submitted more than one that vsync, the latest one is displayed and older ones will just be returned).

wm4, 2015-07-06:
This answers all my questions. Thanks a lot.

I'm not entirely sure why my results are "mixed", but considering it happened with the same file and screen on kodi too, the source of the error is probably something else.

>Sounds similar to how kodi works 

Yep, that's where I got the idea from. But kodi also has a lot of other code, including mysterious OMX things.

By the way, is there a reason why it creates and destroy a display handle on each frame?

closed 2015-07-06 by wm4

popcornmix, 2015-07-06:
> By the way, is there a reason why it creates and destroy a display handle on each frame?

Yes. Because the HDMI mode switching (annoyingly) occurs just after the start of playback. Changing hdmi mode invalidates any dispmanx display handles.

Now since that code was added we now should be wrapping the open/close displays calls:
https://github.com/xbmc/xbmc/blob/master/xbmc/linux/RBP.cpp#L105

So it's possible this is no longer required and the class m_display could be used. However opening/closing the display is a cheap operation done at most 60 times a second, so it won't make much difference.