raspberrypi/firmware #449: OpenMAX hangs in OMX_EmptyThisBuffer

carlonluca, 2015-07-16:
Not sure if this is a fault of my code or not, but I'm experiencing a hang in OMX_EmptyThisBuffer. It seems very similar to #134.

My project is open: https://github.com/carlonluca/pi. I've been reported hangs to happen very rarely when playing a single video file (including audio), it takes many days of continuous playback to reproduce. I had runs of more then 40 days without issues.
As you'll see in the backtrace, one thread is decoding video to texture using EGLImage, one thread is decoding audio and one thread asks OpenMAX for the current position in the media [here](https://github.com/carlonluca/pi/blob/master/piomxtextures_src/omxplayer_lib/OMXClock.cpp#L319). Part of the code is taken from omxplayer.

What seems to be the same issue can be reproduced in seconds when trying to decode multiple video files at the same time on Pi1. Pi2 instead seems to work better. The firmware version on Pi2 is:

    pi@raspberrypi ~ $ /opt/vc/bin/vcgencmd version
    Apr 21 2015 14:42:19 
    Copyright (c) 2012 Broadcom
    version 2d5ad04b63af4233440c3f7c8587108223201102 (clean) (release)
    pi@raspberrypi ~ $ uname -a
    Linux raspberrypi 3.18.11-v7+ #781 SMP PREEMPT Tue Apr 21 18:07:59 BST 2015 armv7l GNU/Linux

The kernel version on Pi1:

    pi@raspberrypi ~ $ uname -a
    Linux raspberrypi 3.18.11+ #781 PREEMPT Tue Apr 21 18:02:18 BST 2015 armv6l GNU/Linux

Can't determine the GPU firmware version on Pi1 cause I'm keeping the locked "scene" under gdb and vcgencmd is not returning (would take other days to reproduce again), but the firmware should be the same of Pi2 (latest Raspbian image available on the Raspberry web site 2015-05-05).

Do you think this is a fault of my code or something firmware related?
This is a backtrace I got from the locked Pi1.

```
(gdb) t a a where

Thread 13 (Thread 0xb3ead450 (LWP 2598)):
#0  0xb58b1dcc in ioctl () from /lib/arm-linux-gnueabihf/libc.so.6
#1  0xb4212014 in completion_thread () from /opt/vc/lib/libvchiq_arm.so
#2  0xb423eda0 in vcos_thread_entry () from /opt/vc/lib/libvcos.so
#3  0xb5a8fc00 in start_thread () from /lib/arm-linux-gnueabihf/libpthread.so.0
#4  0xb58b9728 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
#5  0xb58b9728 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 12 (Thread 0xb36ad450 (LWP 2599)):
#0  0xb5a96770 in sem_wait@@GLIBC_2.4 () from /lib/arm-linux-gnueabihf/libpthread.so.0
#1  0xb422cbec in dispmanx_notify_func () from /opt/vc/lib/libbcm_host.so
#2  0xb423eda0 in vcos_thread_entry () from /opt/vc/lib/libvcos.so
#3  0xb5a8fc00 in start_thread () from /lib/arm-linux-gnueabihf/libpthread.so.0
#4  0xb58b9728 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
#5  0xb58b9728 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 11 (Thread 0xb2ead450 (LWP 2600)):
#0  0xb5a96770 in sem_wait@@GLIBC_2.4 () from /lib/arm-linux-gnueabihf/libpthread.so.0
#1  0xb4228b84 in tvservice_notify_func () from /opt/vc/lib/libbcm_host.so
#2  0xb423eda0 in vcos_thread_entry () from /opt/vc/lib/libvcos.so
#3  0xb5a8fc00 in start_thread () from /lib/arm-linux-gnueabihf/libpthread.so.0
#4  0xb58b9728 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
#5  0xb58b9728 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 10 (Thread 0xb26ad450 (LWP 2601)):
#0  0xb5a96770 in sem_wait@@GLIBC_2.4 () from /lib/arm-linux-gnueabihf/libpthread.so.0
#1  0xb4229744 in cecservice_notify_func () from /opt/vc/lib/libbcm_host.so
#2  0xb423eda0 in vcos_thread_entry () from /opt/vc/lib/libvcos.so
#3  0xb5a8fc00 in start_thread () from /lib/arm-linux-gnueabihf/libpthread.so.0
#4  0xb58b9728 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
#5  0xb58b9728 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 9 (Thread 0xb1b51450 (LWP 2602)):
#0  0xb58ac830 in poll () from /lib/arm-linux-gnueabihf/libc.so.6
#1  0xb42c25f0 in ?? () from /lib/arm-linux-gnueabihf/libglib-2.0.so.0
#2  0xb42c25f0 in ?? () from /lib/arm-linux-gnueabihf/libglib-2.0.so.0
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

---Type <return> to continue, or q <return> to quit---
[... 60 lines cut ...]
#2  0xaf7bb658 in ilcs_pass_buffer () from /opt/vc/lib/libopenmaxil.so
#3  0xaf8c86a0 in COMXCoreComponent::EmptyThisBuffer(OMX_BUFFERHEADERTYPE*) () at /home/luca/project/pi/pi/piomxtextures_src/omxplayer_lib/OMXCore.cpp:369
#4  0xaf8d6d24 in COMXAudio::AddPackets(void const*, unsigned int, double, double, unsigned int) ()
    at /home/luca/project/pi/pi/piomxtextures_src/omxplayer_lib/OMXAudio.cpp:953
#5  0xaf8c54b8 in OMXPlayerAudio::Decode(OMXPacket*) () at /home/luca/project/pi/pi/piomxtextures_src/omxplayer_lib/OMXPlayerAudio.cpp:256
#6  0xaf8c5790 in OMXPlayerAudio::Process() () at /home/luca/project/pi/pi/piomxtextures_src/omxplayer_lib/OMXPlayerAudio.cpp:313
#7  0xaf89bac4 in OMXThread::Run(void*) () from /usr/local/Qt-rasp-5.4.1/plugins/mediaservice/libopenmaxilmediaplayer.so
#8  0xb5a8fc00 in start_thread () from /lib/arm-linux-gnueabihf/libpthread.so.0
#9  0xb58b9728 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
#10 0xb58b9728 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 1 (Thread 0xb41cc000 (LWP 2597)):
#0  0xb5a96770 in sem_wait@@GLIBC_2.4 () from /lib/arm-linux-gnueabihf/libpthread.so.0
#1  0xaf7ba7e0 in ilcs_execute_function_ex () from /opt/vc/lib/libopenmaxil.so
#2  0xaf7bb3f0 in ilcs_execute_function () from /opt/vc/lib/libopenmaxil.so
#3  0xaf7bbdf8 in vcil_out_get () from /opt/vc/lib/libopenmaxil.so
#4  0xaf8cad04 in COMXCoreComponent::GetConfig(OMX_INDEXTYPE, void*) const () at /home/luca/project/pi/pi/piomxtextures_src/omxplayer_lib/OMXCore.cpp:1199
#5  0xaf8cfec4 in OMXClock::OMXMediaTime(bool) () at /home/luca/project/pi/pi/piomxtextures_src/omxplayer_lib/OMXClock.cpp:337
#6  0xaf87b364 in OMX_MediaProcessor::streamPosition() () at /home/luca/project/pi/pi/piomxtextures_src/omx_mediaprocessor.cpp:558
#7  0xaf8e3b00 in OpenMAXILPlayerControl::position() const () at /home/luca/project/pi/pi/piomxtextures_qt_driver/mediaplayer/openmaxilplayercontrol.cpp:518
#8  0xb026ffe4 in ?? () from /usr/local/Qt-rasp-5.4.1/qml/QtMultimedia/libdeclarative_multimedia.so
#9  0xb026ffe4 in ?? () from /usr/local/Qt-rasp-5.4.1/qml/QtMultimedia/libdeclarative_multimedia.so
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

dimovnike, 2015-07-31:
I experience this issue on many running PI's too, but the only difference is that I can not restart the aplication. Also when trying to run the app again, strace says:

```
lags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, parent_tidptr=0xb261c4b8, tls=0xb261c910, child_tidptr=0xb261c4b8) = 5699
ioctl(4, 0xc40c, 0x70a003)              = 0
ioctl(4, 0xc40c, 0x70a003)              = 0
ioctl(4, 0x400cc404, 0xbe9eb67c)        = 0
ioctl(4, 0xc40d, 0x70a003)              = 0
ioctl(4, 0xc40c, 0x70a003)              = 0
ioctl(4, 0xc010c408, 0xbe9eb668)        = -1 EAGAIN (Resource temporarily unavailable)
futex(0xb41a71b8, FUTEX_WAIT_PRIVATE, 0, NULL
```

... and waits. Also even with my application killed, the /opt/vc/bin/vcgencmd can not start, strace says it is waiting for the same FUTEX.

popcornmix, 2015-08-06:
Can you describe how to reproduce the problem?
E.g. provide binaries or instructions to build it, a sample file and instructions of how to run it.
I can look on the firmware side to see if there is an obvious problem.

carlonluca, 2015-08-08:
Thank you for answering. Unfortunately it won't be simple... the backtrace I provided was related to a run that lasted some days playing a single video file. As you can see there is one thread for audio and one for video. I wasn't able to increase the frequency of the locks, but I noticed that using multiple players seems to make it very very simple to lock. It just takes a few seconds. This is the resulting backtrace: http://pastebin.com/CmjRJAgm. In this case you can see two video/audio players and one audio player.

Seems pretty similar to me. If you consider this reasonable, I can provide what is needed to reproduce this in seconds/minutes. If we are lucky it is the same issue.

I'm assuming you start from a clean Raspbian (I'm using 2015-05-05) using a Pi1. Pi2 seems to work a little better.

1. Download the Qt libraries from here: https://goo.gl/MhsBGA. lrunzip and untar.
2. Untar Qt-rasp-5.5.0.tar and qtdeps.tar. Place the directory Qt-rasp-5.5.0 in /usr/local into the pi and qtdeps wherever you want (in the Pi).
3. Allocate 256MB of memory for the GPU in /boot/config.txt.
4. Place this QML file into the Pi: https://github.com/carlonluca/pi/blob/master/piomxtextures_samples/video_audio_multi.qml.
5. Add the lib dir which is into qtdeps to LD_LIBRARY_PATH.
6. Use the qmlscene command to start the QML file providing the proper arguments:

/usr/local/Qt-rasp-5.5.0/bin/qmlscene video_audio_multi.qml file:///home/pi/big_buck_bunny_720p_h264.mov file:///home/pi/sintel_trailer_720p.mp4 file:///home/pi/master_of_puppets.mp3

So provide two video files and an audio file (like wave or mp3), in this order. Use a URI, not a path.

I just tried this procedure and it seems to work. I hope I didn't forget anything. Please let me know if something goes wrong.
If after some minutes everything seems to work properly, try to kill and restart the app or restart the Pi and try again. There are cases in which, for some reason, it locks in a few seconds, some others where it keeps working.

I could even reproduce with this QML: https://github.com/carlonluca/pi/blob/master/piomxtextures_samples/video_position.qml. One video file only and a timer frequently asking for the current position. But it really requires days to lock with the baktrace of the first message.

dimovnike, 2015-08-27:
Hello, I catched another lock up and created a coredump from GDB. Is it of any use (i can send it with the QT app)?

carlonluca, 2015-08-29:
@popcornmix I read a comment you wrote some time ago in #377:

"I believe mmal and openmax are thread safe, so you don't need additional locking.
Obviously you need to protect your own structures (e.g. when accessed from a mmal callback)."

And in fact in omxplayer omx calls seem not to be guarded. Can you confirm that OpenMAX libs in raspberry are thread-safe?

popcornmix, 2015-09-01:
Yes, openmax calls are thread safe.

carlonluca, 2015-10-13:
By doing a large amount of seeks I can now reproduce a similar situation "quickly". This backtrace is a little clearer:

```
(gdb) t a a where

Thread 13 (Thread 0xb3c52450 (LWP 2415)):
#0  0xb5890dcc in ioctl () at ../sysdeps/unix/syscall-template.S:82
#1  0xb4068014 in completion_thread () from /opt/vc/lib/libvchiq_arm.so
#2  0xb4094da0 in vcos_thread_entry () from /opt/vc/lib/libvcos.so
#3  0xb5a6ec00 in start_thread (arg=0xb3c52450) at pthread_create.c:306
#4  0xb5898728 in ?? () at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:116 from /lib/arm-linux-gnueabihf/libc.so.6
#5  0xb5898728 in ?? () at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:116 from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 12 (Thread 0xb3452450 (LWP 2416)):
#0  0xb5a75770 in __new_sem_wait (sem=0xb4090d70) at ../nptl/sysdeps/unix/sysv/linux/sem_wait.c:60
#1  0xb4082bec in dispmanx_notify_func () from /opt/vc/lib/libbcm_host.so
#2  0xb4094da0 in vcos_thread_entry () from /opt/vc/lib/libvcos.so
#3  0xb5a6ec00 in start_thread (arg=0xb3452450) at pthread_create.c:306
#4  0xb5898728 in ?? () at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:116 from /lib/arm-linux-gnueabihf/libc.so.6
#5  0xb5898728 in ?? () at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:116 from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 11 (Thread 0xb2c52450 (LWP 2417)):
#0  0xb5a75770 in __new_sem_wait (sem=0xb408feec) at ../nptl/sysdeps/unix/sysv/linux/sem_wait.c:60
#1  0xb407eb84 in tvservice_notify_func () from /opt/vc/lib/libbcm_host.so
#2  0xb4094da0 in vcos_thread_entry () from /opt/vc/lib/libvcos.so
#3  0xb5a6ec00 in start_thread (arg=0xb2c52450) at pthread_create.c:306
#4  0xb5898728 in ?? () at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:116 from /lib/arm-linux-gnueabihf/libc.so.6
#5  0xb5898728 in ?? () at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:116 from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 10 (Thread 0xb2452450 (LWP 2418)):
#0  0xb5a75770 in __new_sem_wait (sem=0xb4090c74) at ../nptl/sysdeps/unix/sysv/linux/sem_wait.c:60
#1  0xb407f744 in cecservice_notify_func () from /opt/vc/lib/libbcm_host.so
#2  0xb4094da0 in vcos_thread_entry () from /opt/vc/lib/libvcos.so
#3  0xb5a6ec00 in start_thread (arg=0xb2452450) at pthread_create.c:306
#4  0xb5898728 in ?? () at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:116 from /lib/arm-linux-gnueabihf/libc.so.6
#5  0xb5898728 in ?? () at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:116 from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 9 (Thread 0xb1a48450 (LWP 2419)):
#0  0xb588b830 in __GI___poll (fds=0xa40e70, nfds=1, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:87
#1  0xb56d45f0 in ?? () from /lib/arm-linux-gnueabihf/libglib-2.0.so.0
#2  0xb56d45f0 in ?? () from /lib/arm-linux-gnueabihf/libglib-2.0.so.0
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

---Type <return> to continue, or q <return> to quit---
Thread 8 (Thread 0xaedde450 (LWP 2420)):
#0  0xb5a73504 in __pthread_cond_wait (cond=0xae95e8, mutex=0xae95cc) at pthread_cond_wait.c:153
#1  0xb4094e80 in _timer_thread () from /opt/vc/lib/libvcos.so
#2  0xb5a6ec00 in start_thread (arg=0xaedde450) at pthread_create.c:306
#3  0xb5898728 in ?? () at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:116 from /lib/arm-linux-gnueabihf/libc.so.6
#4  0xb5898728 in ?? () at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:116 from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 7 (Thread 0xae5de450 (LWP 2421)):
#0  0xb5a75770 in __new_sem_wait (sem=0xae977c) at ../nptl/sysdeps/unix/sysv/linux/sem_wait.c:60
#1  0xb406aab0 in vchiu_queue_pop () from /opt/vc/lib/libvchiq_arm.so
#2  0xaf9b2b9c in ilcs_task () from /opt/vc/lib/libopenmaxil.so
#3  0xb4094da0 in vcos_thread_entry () from /opt/vc/lib/libvcos.so
#4  0xb5a6ec00 in start_thread (arg=0xae5de450) at pthread_create.c:306
#5  0xb5898728 in ?? () at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:116 from /lib/arm-linux-gnueabihf/libc.so.6
#6  0xb5898728 in ?? () at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:116 from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 6 (Thread 0xaddde450 (LWP 2422)):
#0  0xb588b830 in __GI___poll (fds=0xaf1728, nfds=1, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:87
#1  0xb56d45f0 in ?? () from /lib/arm-linux-gnueabihf/libglib-2.0.so.0
#2  0xb56d45f0 in ?? () from /lib/arm-linux-gnueabihf/libglib-2.0.so.0
[... 67 lines cut ...]

So, as I see that OMX_EmptyThisBuffer call is not returning, do you know of any possible reason why OMX_EmptyThisBuffer may behave like this? If I knew a possible reason for this I may be able to somehow fix my implementation.

In this situation the Pi is stuck. I won't be able to run applications like omxplayer or vcgencmd. If I try to debug vcgencmd with parameter "version" I get this:

```
(gdb) t a a where

Thread 2 (Thread 0xb6e36470 (LWP 2501)):
#0  0xb6efadcc in ioctl () at ../sysdeps/unix/syscall-template.S:82
#1  0xb6fa2014 in completion_thread () from /opt/vc/lib/libvchiq_arm.so
[... cut ...]
#5  0xb6f02728 in ?? () at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:116 from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 1 (Thread 0xb6ff9000 (LWP 2498)):
#0  0xb6f8e770 in __new_sem_wait (sem=0x12838) at ../nptl/sysdeps/unix/sysv/linux/sem_wait.c:60
#1  0x000099a4 in vc_gencmd_read_response ()
#2  0x000091bc in main ()
```

The process never exits. Probably useless but I try any way.
@popcornmix do you have any other advice on how to investigate further and track down what I may be doing wrong?

6by9, 2015-10-14:
It sounds like you've managed to get the GPU stuck such that it can't process the EmptyThisBuffer call.
There may be some useful logging if you add "start_debug=1" to config.txt, then "sudo vcdbg log assert" should show any asserts that have fired in the firmware (it directly accesses memory rather than using an IPC, so should work even if the GPU has locked).

NB There is a difference between blocking such that buffer queues can be made thread safe, rather than blocking for the entire buffer to be processed. In the Pi case, the buffer queues are on the GPU, so have to go via

popcornmix, 2015-10-14:
@carlonluca If you can provide a test app I can run (e.g. a tarball I can unpack and run an executable, I may be able to see how the GPU is stuck). But, get the assert log suggested by @6by9 first, as that may contain some clues.

carlonluca, 2015-10-14:
This is what I had before starting the test:

```
001018.913: assert( source ) failed; ../../../../../middleware/confzilla/cp_front_fdt.c::cp_front_fdt_load_builtin line 115
004962.490: assert( limit_turbo == last_limit_turbo ) failed; ../../../../../helpers/arm_loader/arm_loader.c::tempcheck_task line 2380
008167.483: assert( limit_turbo == last_limit_turbo ) failed; ../../../../../helpers/arm_loader/arm_loader.c::tempcheck_task line 2380
```

this is what I got after the lock:

```
pi@raspberrypi ~ $ sudo vcdbg log assert
001018.913: assert( source ) failed; ../../../../../middleware/confzilla/cp_front_fdt.c::cp_front_fdt_load_builtin line 115
004962.490: assert( limit_turbo == last_limit_turbo ) failed; ../../../../../helpers/arm_loader/arm_loader.c::tempcheck_task line 2380
008167.483: assert( limit_turbo == last_limit_turbo ) failed; ../../../../../helpers/arm_loader/arm_loader.c::tempcheck_task line 2380
1771862.166: assert( elem ) failed; ../../../../../middleware/dispmanx/dispmanx.c::dispmanx_element_change_source line 5551
4112655.155: assert( elem ) failed; ../../../../../middleware/dispmanx/dispmanx.c::dispmanx_element_change_source line 5551
2353721.195: assert( elem ) failed; ../../../../../middleware/dispmanx/dispmanx.c::dispmanx_element_change_source line 5551
```

I made a couple of tests and the second time I got nothing at all, just the first three lines.

The procedure is similar to what I described above, unfortunately running requires some libs, plugins etc... I can provide the newer version, I can lock even with one player only now, but if you want something quicker I could provide a raspbian image with everything in place. You just run the binary. Would that make things simpler for you?
Thanks you for your answers.

popcornmix, 2015-10-14:
I don't think any of the asserts are necessarily relevant. 
A raspbian sdcard image is okay.

carlonluca, 2015-10-21:
Hello, I prepared the image (https://goo.gl/cL3fld). It is intended for armv6 as the issue seems to manifest quicker on Pi1. I built the code in debug so you can get a backtrace if you need it. What you have to do is simply flash it, boot it and, from the home of the pi user run the proper command.
I can reproduce similar situations in several ways, always hanging in ilcs_execute_function_ex. I can advice a couple.

* Running this makes my Pi's hang quickly, even after a few seconds. This almost always locks:

```
./piomxtextures_pocplayer --multipleanimtest /home/pi/big_buck_bunny_1080p_h264.mov /home/pi/big_buck_bunny_1080p_h264.mov
```

I suggest you use a 720p video instead of the 1080p I provided in the home, because this player plays both videos concurrently. What you should see is something like this: https://youtu.be/MuPGAWp7vag. The framerate is going to be far worse as the one in the video is a Pi2 and without all those debug logs. It should hang pretty soon always with the same backtraces I already reported.

* Running this instead takes much more time, but it typically hangs anyway:

```
/usr/local/Qt-rasp-5.5.0/bin/qmlscene samples/video_seek.qml file:///home/pi/big_buck_bunny_1080p_h264.mov
```

this uses just a single player, like omxplayer, but still seems to hang in ilcs_execute_function_ex as I reported above. This typically requires from 10,000 to 100,000 seeks to hang (500ms for a seek so it takes hours). I understand it is difficult to reproduce something like this, so the first option may be the best to start with.

Running a single player without further interaction with openmax seems to be ok (I had a run of over 45 days). But trying to get the current position often using OMX_GetConfig or similar may result in these situations I reported after hours or days of playback. These are the best codes I found to let you reproduce quickly.
Any possible advice you can provide may be useful to fix my code or to find a workaround. Please let me know if you need anything else. Thanks!

carlonluca, 2015-11-03:
@popcornmix were you able to reproduce the issue? Is there anything else I can do to help you with this?

popcornmix, 2015-11-03:
I've had a look. I don't see anything suspicious on gpu side. No memory corruption. All tasks call stacks are valid with no sign of deadlock. Everything is idle waiting for a command.

The gencmd hanging was suspicious. Quit the piomxtextures_pocplayer process and the gencmd completes.

I wondered if this was due to too many gencmd clients (we only support 3). I tried rebuilding the firmware to support 10 gencmd clients and it still hangs. Checking, I am only getting one VCHI_CALLBACK_SERVICE_OPENED  and a few VCHI_CALLBACK_MSG_AVAILABLE when launching piomxtextures_pocplayer. 

After piomxtextures_pocplayer has stalled, then gencmd causes a VCHI_CALLBACK_SERVICE_OPENED message to get though but no VCHI_CALLBACK_MSG_AVAILABLE which is why it hangs. When piomxtextures_pocplayer is quit I see the VCHI_CALLBACK_MSG_AVAILABLE and the two VCHI_CALLBACK_SERVICE_CLOSED messages.

So, this feels like a vchiq deadlock issue, rather than an openmax issue. I think I need to call in @pelwell for vchiq help here. Phil is it possible you can look at this? I have an sdcard image that provokes the hang in a small number of seconds.

pelwell, 2015-11-03:
OK. Leave it where I can look at it in the morning.

pelwell, 2015-11-04:
I think I understand this now. In the logging I captured there were a number of places where VCHIQ - the driver, the user library or the queue utility that the ILCS client uses - was stalling because a queue was full, but I now think this was a symptom of the real problem, which is an ILCS limit on fhe number of concurrent outstanding requests coupled with a complication of allowing application callbacks to make further requests.

When a message is sent that expects a response, the calling thread blocks until the response comes back. This is implemented by associating a semaphore with a transaction ID, and signalling the semaphore when a response with a matching transaction ID is received. For reasons of either convenience or resource limiting, there is a fixed number of these semaphores available (4). If a thread finds that all semaphores are in use then it must wait for one to become available due to the completion of an outstanding request.

This mechanism would be simple and safe were it not for the fact that callbacks into applications are allowed to make requests. Such callbacks are made in the context of the server thread, and if the server thread is blocked then no responses are processed, thus the consequence of the server thread failing to get a slot is deadlock.

The function in question (ilcs_execute_function_ex) pays lip service to the problem without actually doing enough to avoid it. My work around is for the server thread to try to claim the event in a non-blocking fashion, and on failure to process some messages before trying again. If the second attempt fails then it sleeps for 1 millisecond before going round the loop again. This logic has been enough to stop the test from stalling, without the need to increase the concurrency above 4.

You can find the patch in [this](https://gist.github.com/pelwell/1bd72ee37258b9851fb4) gist, to be applied against the userland repo. Copy the build products from build/lib to /opt/vc/lib .

pelwell, 2015-11-05:
With that patch applied the test has been running for 15 hours without stalling. Since there are going to be performance penalties caused by the server thread waiting and retrying, I'll also double the thread limit to 8 - the additional memory and search time is negligible.

carlonluca, 2015-11-05:
Thank you @pelwell for you work! I'll test your patch asap. This may explain why I was able to prevent a bit the lockup by calling omx only inside a mutex zone.
Do you think the seek lockup (shown above https://github.com/raspberrypi/firmware/issues/449#issuecomment-147879790) is another symptom of the same issue? I'm asking because in that case, using a mutex zone on my side was not enough for preventing lockup. However as you can see from the backtrace, the situation seemed to be exactly the same.

pelwell, 2015-11-05:
I would say that the seek lockup isn't another manifestation of the same problem - I would expect to see at least four threads waiting in ilcs_execute_function_ex, when in the traces above there is only one.

carlonluca, 2015-11-05:
Do you have any advice on how I may be able to investigate that problem further? Are there other possible reasons why that thread may be stuck in ilcs_execute_function_ex? Any other idea off the top of your head?

pelwell, 2015-11-05:
Sorry, no - yesterday was my first real skirmish with the OpenMAX code.

My suggestion would be to fetch the userland repo, modify buildme or the CMake files to build a debug version (this is easily buildable on the Pi), and copy the contents of build/lib into /opt/vc/lib. That way, the backtrace should at least tell you which line of the source it was on.

popcornmix, 2015-11-07:
@pelwell's patch is in latest rpi-update libs.

carlonluca, 2015-11-07:
Thanks @popcornmix, I'm already testing it and I see there is a improvement in the second situation, but just like @pelwell said, the "seek test" still seems to manifest issues. But something may have changed, I'm following @pelwell's advice and I'll try to provide some more info assuming you are interested.

pelwell, 2015-11-07:
We're always interested, it's just a matter of time and priorities.

carlonluca, 2015-11-07:
I perfectly understand time is a limited resource and that this issue may not be affecting many users as not exactly a "regular" use.

However I'm running more tests and what I can see is that the first of the two tests is now far better, but I can confirm I can lock the "seek test". This is the backtrace after a lockup that includes more info related to the userland libs: http://pastebin.com/Ek5P4y5q. What I see is that the situation is somehow similar, but we are probably in another case (vcos_thread_current() != &st->thread). Please note that in this case there is no mutex on my side before calling openmax.

As you probably read above, my code somehow integrates into omxplayer code and is built over it. I therefore enabled the logs from omxplayer code and this is the last portion of 130MB of logs: http://pastebin.com/Mv6idTHA. Not sure if these can be of any use.

Of course this again can be a bug in my code, but it is difficult to say. Do you guys have any other advice?

pelwell, 2015-11-07:
That backtrace looks like the ARM is waiting for responses from the VPU, while the server thread is ready to receive such responses. In other words, I think the problem is likely to be in the VPU.

carlonluca, 2015-11-10:
Is there anything I can do to somehow collect more information about this? May be a bug in my code and more information may help finding the bug... Can I debug the issue in any way? Any other advice?

pelwell, 2015-11-10:
The best thing you can do is to find a simple test case that fails earliest and then give it to us. Cutting out as much as you can helps us to focus on what is left. If we have everything we need to reproduce the problem in some relatively short time then there's a good chance we'll take a look.

carlonluca, 2015-11-17:
@pelwell after months the only way I could find to reproduce quicker is the first method you used. Unfortunately that turned out to be another separate issue it seems. So I am not aware of any quicker way to reproduce it.

pelwell, 2015-11-17:
When you wrote:

> However I'm running more tests and what I can see is that the first of the two tests is now far better, but I can confirm I can lock the "seek test"

I assumed that you would be able to package up the "seek test" so that we can run just that.

carlonluca, 2015-11-21:
Yes, sure you can run just the second one, I provided it in the firmware you already have: https://github.com/raspberrypi/firmware/issues/449#issuecomment-149877489. Just run that second command and you should see it running. Just one note: if you wanted to run it, place replace this line "var length = element.duration - 1000;" with "var length = element.duration - 3000;" in the .qml file appearing in the command. This just because I recently had a case in which the player simply reached end of media and interrupted the test. No need to rebuild anything.

In the firmware I provided both samples because I suspected it was possible those were not the same issue. Now, this test performs random seeks. In case you need it, you can tune the frequency of the seeks by simply editing the .qml file you see in the command. In the Timer element there is an "interval" property. Reducing the period reduces the time needed to reproduce, but I wouldn't go below 300ms, I don't think my implementation is sufficiently solid. I typically run with an interval of 500ms because I'm not interested in doing 1*10^12 seeks in 1 second :-), I'm just providing these samples to reproduce as quickly as possible lockups that have backtraces and behaviours similar to issues that I've been told to manifest after days and days of run of apps using my open project. I understand it is nearly impossible to debug something that locks once in 15 days or more, so these samples are the best I could make up to reproduce. If this was still too much time-consuming for you (requires hours), I'd perfectly understand.

I just ran another test, it lasted 7 hours (it is the average time needed to reproduce). I'm attaching both the bt with all the debug info and the last portion of the output that includes the logs that are included in omxplayer code (my project is based largely on that code).
Thank you for all your help.

[bt_seek.txt](https://github.com/raspberrypi/firmware/files/40858/bt_seek.txt)
[output_seek.txt](https://github.com/raspberrypi/firmware/files/40859/output_seek.txt)

carlonluca, 2016-04-03:
@pelwell, I was able to reproduce a similar issue, actually it seems to be identical, with another sample. In this case it always seems to be the flush that is causing the deadlock. It is pretty simple to reproduce, I can deadlock a few minutes or even less. In this case there are three players and some people reported deadlocks. I attach the backtrace: [backtrace.txt](https://github.com/raspberrypi/firmware/files/201342/backtrace.txt). I took it with a recent build of the userland libs in debug mode.

My interpretation of this situation is that Thread 1 is blocked on a mutex which is held by Thread 10. This is correct according to my code (https://github.com/carlonluca/pi/blob/master/piomxtextures_src/omx_mediaprocessor.cpp#L652). Thread 10 is blocked on a wait condition waiting for the player thread to complete (Thread 19): Thread 10 is executing a synchronous stop command and so is waiting for the player thread to die. That thread instead won't die because it is flushing, waiting in the famous ilcs_execute_function_ex function that you already patched once. Many threads are blocked in there: 19, 18, 17, 15 and 14. This situation is then permanent.

Do you think my interpretation makes sense? I understand a backtrace like this is insufficient but do you think it may be a app issue? Do you have any advice? In this case the situation may be a bit simpler as I can deadlock in minutes.

nenadlazovic, 2016-04-04:
Hi,
we also experience locking issue for quite some time. Scenario is similar to @carlonluca case. When running 2 videos at the same time, threadlock happens quite frequently (< 1 min). Not having tech expertise to debug deep, we had to consider changing platform from RPI2 to something else, where drivers dont threadlock. We would like to stay on RPI2 / Broadcom.

Can you please allocate time to work on this issue. Much appreciated.

pelwell, 2016-04-08:
I think you're now seeing the case I (incorrectly) thought @carlonluca was seeing before, which is that the vchiq completion thread (it handles all ilcs RPC traffic) is blocked because ILCS_SERVICE_T::queue is full. As a result, all threads with an outstanding RPC correctly block in ilcs_execute_function_ex.

Try applying this patch and seeing what happens:
```
diff --git a/interface/vmcs_host/vcilcs.c b/interface/vmcs_host/vcilcs.c
index 75ba1f5..a189c57 100644
--- a/interface/vmcs_host/vcilcs.c
+++ b/interface/vmcs_host/vcilcs.c
@@ -180,7 +180,7 @@ ILCS_SERVICE_T *ilcs_init(VCHIQ_STATE_T *state, void **connection, ILCS_CONFIG_T
       goto fail_timer;
 
    // create the queue of incoming messages
-   if(!vchiu_queue_init(&st->queue, 64))
+   if(!vchiu_queue_init(&st->queue, 256))
       goto fail_queue;
 
    // create the bulk receive event
```
I think there are then three possible outcomes:
a) If the problem goes away, then there is just a transient peak of outstanding responses that is >64 but <256.
b) If the problem takes longer to reproduce then the system can't keep up and will always hit a problem like this without some kind of pacing.
c) If it continues to fail in the same way then I've misunderstood the problem.

pelwell, 2016-04-08:
d) There are more than three outcomes.

carlonluca, 2016-04-08:
Hello, I thought the previous patch was supposed to handle this situation already. And I confirm it was very useful: it was clearly fixing the problem for the sample I gave you. I didn't close the issue because I was still able to reproduce with other procedures. This patch again seems very reasonable then: there are many calls at the same time here: flush, GetConfig, multiple emptyThisBuffer etc...
Now I'm running tests with this new patch: I'll let you know. Thanks!

nenadlazovic, 2016-04-11:
@pelwell @carlonluca 
I am testing now and so far it looks very promising. Will run few more tests. will keep you informed.

THANKS

nenadlazovic, 2016-04-12:
Hi @pelwell,
My testing so far (24 hrs+) run shows ALL OK. No issues. It seems very stable. 
I guess many people  will benefit of this improvement.

I would really appreciate if you generate new release with this improvement in it.

If not a big deal, is it possible to expose (maybe as log message) peek vchiu_queue usage, so that one can see if his app is maxing out vchiu_queue.

THANKS

carlonluca, 2016-04-12:
Thanks for the fix @pelwell. My tests confirm @nenadlazovic's findings. No deadlock in more than 30 hours. And I think that the log @nenadlazovic suggested would be useful: determining the actual reason of a deadlock is often difficult without an indication.
Will this get to the repo soon? I know others are waiting to test it.

pelwell, 2016-04-12:
Increasing the limit is easy and can probably go in today. Apart from using slightly more memory it should have no negative consequences, and you've already tested it fairly thoroughly.

Adding some sort of diagnostic output when the limit is reached needs more care. If I post a small patch, as before, would you be happy to test it with a reduced limit?

nenadlazovic, 2016-04-12:
ok. 
pls release increased vchiu_queue (256) asap. 
send us a patch for diagnostic output and I can test it.
thanks

pelwell, 2016-04-12:
Try this patch:
```
diff --git a/interface/vmcs_host/vcilcs.c b/interface/vmcs_host/vcilcs.c
index 2ea7578..73b761e 100644
--- a/interface/vmcs_host/vcilcs.c
+++ b/interface/vmcs_host/vcilcs.c
@@ -386,8 +386,20 @@ static int ilcs_callback(VCHIQ_REASON_T reason, VCHIQ_HEADER_T *header, void *se
 
    case VCHIQ_MESSAGE_AVAILABLE:
 #ifndef _VIDEOCORE
-      if (vchiq_queue_full(&st->queue))
-         VCOS_ALERT("ILCS message queue full!");
+      {
+        static int queue_warn = 0;
+        int queue_len = st->queue.write - st->queue.read;
+        if (!queue_warn)
+           queue_warn = st->queue.size/2;
+        if (queue_len >= queue_warn)
+        {
+           if (queue_len == st->queue.size)
+              VCOS_ALERT("ILCS queue full");
+           else
+              VCOS_ALERT("ILCS queue len = %d", queue_len);
+           queue_warn = queue_warn + (st->queue.size - queue_warn)/2;
+        }
+      }
 #endif
       vchiu_queue_push(&st->queue, header);
       break;
```
You will get a series of messages to stderr as you reach and exceed each warning level, starting at the halfway mark:
```
ILCS queue len = 128
ILCS queue len = 192
ILCS queue len = 224
ILCS queue len = 240
ILCS queue len = 248
ILCS queue len = 252
ILCS queue len = 254
ILCS queue len = 255
ILCS queue full
ILCS queue full
```
The `queue len =` messages only appear once for each threshold, but the full message is repeated to increase the chance that it gets seen in the event of a hang.

nenadlazovic, 2016-04-13:
Hi @pelwell,
i tested and I am getting behavior exactly as expected: at the time when log shows "ILCS queue full", my app hangs (see 2 cases below).

So I am ok with both improvements you made
- vchiu_queue increase to 256
- logging vchiu_queue usage

Whenever you can, pls release firmware with both improvements. THANKS.

**Case 1**
qml: Video, playMedia: mo.id: mediaObject_2_42_56_608_88
qml: Video, stopMedia: mo.id: mediaObject_2_42_53_821_11
ILCS queue len = 32
ILCS queue len = 48
ILCS queue len = 56
ILCS queue len = 60
ILCS queue len = 62
ILCS queue len = 63
ILCS queue full

**Case 2**
qml: Video, playMedia: mo.id: mediaObject_2_43_3_831_7
06:47:41.851 INFO:       Seek to: 00:00:00
06:47:41.868 INFO:       Stop command issued.
06:47:41.871 INFO:       Seek to: 00:00:00
06:47:41.872 INFO:       Stop command issued.
qml: Video, playMedia: mo.id: mediaObject_2_42_56_608_88
06:47:56.901 INFO:       Seek to: 00:00:00
qml: Video, stopMedia: mo.id: mediaObject_2_42_53_821_11
ILCS queue len = 32
ILCS queue len = 48
ILCS queue len = 56
ILCS queue len = 60
ILCS queue len = 62
ILCS queue len = 63
ILCS queue full

nenadlazovic, 2016-04-16:
@pelwell, thanks for adding logging to show usage state of vchiu_queue.
Is there a plan to release this improvement ?
Any rough estimate on when ?

THANKS.

pelwell, 2016-04-16:
It will probably go out this weekend.

popcornmix, 2016-04-17:
Latest rpi-update optionally includes the warnings.
Set env variable ILCS_WARN=1 to enable.

popcornmix, 2016-07-08:
@carlonluca Okay to close this?

carlonluca, 2016-07-08:
Yes, I think you can close this, thanks.

closed 2016-07-08 by popcornmix

atic-atac, 2016-08-09:
I'm getting exactly these symptoms right now, even with the latest firmware.

popcornmix, 2016-08-09:
Many different bugs in the calling code can cause this symptom. Yours is probably not related to this github issue.
If you believe there is a bug in the firmware, then provide a minimal test app that shows the problem (as a new github issue) and we'll try to help.

atic-atac, 2016-08-09:
You mention bugs in the calling code, can you give any hints so I can check ?

dimovnike, 2017-04-14:
Hello, what configs are necessary to activate thew "ILCS queue full" logs in the latest debian jessie? Is setting env var ILCS_WARN=1 enough?

The firmware version i have is:

```
  * firmware as of b51046a2b2bb69771579a549d157205d9982f858

 -- Serge Schneider <serge@raspberrypi.org>  Fri, 03 Mar 2017 14:30:23 +0000

```

popcornmix, 2017-04-14:
Yes, ILCS_WARN should be supported with that firmware

dimovnike, 2017-04-15:
Thanks for the answer. We are having a very similar issue (backtrace attached) but its very hard to reproduce. It just happens once in 2-3 days on a random PI (we are running aboout 30 pi's ). However, I enabled ILCS_WARN but there is no sign of queue being full. When this happens. the application can not be restarted and full system restart is needed (the second backtrace is taken after the restart when the app is stuck). 

Any idea how to debug this? What extra logs I can activate in order to gather more info about it?

[backtrace.zip](https://github.com/raspberrypi/firmware/files/923485/backtrace.zip)