raspberrypi/firmware #450: Device tree and initramfs cannot coexist peacefully

kenrestivo, 2015-07-17:
Following instructions on https://github.com/notro/fbtft/wiki/Bootsplash , initramfs refuses (silently) to load.

adding "devicetree=" to config.txt causes the initramfs to load, but the system is thus without device tree and the drivers for the fbtft thus don't work. This is not a solution since I need device tree AND initramfs to coexist peacefully, and it seems they do not.

This is on a Pi 2, 4.07 kernel and firmware.

(also xposted https://github.com/notro/fbtft/issues/302) not sure who has the power to resolve this issue, suspect it is a bug in the proprietary, closed-source Raspbrry Pi bootloader)

pelwell, 2015-07-21:
The silent failure is because the parser is looking for either "initramfs <filename> <address>" or "initramfs <filename> followkernel", and ignores anything else (error messages are a bit tricky since there is currently no mechanism to bring message from the loader to the attention of the user, but you can retrieve them manually using `sudo vcdbg log msg`). Try appending "followkernel" to see if you get further.

My testing hasn't been very successful so far - it isn't getting as far as the login prompt...

kenrestivo, 2015-07-21:
I am using followkernel.

There's no vcdbg on this distro. I've found vcgencmd in https://github.com/raspberrypi/userland, but not vcdbg.

pelwell, 2015-07-22:
OK, I've succeeded in making it work without any firmware or kernel changes. I'm on the latest rpi-update image, which is 4.0.8-v7+. The steps are:

1) Get yourself a clean Raspbian image.
2) `sudo rpi-update`, which will bring you to 4.0.8-v7+
3) ```
sudo update-initramfs -c -k `uname -r`
```
4)
```
sudo sh -c "echo initramfs initrd.img-`uname -r` followkernel >>/boot/config.txt"
```

5) Add "debug" to /boot/cmdline.txt.
6) Reboot.
7) Log in.
8) dmesg | grep initrd

You should see a message like:
```
[    1.428152] Freeing initrd memory: 3932K (bac19000 - baff0000)
```
Now try the same on your distribution starting with an empty /etc/initramfs-tools/modules, empty /etc/initramfs-tools/conf.d/ and no plymouth additions to cmdline.txt.

If that fails to boot, gather as much info as you can - cmdline.txt, config.txt. what do you see over HDMI etc. - and report back.

kenrestivo, 2015-10-05:
Problem was I was using Debian (not raspian) which mounts the firmware in /boot/firmware not in /boot.

The update-initramfs dumps it in /boot. I added a hook to copy it to /boot/firmware and all is well.

closed 2015-10-05 by kenrestivo