raspberrypi/firmware #451: YUV firmware change.

tvjon, 2015-07-23:
Since updating today, the "hello_video" & derivatives will no longer display any h264 files. The files still play fine in omxplayer

popcornmix, 2015-07-23:
Yes, I have a fix for this - will push later today.

popcornmix, 2015-07-23:
Pushed. Run rpi-update and test please.

tvjon, 2015-07-23:
Well, I can now play *.h264 via, for example, "hello_video", thank you, BUT; your "hello_video_simple" application (no clock or scheduler components) will not play any h264's. BTW, although you posted back that the array subscript error was fixed, I've checked back a few times, & your files are still dated Dec 2014, so maybe your fix didn't make it to Github?

I've just booted from a month old backup, & hello_video_simple.c built plays fine in that "old" firmware.

popcornmix, 2015-07-23:
Could you test this firmware:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_150724.zip

unzip into boot directory of sdcard

tvjon, 2015-07-24:
Initial tests suggest those files have enabled decoding & playback again, thank you, & thank you for the hello_video_ simple file change.

popcornmix, 2015-07-24:
This fix should be in latest rpi-update firmware

popcornmix, 2015-07-28:
There was another update today that could affect this.
Please test and close if okay.

tvjon, 2015-07-28:
Yes, seems fine, thank you.

closed 2015-07-28 by tvjon