raspberrypi/firmware #452: Add possibility to decode JPEGs via mmal in BGRA

LongChair, 2015-07-27:
I have been using the mmal decoder with the provided samples and using qt5.

In some parts of qt5 (ie webengine), chromium requires to use BGRA imagedata which are not available with the decoder.

As the encoding value is defined in https://github.com/raspberrypi/userland/blob/b834074d0c0d9d7e64c133ab14ed691999cee990/interface/mmal/mmal_encodings.h#L79 , i was wondering if that was possible to have the decoder output BGRA rather than the default RGBA data. 

I tried this but it results in an invalid format error when setting up the decoder.

I'm therefore not sure if this is a HW limitation or just the fact that it's not been implemented yet ?

6by9, 2015-07-27:
Lots of formats are defined in mmal_encodings.h, but not all are supported by all components. You can query for the formats potentially supported by a port via MMAL_PARAMETER_SUPPORTED_ENCODINGS.

image_decode supports decoding multiple compressed formats, and only certain image encodings are valid for each compressed format as the component expects the codec to write directly into the output buffer.
At a quick glance, the JPEG hardware block can only write YUV420 (I420) or YUV422 (I422) if the image is encoded not using an RGB colourspace. Adding in a conversion step would be fairly painful and unnecessary for most use cases. You're better off using a separate component to do the conversion - I believe video_splitter should be able to do it for you.

Ruffio, 2016-08-08:
@LongChair Can this issue be closed?

Ruffio, 2017-01-07:
@popcornmix Please consider to close this issue. Last and first time originator wrote was on 27 Jul 2015. @6by9 commented on 27 Jul 2015 and since then there have been no activity...

JamesH65, 2017-12-19:
Closing due to lack of activity. and question answered/issue resolved. Reopen if you feel this issue is still relevant.

closed 2017-12-19 by JamesH65