raspberrypi/firmware #453: TFT screen not working properly

manik9000, 2015-08-04:
Hi, I am using PiTFT 3.5" res touch with RPi2 and OS Image available for this specific screen.
when I boot Pi, it shows all the booting sequence on tft screen but after that it goes black, and I can see only a white '_' symbol in top left corner, that's it.

Any help to resolve this issue?

hkallweit, 2015-08-05:
Most likely you're wrong here in the firmware-related issue section.

I have a PITFT 3.5" Plus and it works perfectly fine on a RPI2.
(Most likely you have the "Plus" version too if you have a RPI2)
With "OS Image available for this specific screen" you mean Adafruit's kernel? Then you should describe your issue in detail in their support forum.

How I operate the PITFT:
- Use hx8357 fbtft display driver from the 4.2 kernel tree.
  (Works fine also with a 4.0 / 4.1 kernel)
- Use Adafruit's DT overlay

manik9000, 2015-08-05:
Thank you for reply
YES it was adafruit Kernal.

But Now I am using fresh raspbian and installed new kernal and all manually.
now screen is working fine only one problem : every time I boot I have to ssh to use 'startx'
[evenif 'raspi-config' settings are for direct GUI boot.]

Ruffio, 2016-06-29:
@manik9000 as I understand your issue has been resolved. Please close this issue if it has been resolved. Thanks.