raspberrypi/firmware #458: vsync callback keeps going after dispmanx display was destroyed

wm4, 2015-08-13:
The vsync callback registered with vc_dispmanx_vsync_callback keeps getting called, even if the associated display is destroyed with vc_dispmanx_display_close.

There is no vsync unregister callback, so I'm assuming the callback should be stopped by the display destruction code.

Is this a bug? If not, what if the display is "lost" (due to mode changes and such), how do you detect that you need to recreate the vsync callback?

wm4, 2015-08-13:
Oh, never mind, it seems it can be stopped with vc_dispmanx_vsync_callback(display, NULL, NULL).

Still could be done automatically, or it could be documented.

JamesH65, 2017-12-21:
Looking at the code, it appears that this has now been fixed - close now calls dispmanx_vsync_callback with NULL.

Closing as fixed.

closed 2017-12-21 by JamesH65