raspberrypi/firmware #459: "overscan_scale=1" in config.txt breaks dispmanX offline compositing

Nixola, 2015-08-13:
I don't think there's much else to say; I didn't have any compositing issue until I added "overscan_scale=1" in my config.txt file. I added it and now I get heavy flickering and glitches whenever I've got more than three layers.

popcornmix, 2015-08-15:
It would help if you described what you are doing. By offline compositing do you mean using dispmanx_offline=1 or some other scheme? The details have a big effect on diagnosing the problem.

Nixola, 2015-08-15:
I mean using dispmanx_offline=1, yeah; if I enable overscan_scale the offline composting doesn't seem to work

popcornmix, 2015-08-15:
Note: overscale_scale=1 is never recommended. Turn off overscan on the display is the best solution. If that is not possible the create the dispmanx overlays at the correct sizes.
Resizing is an expensive operation (especially vertically) and it degrades the image quality.

Can you provide a sample app that demonstrates the problem?

Ruffio, 2016-06-29:
@Nixola is this still an issue or can it be closed?

Nixola, 2016-06-29:
I don't know and I'm afraid I won't be able to test it for a while, as I don't currently have my RPIs

Ruffio, 2016-06-29:
@Nixola, it is just that you haven't responded for 9 months, so you must have found a solution?

Nixola, 2016-06-29:
No, I just started using it headless

closed 2016-06-29 by popcornmix