raspberrypi/firmware #460: HDMI hotplugging and modes

half45, 2015-08-13:
Hello,

When I boot raspberry with connected hdmi, modes are detected correctly:

```
root@raspberrypi:~# tvservice -n
device_name=SAM-SAMSUNG
root@raspberrypi:~# tvservice -m CEA
Group CEA has 16 modes:
           mode 1: 640x480 @ 60Hz 4:3, clock:25MHz progressive 
           mode 2: 720x480 @ 60Hz 4:3, clock:27MHz progressive 
           mode 3: 720x480 @ 60Hz 16:9, clock:27MHz progressive 
           mode 4: 1280x720 @ 60Hz 16:9, clock:74MHz progressive 
           mode 5: 1920x1080 @ 60Hz 16:9, clock:74MHz interlaced 
           mode 7: 720x480 @ 60Hz 16:9, clock:27MHz x2 interlaced 
  (prefer) mode 16: 1920x1080 @ 60Hz 16:9, clock:148MHz progressive 
           mode 17: 720x576 @ 50Hz 4:3, clock:27MHz progressive 
           mode 18: 720x576 @ 50Hz 16:9, clock:27MHz progressive 
           mode 19: 1280x720 @ 50Hz 16:9, clock:74MHz progressive 
           mode 20: 1920x1080 @ 50Hz 16:9, clock:74MHz interlaced 
           mode 22: 720x576 @ 50Hz 16:9, clock:27MHz x2 interlaced 
           mode 31: 1920x1080 @ 50Hz 16:9, clock:148MHz progressive 
           mode 32: 1920x1080 @ 24Hz 16:9, clock:74MHz progressive 
           mode 33: 1920x1080 @ 25Hz 16:9, clock:74MHz progressive 
           mode 34: 1920x1080 @ 30Hz 16:9, clock:74MHz progressive 
root@raspberrypi:~# tvservice -s
state 0x12001a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive
```

If I connect HDMI after boot result is

```
root@raspberrypi:~# tvservice -n
[E] No device present
root@raspberrypi:~# tvservice -m CEA
Group CEA has 2 modes:
           mode 4: 1280x720 @ 60Hz 16:9, clock:74MHz progressive 
           mode 16: 1920x1080 @ 60Hz 16:9, clock:148MHz progressive 
root@raspberrypi:~# tvservice -s
state 0x12001a [HDMI DMT (4) RGB full 4:3], 640x480 @ 60.00Hz, progressive
```

Video present but in low resolution. "tvservice -p" no results
Му config.txt:

```
hdmi_force_hotplug=1
hdmi_drive=2
```
This is expected behavior or bug?

pelwell, 2015-08-13:
Expected behaviour when you have hdmi_force_hotplug=1. See https://github.com/raspberrypi/firmware/issues/376.

half45, 2015-08-13:
Ok, after disabling hdmi_force_hotplug "tvservice -p" works as excepted.
But if I run "tvservice -p" before connecting HDMI, then after connecting HDMI "tvservice -p" wouldn't working before the reboot anyway. Do I need to check that the HDMI connected before calling "tvservice -p"?

```
<HDMI Disconnected>

root@raspberrypi:~# tvservice -p
Powering on HDMI with preferred settings
root@raspberrypi:~# tvservice -s
state 0x40019 [HDMI CEA (1) RGB full 4:3], 640x480 @ 60.00Hz, progressive

<Connect HDMI>

root@raspberrypi:~# tvservice -p
Powering on HDMI with preferred settings
root@raspberrypi:~# tvservice -s
state 0x4001a [HDMI CEA (1) 16:9], 640x480 @ 60.00Hz, progressive
```

josephroffey, 2015-08-20:
Hi all,

Just to say, I have also been experiencing this exact same problem! 
In my environment, pis also need to be able to turned on without a display, but with the possibility of attaching a display later without rebooting...

Does anyone know if there is a way to get tvservice -p to work without knowing which display is going to be attached in advance (when I would just use hdmi_force_hotplug=1)?

popcornmix, 2015-08-20:
When hdmi is powered off, the framebuffer is detached from display.
After ``tvservice -p`` you need to recreate the framebuffer.
``fbset -depth 8 && fbset -depth 16``
should do it. I think changing virtual consoles (ctrl-alt-Fn) should also do it. If running X then ``xrefresh`` will repaint display.

josephroffey, 2015-08-21:
Hi popcornmix - thanks so much for your help!

This didn't seem to work on it's own, but with a combination of 
 tvservice -o //turn HDMI screen off
 tvservice -p //power on HDMI screen to preferred resolution
 a regex to extract the details from tvservice -s and a call to fbset before to recreate the buffer and after to set it correctly, I do seem to have finally got the resolution changes working!!

Thanks all!

popcornmix, 2015-10-14:
@half45 is this still an issue?

half45, 2015-10-18:
Issue solved. Thanks.

closed 2015-10-18 by half45