raspberrypi/firmware #462: Passing OMX_IndexConfigVideoIntraVOPRefresh to OMX_GetConfig: Unsupported index

minfrin, 2015-08-15:
While debugging gst-omx, the attempt to set the OMX_IndexConfigVideoIntraVOPRefresh fails with "Unsupported index".

Turns out that an attempt to get (instead of set) the same index also fails with the same error:

      err = OMX_GetConfig (self->enc->handle, OMX_IndexConfigVideoIntraVOPRefresh, &config);
      if (err != OMX_ErrorNone) {
        GST_ERROR_OBJECT (self, "Failed to force a keyframe at index %d (Could not get config): %s (0x%08x)",
            (int)port->index, gst_omx_error_to_string (err), err);
      }

The source for OMX_GetConfig doesn't seem to be available, is it possible to confirm whether OMX_IndexConfigVideoIntraVOPRefresh is supported by the firmware on the RPi?

Is there any documentation for this?

6by9, 2015-08-15:
Docs for all the IL components are in https://github.com/raspberrypi/firmware/blob/master/documentation/ilcomponents/ including the indices that should be supported and on what ports.
No, IndexConfigVideoIntraVOPRefresh is not supported.

There is support for indices OMX_IndexParamVideoIntraRefresh and OMX_IndexConfigBrcmVideoIntraRefresh. I haven't looked at what specifically OMX_IndexConfigVideoIntraVOPRefresh is trying to do, but from the error message, I'd guess OMX_IndexConfigBrcmVideoRequestIFrame may be related.

minfrin, 2015-08-16:
What the gstreamer code wants to do is force a keyframe while encoding, which call would achieve that?

F5OEO, 2015-08-16:
Could be equivalent to OMX_IndexConfigBrcmVideoRequestIFrame.

6by9, 2015-08-16:
> OMX_IndexConfigBrcmVideoRequestIFrame:
Request an I frame.  If set to OMX_TRUE, the next job submitted 
will be flagged to be converted as an I-frame. This automatically resets
itself. This works with proprietary tunnelling and OMX buffers.

NB This is the next job submitted to the codec. There is an output FIFO within the component, so if you have not been draining the pipe fast enough, there could be a backlog of frames in there before your new I frame appears.

sdroege, 2015-08-22:
Why is the RPi using its own non-standard API for this instead of OMX_IndexConfigVideoIntraVOPRefresh?

6by9, 2015-08-24:
> Why is the RPi using its own non-standard API for this instead of OMX_IndexConfigVideoIntraVOPRefresh?

Probably because it was implemented before the IL spec supported it (it's certainly not in the 1.0 spec, and I'm not going to hunt down when it was added) and no one noticed that it had been officially adopted into the spec.
It is trivial to add as an alternative index, so I'll add it to the list of things to do.

6by9, 2016-02-04:
OMX_IndexConfigVideoIntraVOPRefresh should now be supported. It'll do exactly the same as OMX_IndexConfigBrcmVideoRequestIFrame, though it takes a different structure type.

(Just realised I forgot to update the docs for the list of supported indices though. Oh well)

JamesH65, 2017-12-21:
@6by9  OK to close?

6by9, 2017-12-21:
Yes, done as far as I'm concerned, and OP has never come back on it.

closed 2017-12-21 by JamesH65

tmm1, 2019-08-17:
I'm trying to add support for this to ffmpeg. Is my understanding correct that to force a key-frame, before the call to `OMX_EmptyThisBuffer(handle, buffer);` I need to add a `OMX_SetConfig(handle, OMX_IndexConfigBrcmVideoRequestIFrame, &config)` (with an appropriately initialized config)?