raspberrypi/firmware #464: HDMI hotplug incorrectly detected as faulty

net147, 2015-08-18:
I have an TV connected to a Raspberry Pi 2 Model B using HDMI cable.

If the TV is powered off and I power on the Raspberry Pi, the TV turns on automatically and switches input to the Raspberry Pi.
"tvservice -s" shows "state 0x12001a [HDMI DMT (39) RGB full 16:9], 1360x768 @ 60.00Hz, progressive"
"vcgencmd get_config int" shows "hdmi_force_hotplug=1"

As a result, the state does not change when I unplug the HDMI cable or if I plug it back in.
If I power on the Raspberry Pi while the TV is connected, "vcgencmd get_config int" does not show "hdmi_force_hotplug=1" and the state changes as expected when I unplug the HDMI cable or plug it back in.

So it seems the Raspberry Pi is reading the EDID correctly but thinks hotplug is faulty. Is there a way I can disable composite and force HDMI hotplugging so that I am always able to detect HDMI being active or not when the connected devices always assert hotplug properly?

popcornmix, 2015-08-18:
If hdmi_force_hotplug=1 and you haven't set it in config.txt then it means hotplug was not asserted *but* it was possible to read the EDID of the TV when we booted.
That usually means a faulty HDMI cable (with hotplug not connected) or a badly behaving TV.

hotplug is really only supported on boot to detect if display output should default to composite or hdmi output. In general dynamic unplugging and replugging won't work as new hdmi display won't necessarily support the same hdmi modes and framebuffer will no longer be suitably sized. 

Can you explain exactly why you need to both boot without display powered and to also detect hotplug state?

net147, 2015-08-18:
This has occurred on 2 TVs I have tested. I have an application that is streaming live data from network and displaying it on the TV. The application uses the HDMI state to determine if the TV is powered on or not. If the TV is powered off, it stops streaming. If the TV is powered back on, it starts streaming again.

Most of the time the TV is off and I would like to be able to flash a new image and reboot the Raspberry Pi without having to turn the TV on first - especially if I am doing so remotely.

net147, 2015-08-19:
I am using rpi-4.1.y branch of raspberrypi/linux with devicetree enabled.
Is there a way I can configure the kernel so I can get interrupts for GPIO 46 (HDMI hotplug)?

The following example program doesn't detect interrupts for GPIO 46 when I unplug/plug the HDMI cable:
https://developer.ridgerun.com/wiki/index.php/Gpio-int-test.c

popcornmix, 2015-09-29:
Are you seeing the value change?
```
sudo su
cd /sys/class/gpio/
echo 46 > export
cat gpio46/value 
(unplug/plug hdmi)
cat gpio46/value 
```
I see 0 when connected and 1 when unplugged.

net147, 2015-09-29:
@popcornmix The value does change but I don't get interrupts for the changes.

popcornmix, 2015-09-29:
Have you tried ``tvservice -S``?

net147, 2015-09-29:
tvservice -M shows the changes in realtime but only if the HDMI TV is powered on when the Raspberry Pi is booted. I would like to be able to get interrupts for gpio46 as an alternative that will work regardless of whether the TV is powered on or not when the Raspberry Pi is booted.

tvservice -s only shows proper state when TV is powered on when the Raspberry Pi is booted.

net147, 2016-03-26:
Any update on this?

TheSin-, 2016-05-12:
the interrupt should would be nice, so that something like gpio-watch would work on it.  I wrote something to monitor value for a change, but that isn't as nice as an interrupt would be to tell me to check the value.  There is a chance to miss it if I set the loop too long and is it's short then CPU gets consumed.

Just wondering if this is something that is possible to fix?

TheSin-, 2016-05-16:
can you tell me where in the source the code to change the gpio 46 value is?  I'd like to see if I can find a fix and make a PR for it, but i have no idea where to start looking for it.

TheSin-, 2016-06-06:
@popcornmix @pelwell could you show me where in the rpi/linux repo that I can find this?  I'd like to see if I can figure out why there is no interrupt on value change and submit a PR if I can figure it out.  I'd LOVE to get this fixed and working properly.

net147, 2016-12-05:
@popcornmix Note that booting with HDMI disconnected on Raspberry Pi 3 and checking /sys/class/gpio46/value no longer works in the latest version of the firmware when plugging/unplugging HDMI. Can you please check this?

pelwell, 2016-12-05:
On a Pi3, hotplug detect is done via the GPIO expander (not directly addressable by the ARM) - GPIO 46 is used for I2C. There is a virtual GPIO driver on the ARM that uses a combination of mailbox messages and shared memory to access the GPIO expander, but at the moment that is output-only and used for the activity LED.

net147, 2016-12-05:
Can we have an option in the firmware to keep hotplug detect and events working even if it is not asserted around the time the EDID is read but the EDID is detected successfully? It has been over a year with no progress and I would like to stop having to do workarounds in software...

popcornmix, 2016-12-05:
@net147 out of interest what is the make/model of the TVs that have this behaviour?

net147, 2016-12-05:
@popcornmix 
- Teac LEDV29G72HD (CEC supported)
- Soniq E23Z13A-AU Rev AB (CEC not supported)

AndrewDJohnson, 2017-08-12:
You have to make sure your config.txt file has this line:

hdmi_ignore_cec_init=0 

Sometimes, it's set to 1 and so the TV service doesn't detect anything...

JamesH65, 2017-12-21:
@popcornmix Is this fixed with the hdmi_ignore_cec_init option in config.txt?

closed 2018-10-29 by JamesH65