raspberrypi/firmware #466: Framerate jittering when using OpenGL for hours

dividuum, 2015-08-24:
Hi.

While running my opengl based program [info-beamer](https://info-beamer.com/pi) I noticed jittering after a while: It seems that random frames just take longer to render resulting in stuttering when displaying animated content. I think I'm able to reproduce this problem using one of the provided examples in hello_pi. See [my changes](https://gist.github.com/dividuum/d71c4b42b48594799b5e/revisions) which just modify the hello_triange example code by adding frame time measurements. Just replace the existing triangle.c with my code and compile it.

It seems like some timer is slowly drifting off. When I start the program I get the expected framerate of 60 frames per second. Each frame takes the expected 0.0166 seconds to render. After an hour it seems wandering off slowly:

    0.0170898438
    0.0166015625
    0.0166015625
    0.0166015625
    0.0166015625
    0.0166015625
    0.0166015625
    0.0166015625
    0.0166015625
    0.0170898438

After 6 hours it looks like this:

    0.0156250000
    0.0175781250
    0.0156250000
    0.0175781250
    0.0156250000
    0.0175781250
    0.0175781250
    0.0156250000
    0.0175781250

After 9 hours:

    0.0195312500
    0.0156250000
    0.0156250000
    0.0195312500
    0.0156250000
    0.0156250000
    0.0156250000
    0.0195312500
    0.0156250000
    0.0156250000
    0.0156250000

After 23 hours:

    0.0156250000
    0.0156250000
    0.0156250000
    0.0156250000
    0.0156250000
    0.0156250000
    0.0234375000
    0.0156250000
    0.0156250000
    0.0156250000
    0.0156250000

The odd thing is, that it still results in almost exactly 60 frames per second. But each frame takes a different time to render. My guess is now that this results in visible frame skips once a single frame takes more than 2 × 0.0166666 seconds. Restarting the program results in perfect 0.016666 second frames again.

Any idea what the reason might be?

dividuum, 2015-08-24:
Forgot to mention this. It seems to happen regardless of which firmware/kernel/gl driver version I'm using:

    Aug 20 2015 17:44:52 
    Copyright (c) 2012 Broadcom
    version 3f87086448257aa66881de858a0b4b0e6daa8719 (clean) (release)

    Linux raspberrypi 4.1.6-v7+ #810 SMP PREEMPT Tue Aug 18 15:32:12 BST 2015 armv7l GNU/Linux

and 

    Apr 21 2015 14:42:19 
    Copyright (c) 2012 Broadcom
    version 2d5ad04b63af4233440c3f7c8587108223201102 (clean) (release)

    Linux info-beamer-2840694443 3.18.11+ #781 PREEMPT Tue Apr 21 18:02:18 BST 2015 armv6l GNU/Linux

result in the same behaviour.

popcornmix, 2015-08-25:
At a guess I'd say it's the precision of a float used in time_get.
Once tv_sec gets large, you lose precision in the sub-second component.

Use long long for the timer calculations. Print the difference integer nanoseconds and report back.

dividuum, 2015-08-25:
I really hope that's not the case, since the problem is also visible. But I'll check that and report back.

dividuum, 2015-08-25:
And you are correct. Well that is embarrassing. Thanks.

closed 2015-08-25 by dividuum