raspberrypi/firmware #470: H264 encoder component drops I-frame requests (expected behaviour?)

malc0, 2015-09-01:
My application uses MMAL_PARAMETER_VIDEO_REQUEST_I_FRAME to request output file splits.  If the encoder's output callback is being slow (contention for file I/O) these requests can get dropped and I don't get the expected file split.  Allocating a large pool of encoder output buffers postpones the problem, but is the request dropping a) expected behaviour in the first place, and b) detectable as having happened, or avoidable, by other means?

To illustrate the problem I've written a small patch against RaspiVid.c ( https://gist.github.com/malc0/b325344899b6015d6f06 ), which should be run with `-ih -o foo.h264 -t 50000 -n`.  It avoids tunnelling to get a pts timestamp on the encoder input for when the split should appear, slows down encoder_buffer_callback(), and prints the pts of the buffers following the  MMAL_BUFFER_HEADER_FLAG_CONFIG buffers on the output side.

A typical failing output begins (note the pts = 5762889 request does not appear in the output):
```
requesting split just before pts 0
..
buffer after config header has pts 33312
.................
requesting split just before pts 899412
.............
buffer after config header has pts 899412
......
requesting split just before pts 1765510
...................
requesting split just before pts 2664921
........
buffer after config header has pts 1765510
...........
requesting split just before pts 3730889
...................
requesting split just before pts 4663610
..
buffer after config header has pts 2664921
.................
buffer after config header has pts 3730889

requesting split just before pts 5762889
.......
buffer after config header has pts 4663610
............
requesting split just before pts 8128005
...................
requesting split just before pts 9560400
.........
buffer after config header has pts 8128005
```

Using `encoder_output->buffer_num = 500` an acceptable output begins:
```
requesting split just before pts 0
..
buffer after config header has pts 33194
.................
requesting split just before pts 1294540
..
buffer after config header has pts 1294540
.................
requesting split just before pts 2887821
..
buffer after config header has pts 2887821
.................
requesting split just before pts 4414713
..
buffer after config header has pts 4414713
.................
requesting split just before pts 6074379
.
buffer after config header has pts 6074379
..................
requesting split just before pts 7767239
..
buffer after config header has pts 7767239
.................
requesting split just before pts 9493292
.
buffer after config header has pts 9493292
```

Alternatively, I haven't seen the problem when using MMAL_ENCODING_OPAQUE, but unfortunately this is not acceptable for the application:
```
requesting split just before pts 0
..
buffer after config header has pts 33194
.................
requesting split just before pts 663867
...................
requesting split just before pts 1327733
..
buffer after config header has pts 663867
.................
requesting split just before pts 2124373
.........
buffer after config header has pts 1327733
..........
requesting split just before pts 3518492
........
buffer after config header has pts 2124373
...........
requesting split just before pts 4945807
..........
buffer after config header has pts 3518492
.........
requesting split just before pts 6505892
........
buffer after config header has pts 4945807
...........
requesting split just before pts 7999592
.........
buffer after config header has pts 6505892
..........
requesting split just before pts 9426905
.........
buffer after config header has pts 7999592
..........
requesting split just before pts 10821025
..........
buffer after config header has pts 9426905
```

popcornmix, 2015-09-01:
ping @6by9

6by9, 2015-09-01:
At a guess (sorry, in the office so no time to look in detail), when using I420 the video_encode component has to format convert the I420 into a pool of images of an internal format (opaque buffers typically are already in that format). Hopefully the allocation of the internal buffers is of the same number as input images (or even +1 for a reference image), but may not be (a hardcoded value of 3 wouldn't surprise me). 
I couldn't swear to the behaviour if we haven't got an internal buffer to convert into - I'd hope it would just hold the I420 buffer until it did have one.

The requesting of an I frame just gets passed in to the codec code for the next requested job (frame) to produce an I-frame. Where it goes within the codec I don't know off the top of my head.

Do you get any form of frame out for the dropped PTS? ie is it solely the codec headers and I-frame you're missing, or is the whole frame gone? Rate control will drop frames totally if the resulting size significantly exceeds the target bitrate. Again I'd hope that it would remember that an extra I-frame had been requested and attempt it for the next frame, but don't know. IIRC there is a warning message dropped in vcdbg log msg when it happens.

malc0, 2015-09-01:
The PTS emerges as a normal frame; it's the headers and I-frame that are missing.  All `vcdbg log msg` reports is:
```
1966878.613: mmal: mmal_vll_load: could not load VLL 'videnc.vll': 
1992507.817: gpioman: gpioman_get_pin_num: pin CAMERA_LED not defined
```
which seems of limited relevance.

Given that the problem is triggered purely by the encoder output pipeline 'backing-up', which then must have an affect on the ability to recycle buffers on the encoder's input, I wondered if there was some possibility of a silent allocation failure when attempting to pass the request over to the VC side, but not knowing what's going on that's probably just unhelpful speculation :)

6by9, 2015-09-01:
Nothing obvious from a quick squint at the code, but there does half appear to be a way for the flag to get cleared before it is requested under allocation failure conditions.

What does appear to be present is that setting MMAL_BUFFER_HEADER_FLAG_DISCONTINUITY on an input buffer should also request an I-frame. That might be a better approach for you, as then it shouldn't be possible for the request to get lost.

malc0, 2015-09-02:
Using MMAL_BUFFER_HEADER_FLAG_DISCONTINUITY certainly fixes the test-case, thanks!  So far it is working as expected in my application too, but it can take some days to hit the problem under normal conditions.

I'll leave the bug open for the original MMAL_PARAMETER_VIDEO_REQUEST_I_FRAME problem though, and am happy to test anything in future if it gets looked at.

6by9, 2015-09-02:
OK, I'll have a read through of the code again when I get a chance.
There was a block which read along the lines of 
```
bool i_frame = false
if(thing1) {
   if(REQUEST_I_FRAME flag)
   {
       iframe = true;
       clear REQUEST_I_FRAME flag
    }
}
if (thing2)
{
   submit_job_to_codec(i_frame)
}
```
If thing1 is true but thing2 isn't, then the flag would be cleared but the job never submitted. I didn't work through the logic of whether that could happen in reality. Another to add to my list of jobs.

6by9, 2015-09-02:
It looks like the internal image pool is fixed at 3 images, and the input buffer is released back to the client as soon as the conversion into one of these images has taken place, not when the codec has encoded it. That in itself is fine, but does mean that we can end running out of images in the pool.

If we get that failure when an I-frame has been requested, then the buffer is put back, job cancelled, but as suspected the I-frame request flag doesn't get set again. 
Looking at it, I see no reason why I can't move the check for the an I-frame into the thing2 clause which is guaranteed to be submitted to the codec. It's very low risk, so I'll sort out a patch for @popcornmix and let him push it into the main release (unless he objects to that plan).

6by9, 2015-09-02:
Ah, it's due to locking. thing1 clause is done whilst holding a semaphore, and thing2 clause is done after it has been released. Slightly different solution required, but at least I can see the logic of why it was split.

6by9, 2015-09-16:
I sent my firmware change to popcornmix and it appears to have been merged to the tree on 4th Sept. So whilst it wasn't logged in the firmware release commit text, this should be fixed with the latest rpi-update releases. 
@malc0 Could you retest with your original test case, please?

malc0, 2015-09-16:
@6by9 That seems to have fixed it, thanks (both for this and the MMAL_BUFFER_HEADER_FLAG_DISCONTINUITY trick)!

closed 2015-09-16 by malc0