raspberrypi/firmware #472: Using deinterlacer with egl_render OpenMAX component

carlonluca, 2015-09-06:
I tried to enable deinterlacer while using the egl_render OpenMAX component. What I found out is that the resulting frame rate of the component seems low. I'm trying to find out if there is a mistake in my code, but I also tried to use the vcdbg hist command, which seems to track the GPU usage if I'm not mistaken. The results are reported here:

These were taken with my code running:
deinterlacer on: http://pastebin.com/PpMJnYc8
deinterlacer off: http://pastebin.com/kGw9mkVn

This was taken with omxplayer running with the same video file (interlacer seems to be enabled):
deinterlacer on: http://pastebin.com/cprs2KJs

What I notice is that in the omxplayer case ILVRender is 0.71% and ILImageFX is 28.8% (I suppose ILImageFX is the component executing the deinterlacing algorithm itself) while in my code LEGLRender is taking 86.1% while ILImageFX is still around 24.9%. LEGLRender is saturating the core. This may be my fault, but during a regular playback (same video file but with deinterlacing off) ILEGLRender is around 23.0%, which leaves some idle time for the core. Any idea what may be happening here? Can someone else reproduce this behaviour with another implementation?

6by9, 2015-09-06:
Image_fx is the component doing deinterlacing - yes.
EGLRender has to do a format conversion to an image format that EGL can consume - IIRC that also includes a conversion from YUV to RGB, so is moderately compute intensive. VideoRender just passes frames to the hardware composition engine, so has minimal actual work to do.

Deinterlacing doubles the framerate, so EGLRender is now seeing twice the load.
I couldn't say off the top of my head how EGLRender deals with frame drops, but it sounds like you are hitting the compute limit and frames are being dropped.
(Surprised vcdbg hist is in there - I didn't think there were sufficient symbols exported for that. Makes me want to go and look again, as the updated version could also export to HTML!)

carlonluca, 2015-09-06:
Ok so I guess this is expected, isn't it? It means it is impossible to deinterlace 1080p with this OpenMAX component.
Just asking: why is the performance of the two render components so different? I suppose both need to convert the yuv format of the h264 to some rgb. Why is the performance so different? No way to improve the EGL component? I suppose improving it may also improve OpenGL performance while rendering video.

6by9, 2015-09-06:
Nope, the hardware composition engine will happily accept YUV or RGB, and renders it to either YUV or RGB targets in realtime (up to a certain scene complexity). No format conversion on the GPU processor (aka VPU) is required when using video_render :-)

There are 2 VPU cores, however there is no automatic process migration between cores.
A quick glance at the firmware source implies that we already have image_fx and egl_render on different cores, and your task histories appear to confirm that. If egl_render is already consuming >80%, then there's not much left. 
Memory says that on other platforms we cheated and actually subsampled the image by a factor of 2 in each direction when converting to a texture, but that was mainly as we knew the end display devices. The same isn't true on Pi. There's also a reason why under Android video is only rendered via EGL whilst applying a transition before returning to native rendering - it saves a huge amount of power.

@popcornmix may have some ideas on improving egl_render, but I think you may just be pushing the limits. 
Which deinterlace algo are you using? It appears that he has been tweaking the advanced algorithm to use the QPUs for the processing - that may be counter productive for you as you are trying to render the scene via EGL and hence the QPUs, but may help. 

I think that the main issue is that deinterlace will double the framerate, hence you've double the work involved, and that just pushes the limits a bit too far. It does seem a little odd though that doubling the framerate pushes egl_render up from 23% to 86%. It may be contention on other system hardware, but that's a pure guess.

popcornmix, 2015-09-07:
sdram bandwidth becomes a big factor when working with 32bpp 1080p60 frames, so it's not that surprising that egl_render gets much slower when the system is so much busier - I suspect the egl_render format conversion code is seeing a lot of sdram stalls due to the other core saturating the sdram bus.

I believe that 1080i + double rate deinterlace + convert to texture and rendering with 3D is too much for the pi.

Choosing the half rate deinterlace, or rendering the video on a separate YUV layer (like kodi does) are options that may work.

carlonluca, 2015-09-07:
Thanks for your answers guys. @popcornmix what is this "half rate deinterlace"? Is it a mode that can be set with omx API?

popcornmix, 2015-09-07:
Yes, one of the parameters to image_fx:
https://github.com/xbmc/xbmc/blob/master/xbmc/cores/omxplayer/OMXVideo.cpp#L281

carlonluca, 2015-09-20:
I'm doing some tests on this topic and I noticed this commit in omxplayer: https://github.com/popcornmix/omxplayer/commit/d99bd86c9fd80ea01f98761e130d7ac9a2bb7588. Can it have a role? I tried quickly and seemed to improve a bit. Is there documentation about that param? Thanks.

popcornmix, 2015-09-24:
The final "1"?
Originally the advanced deinterlace used the VPU processors. It's an expensive algorithm, and we only supported up to DVD resolution.
Recently the algorithm has been ported to the QPUs (part of the 3D hardware used for shaders) which has a bit more performance, and can (just about) handle 1080i.
The final "1" parameter requests using the QPUs rather than the VPUs.

Will it help your case? Perhaps a little.

carlonluca, 2015-09-28:
I wrote some notes about this topic here: http://tinyurl.com/q2c8blp. I also provided a demo of the final result, which seems pretty good. I suppose you can close the issue. Thank you for your help!

popcornmix, 2015-09-28:
Looks good.

closed 2015-09-28 by popcornmix