raspberrypi/firmware #473: error 110 - transferring data - noobs cannot install raspbian on the TS32GUSDU1E Transcend 32GB Premium 300x card

saraf, 2015-09-16:
[This card](http://www.amazon.com/Transcend-MicroSDHC-Class10-Adapter-TS32GUSDU1E/dp/B00CES44EO) does not work. There is a stream of mmcblk0 error 110 transferring data errors on the console.

And now I see [this list of cards that are verified to work and not work](http://elinux.org/RPi_SD_cards)

I am surprised to see a large number of cards that do not work!

Can this be fixed?

popcornmix, 2015-09-16:
It is quite likely is fixed in latest kernel, but you will have to wait for the new NOOBS image which should be out in next couple of weeks.
There have been fixes to the current sdcard driver (mmc) as well as an optional new driver (sdhost). 
We are not aware of any sdcards that are unusable now.

saraf, 2015-09-17:
Thanks for the good information!

What is the highest speed microSD card that can be supported by a Raspberry Pi 2 B?
What is the limiting factor? 

What card is recommended for the highest performance? 
The above website further shows that the highest actual write speed observed with the raspberry pi is around 25 MB/s - [for a card that has 40MB/s write - 60MB/s to 90MB/s read speed as advertised](http://www.amazon.com/SanDisk-Extreme-microSDHC-Adapter-SDSQXNE-016G-GN6MA/dp/B013CP5F90/ref=sr_1_1?ie=UTF8&qid=1442464815&sr=8-1&keywords=SDSDQXN-016G-G46A)

The max read speed also seems to have a limit at 25 MB/s.

popcornmix, 2015-09-17:
Yes, max official clock speed for sdcard on Pi is 50MHz with a 4-bit bus, so 25MB/s is the theoretical limit for reads or writes.
There are ways of [overclocking the sdcard interface](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=6201&p=761030#p761030), but no guarantees that won't cause problems.

For most usage on a Pi it is the 4K random access read/write speed that is most important, and that is always well below the 25MB/s limit. See [the sdcard benchmark thread](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=4076) for details.

popcornmix, 2015-10-14:
Have you tried with the latest NOOBS image?
If that doesn't work I'd be interested if latest raspbian image works okay.

saraf, 2015-10-15:
Tried the latest NOOBs just now. Still getting the same stream of errors.

However, writing the latest Raspbian Jessie image directly works and does not give any errors - even when I filled up the entire root partition with file containing zeros using /dev/zero.

popcornmix, 2015-10-15:
I think NOOBS is still on 3.18 kernel, so doesn't have latest sdcard fixes from raspbian (4.1) kernel.
Assuming you are happy with raspberry kernel, then this should be reported here:
https://github.com/raspberrypi/noobs/issues/

Ruffio, 2016-06-30:
@saraf do you still experience this issue, or can it be closed?

saraf, 2016-06-30:
Ok. I moved to another micro sd card and a newer version of raspbian.
Thanks for your help on this! We can close this.

Ruffio, 2016-06-30:
So you will close it? Thanks.

closed 2016-06-30 by saraf