raspberrypi/firmware #476: Unable to use OpenGL ES 2 extensions

Xfennec, 2015-09-24:
Because of this strange thing at the beginning of `gl2ext.h`, I can't find any clean way to link to extensions functions like `glBindVertexArrayOES()`, for instance:
```
/* We want this */
#ifndef GL_GLEXT_PROTOTYPES
#define GL_GLEXT_PROTOTYPES
#endif
```

Since prototypes are already declared, the only option left is to link to some library at compile time, but `readelf -Ws /opt/vc/lib/* | grep glBindVertexArray` gives me nothing.

I would like to be able to do this:
```
#include <GLES2/gl2ext.h>
PFNGLBINDVERTEXARRAYOESPROC glBindVertexArrayOES;
glBindVertexArrayOES = (PFNGLBINDVERTEXARRAYOESPROC)eglGetProcAddress("glBindVertexArrayOES");
```
… but obviously, the already defined prototype conflicts with my own declaration.

I'm using some other GLES2 implementations on some other boards, and none of them force the definition of `GL_GLEXT_PROTOTYPES`. Same story with Khronos specification `gl2ext.h` file.

Shouldn't this "forced" definition be removed ? If not, what's the correct / recommended way to link to extensions ?

popcornmix, 2015-09-24:
We don't support glBindVertexArrayOES (it is not part of the base OpenGLES spec).

Xfennec, 2015-09-24:
Sorry for the noise, I just realized that GL_OES_vertex_array_object is not reported as a supported extension.

closed 2015-09-24 by Xfennec