raspberrypi/firmware #477: rpi-update inconsistency

MrEngman, 2015-09-25:
Current commits for Hexxeh/rpi-firmware 7ee23ac14, 23 Sep, and raspberrypi-firmware 960832a6c, 23 Sep. 

raspberrypi-firmware git_hash shows kernel commit 59e76bb7e and uname_string is 4.1.7+ #815 e.g.uname -a output is 
```Linux raspberrypi 4.1.7+ #815 SMP PREEMPT Thu Sep 17 17:59:24 BST 2015 armv6l``` 

however, after running rpi-update on Pi B+ it loads kernel 4.1.7+ #817. e.g. uname -a output is 
```Linux raspberrypi 4.1.7+ #817 PREEMPT Sat Sep 19 15:25:36 BST 2015 armv6l```

Reviewing the kernel github there is no kernel update for Sep 19.

What's going on? Some update in the raspberrypi-linux github missing? Makes updating the wifi drivers I compile rather impossible as I am unable to select the kernel revision for 4.1.7+ #817

popcornmix, 2015-09-29:
There's been a kernel update. Does it look okay now?

popcornmix, 2015-10-14:
I'll assume it's okay now. Reopen if you believe there is still a problem.

closed 2015-10-14 by popcornmix