raspberrypi/firmware #479: usb stops working with firmware from 23rd May

julianscheel, 2015-10-02:
I'm having trouble to run any firmware from 23rd May (https://github.com/raspberrypi/firmware/commit/f6ce0b0c36c6f7bf9530d64d0a240c6d9a270c46) or later on my systems. No USB peripheral including the smsc USB hub+ethernet are detected at all.
My buest guess is that the change `firmware: Replacing board rev functions with board_info library` is causing some board specific init to be non-working because of some missing bit in my setup.
So far I tested with kernels:

- 3.18.16 - no device tree, my custom defconfig
- 4.1.9 - no device tree, my custom defconfig
- 4.1.9 - no device tree, bcmrpi_defconfig
- 4.1.9 - bcm2708-rpi-b.dtb, bcmrpi_defconfig

System details and USB related dmesg output with todays firmware and kernel:
```
# vcgencmd version
Sep 29 2015 13:06:19 
Copyright (c) 2012 Broadcom
version 3bbf5122d0c4ff421c92d72871dbc8fa104a976d (clean) (release)

# uname -a
Linux nvr-101 4.1.9 #2 PREEMPT Fri Oct 2 13:07:10 CEST 2015 armv6l GNU/Linux

# dmesg | grep -E "(usb|otg)"
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708.boardrev=0xe bcm2708.serial=0x3628f334 smsc95xx.macaddr=B8:27:EB:28:F3:34 bcm2708_fb.fbswap=1 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 elevator=deadline rootwait
[    0.626604] usbcore: registered new interface driver usbfs
[    0.632210] usbcore: registered new interface driver hub
[    0.637757] usbcore: registered new device driver usb
[    4.232148] usbcore: registered new interface driver smsc95xx
[    4.238132] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
[    4.681320] WARN::dwc_otg_hcd_init:1047: FIQ DMA bounce buffers: virt = 0xd2414000 dma = 0x52414000 len=9024
[    4.707961] dwc_otg: Microframe scheduler enabled
[    4.726475] dwc_otg 20980000.usb: DWC OTG Controller
[    4.731561] dwc_otg 20980000.usb: new USB bus registered, assigned bus number 1
[    4.738925] dwc_otg 20980000.usb: irq 32, io mem 0x00000000
[    4.752164] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    4.758962] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    4.766240] usb usb1: Product: DWC OTG Controller
[    4.770993] usb usb1: Manufacturer: Linux 4.1.9 dwc_otg_hcd
[    4.776568] usb usb1: SerialNumber: 20980000.usb
[    4.790881] dwc_otg: FIQ enabled
[    4.790905] dwc_otg: NAK holdoff enabled
[    4.790918] dwc_otg: FIQ split-transaction FSM enabled
[    4.791597] usbcore: registered new interface driver usb-storage
[    4.899090] usbcore: registered new interface driver usbhid
[    4.906850] usbhid: USB HID core driver
```

Full dmesg can be found here: https://gist.github.com/julianscheel/cd2816fb466e7bcef0f3
config.txt is this: https://gist.github.com/julianscheel/dd30f524e795925164b8

Any thoughts on what might go wrong are welcome...

popcornmix, 2015-10-02:
What version of Pi are you running on? What does /proc/cpuinfo say?
What does ``sudo vcdbg log msg`` report.
Are you just testing with kernels you have built? Does the official kernel behave the same?
Why are you specifying ``device_tree=bcm2708-rpi-b.dtb``?
That should be automatically chosen, and if that isn't the board you are using, then that would cause issues.
ping @pelwell

pelwell, 2015-10-02:
Also, what is the output of `sudo vcdbg log msg`, and do you have a dt_blob.bin?

pelwell, 2015-10-02:
Your dmesg log shows that you have a module/kernel version mismatch:
```
bcm2835_gpiomem: disagrees about version of symbol module_layout
[    7.263166] led_class: disagrees about version of symbol module_layout
[    8.030402] rc_core: disagrees about version of symbol module_layout
[    8.194329] fat: disagrees about version of symbol module_layout
[    8.234116] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[    8.383212] snd: disagrees about version of symbol module_layout
[    8.402542] snd: disagrees about version of symbol module_layout
[    9.609073] ledtrig_netdev: disagrees about version of symbol module_layout
[   10.021939] bcm2835_gpiomem: disagrees about version of symbol module_layout
[   10.293783] led_class: disagrees about version of symbol module_layout
[   11.260002] bcm2708_wdog: disagrees about version of symbol module_layout
```
Generally speaking that doesn't make for a happy system.

pelwell, 2015-10-02:
dmesg shows `bcm2708.boardrev=0xe`, so it should be a rev 1.2 Model B.

pelwell, 2015-10-02:
FYI your `device_tree=bcm2708_rpi_b.dtb` line is redundant, since that is the default for your Pi. This is harmless, but it does make it more difficult to boot your card in another Pi.

julianscheel, 2015-10-02:
@pelwell Thanks for the hints. The key was the dt-blob, which I did not update as I had some custom changes in there and did not realize that it has undergone major upstream changes. USB is back alive now :)

closed 2015-10-02 by julianscheel

pelwell, 2015-10-02:
I'm happy to hear that your immediate problem is solved. Are you sure that a custom dt-blob is the best solution to your problem? Would a DT overlay not work for you?

julianscheel, 2015-10-02:
I introduced the dt-blob.bin to reconfigure some gpio pull settings for different base-boards. At that time the kernel was not dt aware, so the GPU dt seemed the best choice as I had not to hardcode stuff in the kernel. If the kernel dt support allows full gpio configuration including termination, direction and polarity nowadays then I might switch over to use that.