raspberrypi/firmware #480: pi2 hangs on boot when connected to hdmi-cec screen and mains applied first

sunnysmith, 2015-10-06:
When using certain HDMI-CEC compliant screens with the feature turned on or off on the screen while utilizing a HDMI-CEC HDMI cable (many are not even though they advertise they are), when you plug in the pi2 it will hang immediately with both the red and green led’s solid.  Doing it enough times corrupts the FAT partition on the sd card. 

The solution is to plug the micro usb into the pi BEFORE you plug the adapter into the mains.  We tested this in many different combinations and found this to be definitively the problem.  We had seen this problem in the field over the last year but always (erroneously) attributed it to the screen.  The best practice until the problem is formally solved is to always power up the pi in this manner (usb then mains) although it is counter-intutive for most since we plug our mobile phones in the other way (mains then usb).  

There are many people experiencing red+green solid leds on boot and the discussion is usually centered on sd card problems or corruption.  This may be the problem.

popcornmix, 2015-10-06:
Are you sure CEC support is relevant?
Does adding ``hdmi_ignore_cec=1`` to config.txt make any difference?

sunnysmith, 2015-10-06:
Yes, it is relevant as is the HDMI cable.  The screens in question were late model Samsung and VIZIO with CEC support.  We tried many combinations of power adapters and HDMI cables and found that if the HDMI cable was truly CEC-complicant then the problem occurred.  The pi2 boots fine with the same screen and a cable is non-CEC compliant.  

As far as config.txt parameters, we tried that one and all other relevant options.  Adding boot_delay to the config.txt has no effect.  Red and green solid as soon as usb is connected so I am led to think that it doesn't make it that far.

popcornmix, 2015-10-07:
Still finding it hard to understand how CEC is involved.
Can you take a CEC enabled HDMI cable, and use tape to insulate pin 13 (CEC):
https://en.wikipedia.org/wiki/HDMI

and report if that has the problem or not.

I always power the Pi by plugging in the mains first, then the USB connector, and my TV supports CEC and no trouble booting.

sunnysmith, 2015-10-07:
Ok, I could not manage to tape the individual pins so I went the route of cutting the conductors. I initially made a mistake which may have revealed the real problem. 

I first erroneously cut the conductor for pin 4 on the plug which is SDA Serial Data according to the wiki page and this solved the problem.  I reconnected the conductor and verified that the problem still existed.  

Then, I correctly cut the conductor for pin 7 on the plug which is CEC.  Reconnecting the conductor for pin 4 (SDA) and leaving 7 cut (CEC), this solved the problem as well. 

With either or both cut, it boots normally.  Reconnecting both, recreates the condition.  

Working on a hunch that this is a power problem I went further.  I used many different 2a and 2.5a adapters and used a 12watt mobile charger with a usb cable with all the same effect.  The problem still occurred.  

So, I had a few pico ups' from pimodules laying around.  I connected it to the pi without the lipo battery and it solved the problem.  The pico ups has some larger caps on it which may be helping with any voltage/current drop. 

Last test, I connected my Fluke to GPIO Pins 2 and 6 and measured 5.165v.  The final weirdness was that the fix (usb then mains) did not work with the fluke attached.  I connected the fluke, then connected the usb, then plugged in the adapter and it hung (red+green solid). 

We deployed a pi2 on a vizio yesterday that had this problem (mains then usb hang) but the small vizio we got in today does not have the problem so it is brand and model specific not necessarily brand specific.

ghollingworth, 2015-10-16:
One thing you could try would be to:

1) Wire back both the CEC and I2C connections...
2) Take a piece of wire and connect the shells of the HDMI and USB (i.e. common the earths between them)

Then try to reproduce the problem...

Gordon

sunnysmith, 2015-10-16:
Problem still occurs with the shells of the HDMI and USB connected with a jumper.

ghollingworth, 2015-10-17:
Can you just show a picture of what you've done, I'm just worried now reading back what I've written that it's not very easy to understand what I'm trying to test!

I would like the shell of the HDMI cable and the shell of the uUSB power input to be common-ed up before any are plugged into the Raspberry Pi (with nothing else plugged in).  My thought is that the HDMI monitor power supply and the uUSB power supply neither have earth connections and therefore the voltage difference between the two shells when you plug them into the Raspberry Pi can be very high, (as much as 700V) but the total stored charge is very low (like a static shock) but when you plug them both into the Pi the charge is passed between the uUSB ground and the HDMI ground causing a 'ground bounce' which can cause problems with the chips on the Pi.

From the other tests you've done such as breaking the I2C or CEC it doesn't look like this is the problem, but it is something I see quite often on my desk (at least a ground difference not a failure)

Gordon

sunnysmith, 2015-10-19:
Gordon, 

The problem was in the firmware. I thought I fried a PI2 when I was doing the testing that you outlined so I grabbed a different one that I had in a reclaim pile.  Surprisingly, it didn't have the problem and moving its SD card to any PI2 did not replicate the problem.  

The PI2 I had grabbed was 3.18.7-v7+ which did not have the problem.  4.1.10-v7+ fixed the problem.  All of the PI2s that had the problem were running 4.1.6-v7+.  

Sorry to cause you guys any unnecessary work/contemplation.   This one really had me stumped.  I should have updated the firmware as step 1. 

Sunny

Ruffio, 2016-06-29:
@sunnysmith if the issue has been resolved, please close this. Thanks.

closed 2016-06-30 by popcornmix