raspberrypi/firmware #481: Pi 2 freeze after some hours after upgrading to b06b317afb60547b436cd30e056b54cec7ae3b8e commit

iz8mbw, 2015-10-09:
Hi.
Starting from the rpi firmware b06b317afb60547b436cd30e056b54cec7ae3b8e commit, my Raspberry Pi 2 freeze after some hours.
The same rpi firmware b06b317afb60547b436cd30e056b54cec7ae3b8e commit on the Raspberry Pi 1 works well.

I'm sorry but I cannot give more info about the freeze, the Pi 2 does not more respond to ping and the green led of the disk activity remains fixed light on...

Considering that the problem happens only on the Pi 2 and not also on the Pi 1, is not possible that this https://github.com/raspberrypi/firmware/issues/369 is the problem?

pelwell, 2015-10-09:
Can you try reverting just the "start*.elf" and "fixup*.dat" files to those from the previous firmware (6785329266e889552b7a142563cb66d743b25610) and see if that makes a difference?

popcornmix, 2015-10-09:
Possibly duplicate of https://github.com/Hexxeh/rpi-update/issues/193
After reverting the start.elf/fixup.dat can you try:
``sudo rpi-update 608c3b3e994e086d400f77469682ff1e8f346e27``
and reboot which may help.

iz8mbw, 2015-10-09:
Hi.
What do you mean for "reverting just the "start*.elf" and "fixup*.dat" files"?
I have already solved with: "rpi-update 7ee23ac1452ad57dcc0ef2c1f074060b4bd9338d"

So now my Pi2 is OK since 2 days withe the https://github.com/Hexxeh/rpi-firmware/commit/7ee23ac1452ad57dcc0ef2c1f074060b4bd9338d firmware version.

pelwell, 2015-10-09:
The "firmware" consists of several parts - the binary files that make up the boot loader(s), the kernel image, the Device Tree blobs and the configuration files. I was suggesting that you could downgrade to an earlier version, copy the "start\*.elf" and "fixup\*.dat" files, upgrade again and then copy the saved files over the new ones to create a hybrid firmware.

@popcornmix thinks this is a kernel problem, not related to the HYP change. If you found that your hybrid system was still unstable then that would be further evidence that this is indeed a kernel bug, and nothing to do with the #369.

iz8mbw, 2015-10-09:
OK.
So I have done:

1. "rpi-update 7ee23ac1452ad57dcc0ef2c1f074060b4bd9338d"
2. Backup the "start*.elf" and "fixup*.dat" files
3. "rpi-update" (so to last commit https://github.com/raspberrypi/firmware/commit/b06b317afb60547b436cd30e056b54cec7ae3b8e)
4. Overwrite the "start*.elf" and "fixup*.dat" files with the "old" one (https://github.com/Hexxeh/rpi-firmware/commit/7ee23ac1452ad57dcc0ef2c1f074060b4bd9338d)
5. Reboot

Now the Raspberry Pi 2 is UP (Linux iz8mbw 4.1.10-v7+ #820 SMP PREEMPT Sun Oct 4 16:26:51 BST 2015 armv7l GNU/Linux
), I will inform you in the next days.

pelwell, 2015-10-09:
Thanks. If we find the problem in the meantime we'll post a comment.

popcornmix, 2015-10-09:
I'm suspecting it is this issue: https://patchwork.ozlabs.org/patch/507215/
I'll add it to a Milhouse OpenELEC build tonight (where the issue has also been reported) and if it is reported as fixed will push it to rpi-update.

iz8mbw, 2015-10-09:
Hi.
Also with the "workaround" the Raspberry Pi 2 freezes again after 4 hours....
Now I back again to the working https://github.com/Hexxeh/rpi-firmware/commit/7ee23ac1452ad57dcc0ef2c1f074060b4bd9338d

xittos, 2015-10-09:
I have the same problem but for me, even the Hexxeh/rpi-firmware@608c3b3 is safe (kernel 4.1.8). The last two firmware commits (kernel 4.1.9 and kernel 4.1.10) freeze randomly after few hours....  (See those thread https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=122620 and https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=122598)

popcornmix, 2015-10-10:
Please try latest rpi-update and report back.

iz8mbw, 2015-10-11:
Updated this morning to last rpi-update, Pi2 is under testing. I will back in the next few hours.

iz8mbw, 2015-10-11:
Untill now, afetr 11 hours of uptime, all it's OK

xittos, 2015-10-12:
Even for me everything is ok now, 1 day and half of uptime....

popcornmix, 2015-10-13:
@iz8mbw okay to close?

iz8mbw, 2015-10-13:
sure!

closed 2015-10-13 by popcornmix

yy502, 2015-10-30:
Hi guys, I still have this issue with my RPi2. v4.1.17 is the last stable version for my Pi. Any later version will result my Pi to freeze (and auto-reboot because I have enabled watchdog) randomly, from <1 hours to a few hours, even when idling.

FYI, I run my Pi headlessly on Ethernet with 2 USB disks attached.

popcornmix, 2015-11-01:
Can you create a new issue - I don't believe your freeze is the same one as reported here.
Please identify the first version of firmware that has the problem. (Using rpi-update <hash> with versions from https://github.com/Hexxeh/rpi-firmware/commits/master).

yy502, 2015-11-01:
Sure. I'll run through each recent version and create a new ticket.

Is there anything I should do to help debugging? I can SSH into my Pi to monitor system log until it crashes/reboots if that helps.

yy502, 2015-11-02:
just before I create a new ticket, I think my issue is same as https://github.com/Hexxeh/rpi-update/issues/193

The latest stable version for me is v4.1.8. v4.1.9 up to v4.1.12 all lead to random crash.

popcornmix, 2015-11-02:
https://github.com/Hexxeh/rpi-update/issues/193 was a duplicate of this issue and is resolved.
If you still have an issue it is something different.

yy502, 2015-11-02:
ok, so how would I produce some useful trace/logging to help debug this issue please?

popcornmix, 2015-11-02:
Linking to the exact firmware (from https://github.com/Hexxeh/rpi-firmware/commits/master) that introduces the problem is most useful.
After a crash there may be evidence in /var/log/messages, so paste that to a paste site.
Narrowing down the problem would be useful. i.e. does the crash still happen when you unplug the USB disks (or any other non-default peripherals).
Are you overclocking? If so disable that and test.

yy502, 2015-11-02:
OK, will do. Thanks.

tkalinows, 2016-02-05:
Hi, 
I've same problem.  My uname -a:
Linux kali-iHome 4.1.13-v7+ #826 SMP PREEMPT Fri Nov 13 20:19:03 GMT 2015 armv7l GNU/Linux

Logs:

kali@ka-iHome:~$ sudo tail /var/log/daemon.log
Feb  5 06:17:19 kali-iHome apache2[649]: .
Feb  5 06:17:19 kali-iHome systemd[1]: Started LSB: Apache2 web server.
Feb  5 06:17:19 kali-iHome systemd[1]: Starting Multi-User System.
Feb  5 06:17:19 kali-iHome systemd[1]: Reached target Multi-User System.
Feb  5 06:17:19 kali-iHome systemd[1]: Starting Graphical Interface.
Feb  5 06:17:19 kali-iHome systemd[1]: Reached target Graphical Interface.
Feb  5 06:17:19 kali-iHome systemd[1]: Starting Update UTMP about System Runlevel Changes...
Feb  5 06:17:19 kali-iHome systemd[1]: Started Update UTMP about System Runlevel Changes.
Feb  5 06:17:19 kali-iHome systemd[1]: Startup finished in 3.554s (kernel) + 20.095s (userspace) = 23.649s.
Feb  5 17:26:10 kali-iHome systemd[1]: Time has been changed
kali@kali-iHome:~$ sudo tail /var/log/syslog
Feb  5 06:17:19 kali-iHome apache2[649]: .
Feb  5 06:17:19 kali-iHome systemd[1]: Started LSB: Apache2 web server.
Feb  5 06:17:19 kali-iHome systemd[1]: Starting Multi-User System.
Feb  5 06:17:19 kali-iHome systemd[1]: Reached target Multi-User System.
Feb  5 06:17:19 kali-iHome systemd[1]: Starting Graphical Interface.
Feb  5 06:17:19 kali-iHome systemd[1]: Reached target Graphical Interface.
Feb  5 06:17:19 kali-iHome systemd[1]: Starting Update UTMP about System Runlevel Changes...
Feb  5 06:17:19 kali-iHome systemd[1]: Started Update UTMP about System Runlevel Changes.
Feb  5 06:17:19 kali-iHome systemd[1]: Startup finished in 3.554s (kernel) + 20.095s (userspace) = 23.649s.
Feb  5 17:26:10 kali-iHome systemd[1]: Time has been changed

I had off ipv6 because it was last of above logs before freeze - but no result. System steale freeze after few hours

any sugesttion?

popcornmix, 2016-02-05:
If you are overclocking then disable that first.
If that doesn't help then create a new issue. This issue is closed.

tkalinows, 2016-02-05:
ok. Sorrty.