raspberrypi/firmware #482: [rpi2] make fatal error: vchost_config.h: No such file or directory

jalona, 2015-10-11:
not correct location vchost_config.h in /opt/vc/include/interface/vmcs_host/vcgencmd.h:33:27

````
...
Scanning dependencies of target brcm
[ 71%] Building CXX object plugins/egldeviceintegration/brcm/CMakeFiles/brcm.dir/eglfsbrcmintegration.cpp.o
In file included from /opt/vc/include/bcm_host.h:54:0,
                 from /home/alarm/build/albeda-pkgbuilds/greenisland-rpi-git/src/greenisland-rpi-git/plugins/egldeviceintegration/brcm/eglfsbrcmintegration.cpp:29:
/opt/vc/include/interface/vmcs_host/vcgencmd.h:33:27: fatal error: vchost_config.h: No such file or directory
compilation terminated.
plugins/egldeviceintegration/brcm/CMakeFiles/brcm.dir/build.make:62: recipe for target 'plugins/egldeviceintegration/brcm/CMakeFiles/brcm.dir/eglfsbrcmintegration.cpp.o' failed
make[2]: *** [plugins/egldeviceintegration/brcm/CMakeFiles/brcm.dir/eglfsbrcmintegration.cpp.o] Error 1
CMakeFiles/Makefile2:694: recipe for target 'plugins/egldeviceintegration/brcm/CMakeFiles/brcm.dir/all' failed
make[1]: *** [plugins/egldeviceintegration/brcm/CMakeFiles/brcm.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
[alarm@pi greenisland-rpi-git]$ 
````

````
...
#include "vchost_config.h"
...
````
Correct location vchost_config.h /opt/vc/include/interface/vmcs_host/linux

pelwell, 2015-10-11:
What command did you run, and in which directory?

jalona, 2015-10-11:
Close.
solution: raspberrypi/firmware/pull/171

closed 2015-10-11 by jalona