raspberrypi/firmware #483: Kernel Oops - BUG: Bad page map in process

richardkennedy, 2015-10-12:
One of my headless clients running Debian Jessie and Kernel 4.1.8 (608c3b3e994e086d400f77469682ff1e8f346e27), became unresponsive over the weekend due to a Kernel Oops (logs below) however it's the first time I've personally come across a 'Bad page map' error, is this referencing a possible OOM event? At the time of the failure, my records show that the device had a temperature of 54.6 degrees, CPU uitlisation was at 24% and RAM utilisation was at 40%. Any insight appreciated.

Oct 10 09:47:41 device-name kernel: [90492.426587] Unable to handle kernel paging request at virtual address 00003394
Oct 10 09:47:41 device-name kernel: [90492.433809] pgd = b8724000
Oct 10 09:47:41 device-name kernel: [90492.437468] [00003394] *pgd=386a9831, *pte=00000000, *ppte=00000000
Oct 10 09:47:41 device-name kernel: [90492.443752] Internal error: Oops: 17 [#1] PREEMPT SMP ARM
Oct 10 09:47:41 device-name kernel: [90492.449137] Modules linked in: cpufreq_stats joydev evdev bcm2835_gpiomem uio_pdrv_genirq uio snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd fuse ipv6
Oct 10 09:47:41 device-name kernel: [90492.463534] CPU: 3 PID: 457 Comm: custom-script Not tainted 4.1.8-v7+ #818
Oct 10 09:47:41 device-name kernel: [90492.470046] Hardware name: BCM2709
Oct 10 09:47:41 device-name kernel: [90492.473439] task: b9361200 ti: b4816000 task.ti: b4816000
Oct 10 09:47:41 device-name kernel: [90492.478830] PC is at down_write+0x24/0x68
Oct 10 09:47:41 device-name kernel: [90492.482830] LR is at anon_vma_clone+0x84/0x1b4
Oct 10 09:47:41 device-name kernel: [90492.487264] pc : [<8055be24>]    lr : [<801268e4>]    psr: 60000013
Oct 10 09:47:41 device-name kernel: [90492.487264] sp : b4817e58  ip : b4817e70  fp : b4817e6c
Oct 10 09:47:41 device-name kernel: [90492.498717] r10: b4aa95c0  r9 : 00003390  r8 : b87c3390
Oct 10 09:47:41 device-name kernel: [90492.503929] r7 : b489d824  r6 : 00000000  r5 : b92b72c0  r4 : 00003394
Oct 10 09:47:41 device-name kernel: [90492.510440] r3 : ffff0001  r2 : a0000013  r1 : 00000001  r0 : 00003394
Oct 10 09:47:41 device-name kernel: [90492.516952] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Oct 10 09:47:41 device-name kernel: [90492.524071] Control: 10c5387d  Table: 3872406a  DAC: 00000015
Oct 10 09:47:41 device-name kernel: [90492.529802] Process custom-script (pid: 457, stack limit = 0xb4816210)
Oct 10 09:47:41 device-name kernel: [90492.535967] Stack: (0xb4817e58 to 0xb4818000)
Oct 10 09:47:41 device-name kernel: [90492.540313] 7e40:                                                       00000000 b4bc1738
Oct 10 09:47:41 device-name kernel: [90492.548476] 7e60: b4817ea4 b4817e70 801268e4 8055be0c b4817eac b489d7e8 80120da0 b4bc1738
Oct 10 09:47:41 device-name kernel: [90492.556638] 7e80: b489d7e8 b4bd1c00 00000001 808664c0 00000000 808664c0 b4817ec4 b4817ea8
Oct 10 09:47:41 device-name kernel: [90492.564800] 7ea0: 80126a48 8012686c b489d7e8 b4bc1738 b4bd1c00 00000001 b4817f4c b4817ec8
Oct 10 09:47:41 device-name kernel: [90492.572963] 7ec0: 80025084 80126a20 b4817f70 b92ba9c0 80569a28 b962d36c 76f31278 01200011
Oct 10 09:47:41 device-name kernel: [90492.581125] 7ee0: b4bc15ec b4bc15e0 b4bc15e8 b4bc15d8 b4bd1c38 b90a2578 b4af0000 00000000
Oct 10 09:47:41 device-name kernel: [90492.589288] 7f00: b4af0288 b90a2540 00000000 00000000 46b09d5e 0000524d 00000000 fffffff4
Oct 10 09:47:41 device-name kernel: [90492.597450] 7f20: b4817f3c 76f31278 00000000 01200011 00000078 00000000 00000000 00000000
Oct 10 09:47:41 device-name kernel: [90492.605612] 7f40: b4817f8c b4817f50 80025d30 800245fc 00000000 00000000 00000008 00000000
Oct 10 09:47:41 device-name kernel: [90492.613775] 7f60: b4817f78 802ff778 76f31278 00000000 76e87000 00000078 8000fb88 b4816000
Oct 10 09:47:41 device-name kernel: [90492.621937] 7f80: b4817fa4 b4817f90 8002614c 80025c90 76f31278 800d06ec 00000000 b4817fa8
Oct 10 09:47:41 device-name kernel: [90492.630099] 7fa0: 8000f980 8002612c 76f31278 00000000 01200011 00000000 00000000 00000000
Oct 10 09:47:41 device-name kernel: [90492.638261] 7fc0: 76f31278 00000000 76e87000 00000078 76f31210 00000000 000001c9 7eb1210c
Oct 10 09:47:41 device-name kernel: [90492.646424] 7fe0: 76f316d0 7eb120e0 00000000 76de9edc 60000010 01200011 00000000 00000000
Oct 10 09:47:41 device-name kernel: [90492.654595] [<8055be24>] (down_write) from [<801268e4>] (anon_vma_clone+0x84/0x1b4)
Oct 10 09:47:41 device-name kernel: [90492.662240] [<801268e4>] (anon_vma_clone) from [<80126a48>] (anon_vma_fork+0x34/0x164)
Oct 10 09:47:41 device-name kernel: [90492.670148] [<80126a48>] (anon_vma_fork) from [<80025084>] (copy_process.part.53+0xa94/0x15d0)
Oct 10 09:47:41 device-name kernel: [90492.678748] [<80025084>] (copy_process.part.53) from [<80025d30>] (do_fork+0xac/0x3ec)
Oct 10 09:47:41 device-name kernel: [90492.686653] [<80025d30>] (do_fork) from [<8002614c>] (SyS_clone+0x2c/0x34)
Oct 10 09:47:41 device-name kernel: [90492.693520] [<8002614c>] (SyS_clone) from [<8000f980>] (ret_fast_syscall+0x0/0x54)
Oct 10 09:47:41 device-name kernel: [90492.701076] Code: e1a04000 f57ff05b f590f000 e59f303c (e1902f9f) 
Oct 10 09:47:41 device-name kernel: [90492.710916] ---[ end trace b67b90c0f8edd4cc ]---
Oct 10 09:47:45 device-name kernel: [90496.954752] swap_free: Bad swap file entry 01010800
Oct 10 09:47:45 device-name kernel: [90496.959665] BUG: Bad page map in process ssh-agent  pte:80840000 pmd:387c2831
Oct 10 09:47:45 device-name kernel: [90496.966845] addr:7edb4000 vm_flags:00100173 anon_vma:b86e1870 mapping:  (null) index:7efe0
Oct 10 09:47:45 device-name kernel: [90496.975158] file:  (null) fault:  (null) mmap:  (null) readpage:  (null)
Oct 10 09:47:45 device-name kernel: [90496.981873] CPU: 0 PID: 1101 Comm: ssh-agent Tainted: G      D         4.1.8-v7+ #818
Oct 10 09:47:45 device-name kernel: [90496.989730] Hardware name: BCM2709
Oct 10 09:47:45 device-name kernel: [90496.993172] [<80018440>] (unwind_backtrace) from [<80013e0c>] (show_stack+0x20/0x24)
Oct 10 09:47:45 device-name kernel: [90497.000956] [<80013e0c>] (show_stack) from [<80558600>] (dump_stack+0x98/0xe0)
Oct 10 09:47:45 device-name kernel: [90497.008222] [<80558600>] (dump_stack) from [<8011ab9c>] (print_bad_pte+0x130/0x1a8)
Oct 10 09:47:45 device-name kernel: [90497.015923] [<8011ab9c>] (print_bad_pte) from [<8011c570>] (unmap_single_vma+0x55c/0x588)
Oct 10 09:47:45 device-name kernel: [90497.024112] [<8011c570>] (unmap_single_vma) from [<8011d3c8>] (unmap_vmas+0x64/0x78)
Oct 10 09:47:45 device-name kernel: [90497.031891] [<8011d3c8>] (unmap_vmas) from [<801239d8>] (exit_mmap+0xe8/0x1f0)
Oct 10 09:47:45 device-name kernel: [90497.039169] [<801239d8>] (exit_mmap) from [<80024348>] (mmput+0x50/0xf4)
Oct 10 09:47:45 device-name kernel: [90497.045915] [<80024348>] (mmput) from [<80028c30>] (do_exit+0x2c0/0xa64)
Oct 10 09:47:45 device-name kernel: [90497.052635] [<80028c30>] (do_exit) from [<80029470>] (do_group_exit+0x50/0xcc)
Oct 10 09:47:45 device-name kernel: [90497.059910] [<80029470>] (do_group_exit) from [<80033ed4>] (get_signal+0x2b0/0x6e0)
Oct 10 09:47:45 device-name kernel: [90497.067609] [<80033ed4>] (get_signal) from [<80013194>] (do_signal+0x98/0x3ac)
Oct 10 09:47:45 device-name kernel: [90497.074876] [<80013194>] (do_signal) from [<80013690>] (do_work_pending+0xb8/0xc8)
Oct 10 09:47:45 device-name kernel: [90497.082461] [<80013690>] (do_work_pending) from [<8000f9e4>] (work_pending+0xc/0x20)
Oct 10 09:47:45 device-name kernel: [90497.090234] swap_free: Bad swap file entry 01034800
Oct 10 09:47:45 device-name kernel: [90497.095156] BUG: Bad page map in process ssh-agent  pte:81a40000 pmd:387c2831
Oct 10 09:47:45 device-name kernel: [90497.102298] addr:7edb5000 vm_flags:00100173 anon_vma:b86e1870 mapping:  (null) index:7efe1
Oct 10 09:47:45 device-name kernel: [90497.110587] file:  (null) fault:  (null) mmap:  (null) readpage:  (null)
Oct 10 09:47:45 device-name kernel: [90497.117375] CPU: 0 PID: 1101 Comm: ssh-agent Tainted: G    B D         4.1.8-v7+ #818
Oct 10 09:47:45 device-name kernel: [90497.125240] Hardware name: BCM2709
Oct 10 09:47:46 device-name kernel: [90497.128700] [<80018440>] (unwind_backtrace) from [<80013e0c>] (show_stack+0x20/0x24)
Oct 10 09:47:46 device-name kernel: [90497.136501] [<80013e0c>] (show_stack) from [<80558600>] (dump_stack+0x98/0xe0)
[... 265 lines cut ...]
Oct 10 09:47:47 device-name kernel: [90499.029069] flags: 0x400(reserved)
[... cut ...]
Oct 10 09:47:48 device-name kernel: [90499.184267] [<80013194>] (do_signal) from [<80013690>] (do_work_pending+0xb8/0xc8)
Oct 10 09:47:48 device-name kernel: [90499.191881] [<80013690>] (do_work_pending) from [<8000f9e4>] (work_pending+0xc/0x20)
Oct 10 09:47:48 device-name kernel: [90499.199968] BUG: Bad rss-counter state mm:b9234e00 idx:0 val:-1
Oct 10 09:47:48 device-name kernel: [90499.205962] BUG: Bad rss-counter state mm:b9234e00 idx:1 val:2
Oct 10 09:47:48 device-name kernel: [90499.211811] BUG: Bad rss-counter state mm:b9234e00 idx:2 val:-13

pelwell, 2015-10-12:
Are you overclocking anything - ARM, SDRAM or GPU?

richardkennedy, 2015-10-12:
Yep (see below), the device is currently set to the 'Pi2' overclocking profile via raspi-config. I was concerned about that being a potential cause yesterday evening, so I ran approximately 2 hours of stress tests (OpenGL, CPU/Storage) but I failed to replicate any form of instability so personally dismissed it as a potential cause.

/boot/config.txt
arm_freq=1000
core_freq=500
sdram_freq=500
over_voltage=2

popcornmix, 2015-10-13:
Can you remove the overclock and see if you get the kernel panic.
Turbo mode is not stable on many Pi2's.

alexara, 2015-11-01:
I had the same issue using overclock ```over_voltage=2```:
```
Nov  1 16:25:28 raspberrypi2 kernel: [14047.105525] BUG: Bad page map in process beam.smp  pte:2514155f pmd:2fce2831
Nov  1 16:25:28 raspberrypi2 kernel: [14047.113349] page:b656cd24 count:1 mapcount:-1 mapping:a5c00bac index:0x72
Nov  1 16:25:28 raspberrypi2 kernel: [14047.120973] flags: 0x2027c(referenced|uptodate|dirty|lru|active|arch_1|mappedtodisk)
Nov  1 16:25:28 raspberrypi2 kernel: [14047.130740] page dumped because: bad pte
Nov  1 16:25:28 raspberrypi2 kernel: [14047.135583] addr:72c3d000 vm_flags:00100077 anon_vma:b51f3270 mapping:  (null) index:72c3d
Nov  1 16:25:28 raspberrypi2 kernel: [14047.146158] file:  (null) fault:  (null) mmap:  (null) readpage:  (null)
Nov  1 16:25:28 raspberrypi2 kernel: [14047.154270] CPU: 2 PID: 3502 Comm: beam.smp Not tainted 4.1.12-v7+ #824
Nov  1 16:25:28 raspberrypi2 kernel: [14047.162192] Hardware name: BCM2709
Nov  1 16:25:28 raspberrypi2 kernel: [14047.166486] [<80018444>] (unwind_backtrace) from [<80013e08>] (show_stack+0x20/0x24)
Nov  1 16:25:28 raspberrypi2 kernel: [14047.176147] [<80013e08>] (show_stack) from [<80557fb4>] (dump_stack+0x98/0xe0)
Nov  1 16:25:28 raspberrypi2 kernel: [14047.184430] [<80557fb4>] (dump_stack) from [<8011a858>] (print_bad_pte+0x130/0x1a8)
Nov  1 16:25:28 raspberrypi2 kernel: [14047.194346] [<8011a858>] (print_bad_pte) from [<8011c168>] (unmap_single_vma+0x544/0x588)
Nov  1 16:25:28 raspberrypi2 kernel: [14047.204782] [<8011c168>] (unmap_single_vma) from [<8011cfdc>] (unmap_vmas+0x64/0x78)
Nov  1 16:25:28 raspberrypi2 kernel: [14047.214967] [<8011cfdc>] (unmap_vmas) from [<801235e8>] (exit_mmap+0xe8/0x1f0)
Nov  1 16:25:28 raspberrypi2 kernel: [14047.223549] [<801235e8>] (exit_mmap) from [<80024360>] (mmput+0x50/0xf4)
Nov  1 16:25:28 raspberrypi2 kernel: [14047.231584] [<80024360>] (mmput) from [<80028c7c>] (do_exit+0x2c0/0xa64)
Nov  1 16:25:28 raspberrypi2 kernel: [14047.239598] [<80028c7c>] (do_exit) from [<800294b8>] (do_group_exit+0x4c/0xcc)
Nov  1 16:25:28 raspberrypi2 kernel: [14047.248106] [<800294b8>] (do_group_exit) from [<80033f1c>] (get_signal+0x2b0/0x6e0)
Nov  1 16:25:28 raspberrypi2 kernel: [14047.258290] [<80033f1c>] (get_signal) from [<80013190>] (do_signal+0x98/0x3ac)
Nov  1 16:25:28 raspberrypi2 kernel: [14047.266850] [<80013190>] (do_signal) from [<8001368c>] (do_work_pending+0xb8/0xc8)
Nov  1 16:25:28 raspberrypi2 kernel: [14047.276974] [<8001368c>] (do_work_pending) from [<8000f9e4>] (work_pending+0xc/0x20)
Nov  1 16:25:28 raspberrypi2 kernel: [14047.287303] Disabling lock debugging due to kernel taint
Nov  1 16:25:28 raspberrypi2 kernel: [14047.294303] BUG: Bad page state in process beam.smp  pfn:25141
Nov  1 16:25:28 raspberrypi2 kernel: [14047.301488] page:b656cd24 count:0 mapcount:-1 mapping:a5c00bac index:0x72
```
I removed the overlock now and report back as well in case the kernel Oops happens again.

popcornmix, 2015-11-01:
@richardkennedy have you tested without overclock?

richardkennedy, 2015-11-01:
Not entirely, reduced it to 900Mhz with a core frequency of 250, sdram of 450 and an over voltage of 2. Still monitoring at the moment, I'll go through my logs this evening to see how it's performing.

popcornmix, 2015-11-01:
The most likely issue is the sdram_freq. If it seems stable with no overclock, you may well be okay with the arm and core overclocks in place, but the sdram overclock. (My guess is that >90% of Pi2's can handle the core and arm overclock, but perhaps only 50% can handle the sdram overclock reliably).

But first, make sure everything is stable without any overclock.

Ruffio, 2016-08-08:
@richardkennedy how did this end up? Can this issue be closed? Please do so, if it has been resolved or provide further information for investigation. Thanks.

Ruffio, 2016-08-10:
@popcornmix this issue looks like it was caused by overclocking sdram. It should be closed as there have been no response to your suggestion 1 Nov 2015

closed 2016-08-10 by popcornmix