raspberrypi/firmware #484: Emulationstation: Screen don't fit

42Swampy, 2015-10-12:
I’m using a Raspberry Pi 2 with Composite Video and an old TV (CRT) to use retropie. Up to the Commits on May 9, 2015 all works fine. Since the Commits on May 13, 2015 I have a screen problem. The Emulationstation screen is too big for the TV. The screen is OK for the emulators and for the boot-process. Only Emulationstation don’t fit.

Here my config.txt:

overscan_left=24
overscan_right=8
overscan_top=-4
overscan_bottom=-4
sdtv_mode=2
gpu_mem_256=128
gpu_mem_512=256
gpu_mem_1024=256
overscan_scale=1
decode_MPG2=0x586f88c2
decode_WVC1=0xb535a4b7
max_usb_current=1
gpu_mem=512
arm_freq=1000
core_freq=500
sdram_freq=500
over_voltage=2

popcornmix, 2015-10-12:
What does ``tvservice -s`` report?

popcornmix, 2015-10-12:
Also have you tried latest rpi-update firmware?

42Swampy, 2015-10-12:
tvservice -s
`state 0x80001 [PAL 4:3], 720x576 @ 50.00Hz, interlaced`

42Swampy, 2015-10-12:
I tried the latest firmware and the problem remains.
I tested also:

kernel: Bump to 4.0.5
6d25078b7cb361052ae280d552561d7250d33f5b
kernel: Bump to 3.18.16
33a6707cf1c96b8a2b5dac2ac9dead590db9fcaa
kernel: Bump to 3.18.14
4f95c894c6a96b265d98f04a889e469833f7e996

The problem exists in all versions newer than:

kernel: Bump to 3.18.13
9354845ea5a961f267bcf440f92c9bd21be29b35

popcornmix, 2015-10-12:
Does it improve if you replace the negative overscan_ options with zero?

42Swampy, 2015-10-13:
Tried it with overscan_top=0 and overscan_bottom=0. No improvement!

42Swampy, 2015-10-13:
As far as I can remember, I had the same problem with raspbian-desktop when I updated the kernel(firmware) to 4.x.

popcornmix, 2015-10-13:
Can you point me at instructions you used for installing emulationstation.
Any other changes needed to see the problem (apart from the config.txt settings listed and using composite output).

joolswills, 2015-10-13:
User is using RetroPie - To replicate you could download a retropie image from http://blog.petrockblock.com/retropie/retropie-downloads/ or alternatively

```
git clone --depth=1 https://github.com/RetroPie/RetroPie-Setup.git
cd RetroPie-Setup
sudo ./retropie_packages.sh emulationstation depends
sudo ./retropie_packages.sh emulationstation install_bin
sudo ./retropie_packages.sh emulationstation configure
sudo ./retropie_packages.sh esthemes install_theme carbon HerbFargus
```

which should set up emulationstation in /opt/retropie/supplementary/emulationstation/emulationstation - (if you prefer to build from source, you can replace the emulationstation lines with just `sudo ./retropie_packages.sh emulationstation`)

popcornmix, 2015-10-14:
I can run emulationstation and I get dialog:
``We can't find any systems! Check that your paths are correct in the systems configuration file, and your game directory has at least one game with the correct extension."
Is that dialog enough to see the problem you have, or will I have to get past this?

42Swampy, 2015-10-14:
I think it is enough to see the problem. I expect, that you are already in the graphical user interface of emulationstation.

42Swampy, 2015-10-14:
If you want to ensure it, you can copy one free rom into `/home/pi/RetroPie/roms/`<system to emulate>.  Perhaps a fakefile will do it too.

joolswills, 2015-10-14:
I assume my instructions were followed to get a minimal install. To add a single system so it doesn't complain

```
sudo ./retropie_packages.sh frotz
```

which will install frotz and zork1/2/3

popcornmix, 2015-10-14:
Still not finding any systems with frotz installed. Does .emulationstation/es_systems.cfg need to be setup manually (it seems to only reference nes)?

Anyway I can reproduce the issue from the error screen. @42Swampy can you try this test build:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_scale.zip

and let me know if all is okay.

joolswills, 2015-10-14:
aah .emulationstation/es_systems.cfg would be overriding the retropie installed one - remove that and it should use /etc/emulationstation/es_systems.cfg which contains the frotz configuration.

But not important if you can reproduce it. thanks.

42Swampy, 2015-10-14:
I have tested the build and all seems to be OK.

popcornmix, 2015-10-21:
The fix should be present in latest "rpi-update" firmware. Please test and close if you are still happy.

42Swampy, 2015-10-22:
I'm still happy. Thanks a lot!

closed 2015-10-22 by 42Swampy