raspberrypi/firmware #486: mmal: Custom pitch for MMAL_ENCODING_I420

julianscheel, 2015-10-14:
Is it possible to make the decoder render into buffers with a custom pitch and custom plane offsets?
I tried to set pitch and offset for the planes before I pass the mmal_buffer with the buffer_header type structure like this:
```
buffer->type->video.planes = picture->i_planes;
for (int i = 0; i < picture->i_planes; i++) {
    if (i > 0)
        offset = offset + picture->p[i - 1].i_pitch * picture->p[i - 1].i_lines;
    buffer->type->video.offset[i] = offset;
    buffer->type->video.pitch[i] = picture->p[i].i_pitch;
}
```

But this is not taken into account, so for now I assume it's not configurable but can only be read back from this structure?

popcornmix, 2015-10-14:
In general pitches are needed due to hardware constraints.
I think you can only set them on the component's port, not through the mmal buffer (which is read only).
The component's port will reject pitch settings that can't be obeyed.
@6by9 may be able to confirm that.

julianscheel, 2015-10-14:
@popcornmix I couldn't spot any point where a pitch could be specified on a port?

popcornmix, 2015-10-14:
In openmax it will be called nStride.
In mmal I believe mmal->format->es->video.width equates to nStride, and
mmal->format->es->video.crop.width equates to nFrameWidth.

julianscheel, 2015-10-14:
Ah yes, make sense. Actually the stride is matching what I need. But the plane offset is wrong. For feeding into avcodec the planes needs to be 580 lines high, but this is not accepted by mmal, when trying it like this:
```
sys->output->format->es->video.height = 580;
sys->output->format->es->video.crop.height = 576;
status = mmal_port_format_commit(sys->output);
```
So I fear it is not supported to have a gap between the yuv planes?

6by9, 2015-10-14:
@popcornmix has directed you to the right point already - format width/height are nStride and nSliceHeight, whilst crop.width/height are nFrameWidth/Height.
Hardware restrictions on the buffer parameters are enforced on nStride and nSliceHeight. For YUV420 the stride must be a multiple of 32, and sliceheight a multiple of 16. Width/height must be a multiple of 2 to avoid nasties in chroma subsampling.

580/16 = 36.25. You should be able to define it as 592 if that helps with avcodec.

(There may be a couple of components that do don't directly obey the port definition on strides/heights - it depends how rigorous the author was being when they were converting the port format into the internal structures defining the images. If you do find oddballs then let us know and they may be easily fixable)

julianscheel, 2015-10-14:
@6by9 Thanks for the details. This allows me indeed to get things going :)

closed 2015-10-14 by julianscheel