raspberrypi/firmware #487: Default pin pull state upon boot

industrial-pi, 2015-10-16:
I am evaluating the use a Raspberry Pi Model B+ v1.2 board in an industrial control system. I am perfectly able to control all the pin pullups and pulldowns from userspace (I have verified all with an oscilloscope) however this does not help me.

In particular, I need to disable the pullup on GPIO17 which is connected to a hardware watchdog circuit in my application. The pin must float, with no more than 10 uA leakage current, until it first drives the watchdog. The 50-60 kOhm internal pullup is enough to drive more than 3 times the allowed leakage current.

After driving the pin with more than 10 uA for the first time, the Pi must contine toggling it faster than 10 Hz to prevent the watchdog circuit from triggering an alarm, leaving no time to boot and start a toggling process from userspace.

The board runs raspbian and raspberrypi-bootloader version 1.20150923-1 straight out of the repository.

I have unsuccesfully tried changing the default pullup resistor state of GPIO17 by following the dt-blob method described at the following link. 

https://www.raspberrypi.org/documentation/configuration/pin-configuration.md

Could you please let me know if the article above is still current and if there is any diagnostic messaging that can be enabled to allow troubleshooting this.

Some more information: I read in the BCM2835 ARM peripherals datasheet (page 100, GPIO Pull-up/down register GPPUD) that "the pullups are maintained even in power-down mode when the core is off, when all register content is lost". And if I disable the pullup from userland, and issue a sudo reboot command, the pull stays in the last state (as expected) until the next cold boot.

Another piece of information is that I am unable to change the deafult pull state to down as well as off. In other words, after a cold boot the pull is always up.

I really like the raspberry and it's a perfect fit for our application, however I'll have to drop it unless I can disable the pull upon boot. 

I added just one line, then compiled it with dtc and put the resulting dt-blob.bin under /boot

```
pin@p17 { function = "input";  termination = "no_pulling"; }; // Watchdog
```

The complete dt-blob.dts source follows

```
/dts-v1/;

/ {
   videocore {
      pins_rev1 {
         pin_config {
            pin@default {
               polarity = "active_high";
               termination = "pull_down";
               startup_state = "inactive";
               function = "input";
            }; // pin
            pin@p2  { function = "i2c1";   termination = "pull_up"; }; // I2C 1 SDA
            pin@p3  { function = "i2c1";   termination = "pull_up"; }; // I2C 1 SCL
            pin@p5  { function = "output"; termination = "pull_down"; }; // CAM_LED
            pin@p6  { function = "output"; termination = "pull_down"; }; // LAN NRESET
            pin@p14 { function = "uart0";  termination = "no_pulling"; drive_strength_mA = < 8 >; }; // TX uart0
            pin@p15 { function = "uart0";  termination = "pull_up"; drive_strength_mA = < 8 >; }; // RX uart0
            pin@p16 { function = "output"; termination = "pull_up"; polarity="active_low"; }; // activity LED
            pin@p27 { function = "output"; termination = "no_pulling";    }; // Camera shutdown
            pin@p40 { function = "pwm";    termination = "no_pulling"; drive_strength_mA = < 16 >; }; // Left audio
            pin@p45 { function = "pwm";    termination = "no_pulling"; drive_strength_mA = < 16 >; }; // Right audio
            pin@p46 { function = "input";  termination = "no_pulling";    }; // Hotplug
            pin@p47 { function = "input";  termination = "no_pulling";    }; // SD_CARD_DETECT
            pin@p48 { function = "sdcard"; termination = "pull_up";    drive_strength_mA = < 8 >; }; // SD CLK
            pin@p49 { function = "sdcard"; termination = "pull_up";    drive_strength_mA = < 8 >; }; // SD CMD
            pin@p50 { function = "sdcard"; termination = "pull_up";    drive_strength_mA = < 8 >; }; // SD D0
            pin@p51 { function = "sdcard"; termination = "pull_up";    drive_strength_mA = < 8 >; }; // SD D1
            pin@p52 { function = "sdcard"; termination = "pull_up";    drive_strength_mA = < 8 >; }; // SD D2
            pin@p53 { function = "sdcard"; termination = "pull_up";    drive_strength_mA = < 8 >; }; // SD D3
         }; // pin_config

         pin_defines {
            pin_define@HDMI_CONTROL_ATTACHED {
               type = "internal";
               number = <46>;
            };
            pin_define@NUM_CAMERAS {
               type = "internal";
               number = <1>;
            };
[... 922 lines cut ...]

      pins_cm {
         pin_config {
            pin@default {
               polarity = "active_high";
               termination = "pull_down";
               startup_state = "inactive";
               function = "input";
            }; // pin
            pin@p14 { function = "uart0";  termination = "no_pulling";    }; // TX uart0
            pin@p15 { function = "uart0";  termination = "pull_up"; }; // RX uart0
            pin@p48 { function = "sdcard"; termination = "pull_up";    drive_strength_mA = < 8 >; }; // SD CLK
            pin@p49 { function = "sdcard"; termination = "pull_up";    drive_strength_mA = < 8 >; }; // SD CMD
            pin@p50 { function = "sdcard"; termination = "pull_up";    drive_strength_mA = < 8 >; }; // SD D0
            pin@p51 { function = "sdcard"; termination = "pull_up";    drive_strength_mA = < 8 >; }; // SD D1
            pin@p52 { function = "sdcard"; termination = "pull_up";    drive_strength_mA = < 8 >; }; // SD D2
            pin@p53 { function = "sdcard"; termination = "pull_up";    drive_strength_mA = < 8 >; }; // SD D3
         }; // pin_config

         pin_defines {
         }; // pin_defines
      }; // pins_cm
   };
};
```

pelwell, 2015-10-16:
Compiling your dt-blob.dts to dt-blob.bin and loading it onto my Rev 1.2 B+ gives the expected behaviour - the pull (down) is removed shortly after power is applied. If I hide the dt-blob.bin and warm reboot (i.e. sudo shutdown -r now) then that lack of a pull persists (this behaviour changed in July - prior to that all pulls were initialised to their defaults, but now only those pins mentioned in the dt-blob are changed). A full power-cycle restores the pull to the usual pull-down.

If you have external hardware connected to the GPIO pins then please remove it and repeat your testing as I suspect the pull-up is external to your Pi. If not, then it sounds like your Pi may have a fault.

swarren, 2016-03-19:
I believe the default pull state documented in BCM2835-ARM-Peripherals.pdf will always apply at cold boot; it's something that is implemented in hardware or non-modifiable boot ROM. Once the FW gets started, it could reprogram the pull based on DT/..., but this will always need some time, so there will always be a window where the default HW configuration is active. A reboot/reset (a) possibly just resets the CPUs and not all HW modules and (b) is a reset not a cold boot anyway. Either would explain why the HW defaults don't get re-applied temporarily during a reboot.

Perhaps you can choose a different pin for this purpose; some pins have a default pull-up and some a default pull-down according to that PDF, but I don't recall any that start out floating. Alternatively, you could insert a tri-state driver chip between GPIO17 and your application, although you'd need to use some other GPIO (or perhaps I2C/SPI) to control that driver's tristate.

closed 2017-12-21 by JamesH65