raspberrypi/firmware #488: hdmi_group = 2 and hmdi_mode=14 not working after Jessie install

Gordon999, 2015-10-21:
This mode doesn't work with Jessie on my Pi. Doesn't switch to 840x480 but on my monitor appears to stay hi res (1920 x 1080?) but look out of focus.

popcornmix, 2015-10-22:
Report output of:
```
tvservice -s
fbset
vcgencmd get_config int
vcgencmd version
```

Gordon999, 2015-10-22:
I had to read it off a fuzzy screen so there maybe some errors

tvservice –s

state 0x120006 [DVI DMT {14} ROB full 16:9] 848x480 @ 60Hz progressive

fbset

mode “1920 x 1080”
       geometry 1920,1080,1920,1080 16
       timings 0 0 0 0 0 0 0
       rgba 5/11.6/5.5/0/0/16
endmode

vcgencmd get_config int

arm_frq=900
config_hdmi_boost=5
disable_commandline_tags=2
disable_l2cache=1
disable_overscan=1
enmc_pll_core=1
force_eprom_read=1
force_pwm_open=1
framebuffer_height=1080
framebuffer_ignore_alpha=1
framebuffer_swap=1
framebuffer_width=1920
hdmi_force_cee_address=65535
hdmi_group=2
hdmi_mode=14
overvoltage_avs=0x1b774
pause_burst_frames=1
program_serial_random=1
sdram_freq=750
temp_limit=85

vcgencmd version

Sep 23 2015 12:21:16
C1561006148c90a3ba28cc376c9c01e05a77d6d5
[clean] (release)
—

closed 2015-10-22 by Gordon999

popcornmix, 2015-10-22:
I assume you closed because you have the answer?
Obviously setting:
``
framebuffer_width=1920
framebuffer_height=1080
``
with a 848x480 display is not going to look good. Just remove those settings from config.txt.