raspberrypi/firmware #489: mmal: Safe to rely on MMAL_EVENT_FORMAT_CHANGED

julianscheel, 2015-10-21:
Can we be sure that at least one MMAL_EVENT_FORMAT_CHANGED will appear when starting to decode a frame? Or can it happen under some circumstances that the decoder starts to output data without having sent a format changed event?

6by9, 2015-10-21:
The code isn't the clearest, but appears that it should unconditionally send an MMAL_EVENT_FORMAT_CHANGED event having either decoded the header bytes, or got the first image out. It then also monitors the images produced and produces another event should the resolution ever vary.
There does appear to be slightly odd code that if you set format crop.width/height on the input port, those get reflected to the output port, but no event is generated, but I don't think that is a situation that is generally worried about.

julianscheel, 2015-10-30:
@6by9 Thanks for clarification

Ruffio, 2016-07-24:
@julianscheel can this issue be closed now?

julianscheel, 2016-07-24:
Yes, thanks.

closed 2016-07-24 by julianscheel