raspberrypi/firmware #490: lirc does not with device tree?

cryoguy, 2015-10-22:
I tried to set up LIRC with Raspberry PI 2B for a couple of days now with no luck.   The reciever works fine, but not for IR emmitter.     This is what I figured out, if I turned off device tree with deivice_tree= in config.txt, LIRC works fine with the right pin assignment.
sudo cat /sys/kernel/debug/gpio
[sudo] password for root: 
GPIOs 0-53, bcm2708_gpio:
 gpio-17  (lirc_rpi ir/out     ) in  lo    
 gpio-18  (lirc_rpi ir/in      ) in  lo    
 gpio-47  (led0                ) out lo  

and following is when I leave the device-tree on:
sudo cat /sys/kernel/debug/gpio
[sudo] password for root: 
GPIOs 0-53, platform/3f200000.gpio, pinctrl-bcm2835:
 gpio-35  (?                   ) in  hi    
 gpio-47  (?                   ) out lo 

Anyone has any advice?  I need the Device_tree on since it is needed for the touch display to work.

Thanks.

Tony

pelwell, 2015-10-22:
Have you read /boot/overlays/README? How are you configuring lirc?

Ruffio, 2016-06-29:
@cryoguy is this still an issue or can it be closed?

Ruffio, 2016-07-24:
@popcornmix this issue should be closed due to inactivity/lack of response to question asked 9 months ago

closed 2016-07-24 by popcornmix